armbian/build #176: Help

renanrfa, 2016-02-24:
I wonder if I run compile file in a ubuntu server, where many have software installed vsftpd , etc.

My question is whether the image /output/images/ generated by compile.sh comes with vsftpd installed?

I appreciate your response and attention.

ThomasKaiser, 2016-02-24:
Please do an 'apt-cache search vsftpd', choose the right package, install it with 'apt-get install', verify that it works, and then either add the package name to debootstrap-ng.sh or add it to the image through [userpatches/customize-image.sh](http://www.armbian.com/using-armbian-tools/) and think about closing this issue afterwards. Thx.

renanrfa, 2016-02-24:
in the file

userpatches / customize-image.sh

$ RELEASE case in
        wheezy )
        # Your code here
        ;;
        jessie )
        # **_Would be correct if I did that ?_**

_apt- get install vim_
        ;;
        trusty )
        # Your code here
        ;;

ThomasKaiser, 2016-02-24:
Well, Github issues aren't intended to help with basics. Please join http://forum.armbian.com/index.php/forum/12-armbian-build-framework/ and let's discuss there :)

igorpecovnik, 2016-02-24:
@renanrfa 

Yes, this is the proper way. I am closing this since it's not an issue. Help=manual and forum.

closed 2016-02-24 by igorpecovnik