armbian/build #177: Add sunxi-tools and man-db to default packages

ThomasKaiser, 2016-02-24:
It would be helpful if sunxi-tools (at least fex2bin/bin2fex) would be part of every image. The sunxi-tools package seems not to be available on every distro we support so what about cross compiling it into the image?

On trusty for whatever reasons no manual pages are available unless one installs 'man-db' package. Is this by intention or not? If the latter should we add it to the trusty default package list?

zador-blood-stained, 2016-02-24:
sunxi-tools are included in board support package (linux-root-$BOARD...) [here](https://github.com/igorpecovnik/lib/blob/master/makeboarddeb.sh#L97). It will be easy to modify this part so they are installed for all boards.

Looks like "man-db" is not marked as important, so it may not be installed automatically. I'll add it to packages list for all distributions later.

WereCatf, 2016-02-24:
sunxi-tools are compiled in common.sh, but at least on Mint 17.2 the compilation fails when it's trying to cross-compile them; compiling for x86-64 works fine, though. Also, sunxi-tools is already available at least in Debian 8's repos, could just install from there while installing everything else. I do not know if they are available in Debian 7's or Ubuntu Wheezy's repos, but they are available in Wily Werewolf, so I would assume Wheezy too.

zador-blood-stained, 2016-02-24:
Jessie and later for [Debian](https://packages.debian.org/stretch/sunxi-tools)
Vivid and later for [Ubuntu](http://packages.ubuntu.com/ru/xenial/sunxi-tools), though only x86 and x64

zador-blood-stained, 2016-02-24:
Build script wasn't tested on Mint, if
```
lsb_release -sc
```
outputs anything that "trusty", there may be missing dependencies.

WereCatf, 2016-02-24:
Cross-compiling other stuff works fine, it's just sunxi-tools that fails. There's something wrong with the Makefiles, I'd assume. I mean, cross-compiling the kernel and, say, MT7601U works just fine. The build-system also changes the permissions of build-host's /tmp to root-only for some reason and I have to always fix the permissions after build.

ThomasKaiser, 2016-02-24:
@WereCatf  on Trusty and Wheezy the first part of this https://github.com/igorpecovnik/lib/blob/ee1546908974063252abb8c4589eb7d9040e26bf/scripts/h3disp#L198 always fails, that's why I added the function. But this takes time and is rather useless if we could add the tools when building the image (in the latest available version so we could also include _sunxi-fel_ to be able to boot other sunxi devices through USB -- see lima-memtester for example)

zador-blood-stained, 2016-02-24:
@WereCatf `output/debug/install.log` should contain error messages from sunxi-tools cross-compilation, based on them, we may find the reason why compilation fails in your case.

WereCatf, 2016-02-24:
It's just `/usr/lib/gcc-cross/arm-linux-gnueabihf/4.8/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: No such file or directory` for the first two tools, and `/usr/include/sys/io.h:83:3: error: impossible constraint in ‘asm’
   __asm__ __volatile__ ("inl %w1,%0":"=a" (_v):"Nd" (__port));` for meminfo.

Also, these don't actually show up in install.log as the errors aren't logged in there :)

zador-blood-stained, 2016-02-24:
@WereCatf 
This looks like a dependency problem (`libc6-dev-armhf-cross` package should provide these files I believe)

WereCatf, 2016-02-24:
Yup, I had missed that and installing it fixed the compilation.

ThomasKaiser, 2016-02-24:
Since I will be away from keyboard (at least from Armbian) for a few days and Igor's busy in Nuremberg, can one of you take care that all the tools (including _sunxi-fel_ if possible) get installed to /usr/local/bin/ with next build?

If my pull request from today got merged IMO we're ready to ship 5.04 images that already really work for headless purposes and can be agressively advertised since this stuff should work then and users can start to continue working on the missing stuff.

BTW: I also developed a script to 'armbianify' existing OS images for H3 and while this worked as a PoC for loboris' Ubuntu Mate 15.04 image I refrain from following this route since users would try this with every (im)possible combination of OS image with the armbianfy script and will complain if it bricks their images/devices.

zador-blood-stained, 2016-02-24:
> Since I will be away from keyboard (at least from Armbian) for a few days and Igor's busy in Nuremberg, can one of you take care that all the tools (including sunxi-fel if possible) get installed to /usr/local/bin/ with next build?

 I'll try to change sunxi-tools part and push necessary changes to my repo after Igor merges this PR (since Igor is the only one who has push access to this repository and I don't want to create merge conflicts if they can be avoided).

ThomasKaiser, 2016-02-24:
@zador-blood-stained Great! Thx a lot. And yes, avoiding merge conflicts is a good idea since my PR touches the same files. I would believe Igor will merge it tomorrow morning...

BTW: Since sunxi-tools are really small and might help in situations where one user wants to _sunxi-fel_ boot another sunxi board or cure broken script.bin stuff (I did that probably 50 times within the last 2 weeks ;) ) it might be also an idea to include the tools in every image and not only in sunxi based?

igorpecovnik, 2016-02-24:
Sunxi tools are present just the linking is not done properly. I noticed that few days ago.

zador-blood-stained, 2016-02-24:
Looks like sunxi-fel may require compilation in chroot (or providing precompiled binary) due to its dependency on libusb.

zador-blood-stained, 2016-02-24:
Especially since ubuntu repositories don't have `libusb-1.0-0-dev` for armhf.

ThomasKaiser, 2016-02-24:
@zador-blood-stained if you manage to get it working including sunxi-fel it would be great since this is really helpful not only (but most importantly) for lima-memtester. And we still need a lot of feedback regarding the OPi One and DRAM settings.

Throwing in a precompiled armhf binary doesn't hurt IMO.

WereCatf, 2016-02-25:
If precompiled binaries are fine I do have sunxi-tools compiled as a .deb at https://github.com/WereCatf/armbian-debs/raw/master/sunxi-tools_1.0-1_armhf.deb -- I have never played around with debian-packaging before, so it's entirely possible it's not exactly as it should be, but the binaries, at least, do seem to work. I did rename meminfo to sunxi-meminfo, because I felt meminfo was too ambiguous.

zador-blood-stained, 2016-02-26:
Should be fixed: [[1]](https://github.com/zador-blood-stained/lib/commit/aad049832a806a8d61649356401738616d3bd101), [[2]](https://github.com/zador-blood-stained/lib/commit/288e34976598b7064ba2717e087ea1bcd5a53c67)

closed 2016-02-26 by ThomasKaiser