armbian/build #181: Adding an easy way to identify the board you're working on.

ThomasKaiser, 2016-02-25:
What about adding

    declare -x PROMPT_COMMAND='echo -n -e "\033]0;$HOSTNAME (${USER}): $PWD\007"'

by default to /root/.bashrc or /root/.profile? Doing my normal work I'm establishing SSH sessions to at least 5 (and up to 15) hosts continually (customer's servers most of the time) and defining this really helps to switch between terminal windows.

While I could add that through the customization process I thought "what about defining this by default?". Any thoughts? Do you expect drawbacks?

zador-blood-stained, 2016-02-25:
Doesn't the default bash prompt look like `user@hostname directory` ? Or do you want to use different prompt colors for different hosts?

ThomasKaiser, 2016-02-25:
It's not about the prompt, this changes the window title in xterm or in Terminal.app (that is what I use on Linux/OS X) and I would believe in Putty too? I've twenty windows open and can see all windows titles with a keyboard shortcut and easily navigate to the one I want to work on. The prompt alone helps only _when_ I've already switched to the specific SSH session.

zador-blood-stained, 2016-02-25:
Ah, right. I'm mostly using KiTTY (a fork of PuTTY), and it has its own settings for window titles
[image]

Well, changing window titles shouldn't break anything, and I believe `.bashrc` should be the right place to put this in (so it applies only to interactive bash sessions).

ThomasKaiser, 2016-02-25:
Yes, in .bashrc it works -- I simply put it at the end of file:

[image]

ThomasKaiser, 2016-02-25:
I will put this in _makeboarddeb.sh_ as:

    echo -e "ZGVjbGFyZSAteCBQUk9NUFRfQ09NTUFORD0nZWNobyAtbiAtZSAiXDAzM10wOyRIT1NUTkFNRSAo\nJHtVU0VSfSk6ICRQV0RcMDA3IicK" | base64 --decode >>"$destination/root/.bashrc"

ThomasKaiser, 2016-02-25:
Just found out that the idea is not the best since it affects negatively shell output when using a serial console. Or maybe something went wrong when inserting to .bashrc. Anyway, I'll close the issue now.

closed 2016-02-25 by ThomasKaiser

zador-blood-stained, 2016-02-26:
I believe you can chek terminal type ($TERM variable) and declare PROMPT_COMMAND only for those types that support setting window titles.