armbian/build #183: Behaviour of desktop image

ThomasKaiser, 2016-02-26:
Just tried out one of our desktop builds (jessie but that shouldn't make a difference) to realise that I'm not really confident that it is sufficient to start nodm and logging in without authentication as root.

At least nodm should be disabled on the first boot and then enabled by _firstrun_ or even better by _check_first_login.sh_ (it's confusing to see X windows starting when you should wait patiently for the 1st reboot) or maybe even better we provide a small script that guides the user through the following process:

- create a normal user account and add this user to sudo group
- let the user choose a specific window manager and install this instead of nodm or exchange _NODM_USER_ in _/etc/default/nodm_

We could also think about disabling nodm at all and start this guided tour from within [check_first_login.sh](https://github.com/igorpecovnik/lib/blob/master/scripts/check_first_login.sh)? This way the user must authenticate once and choose a root password (which he will forget for sure ;) ) and is then forced to create at least one user account and choose a window manager or decide to go with nodm and his account will be added to _/etc/default/nodm_.

What do you think?

BTW: Just suggestions, I lack both time/skills to implement this and am not that familiar with Desktop @ Linux at all.

Update: The more I think about the more I come to the conclusion that we should disable noadm by default and then use something like this as _check_first_login.sh_:

    #!/bin/bash
    # only do this for interactive shells
    if [ "$-" != "${-#*i}" ]; then
        if [ -f "$HOME/.not_logged_in_yet" ]; then
    		echo -e "\n\e[0;31mThank you for choosing Armbian! Support: \e[1m\e[39mwww.armbian.com\x1B[0m\n"
    		echo -e "Please provide a username (eg. igor): \c"
    		read username
    		RealUserName="$(echo "${username}" | tr '[:upper:]' '[:lower:]' | tr -d -c '[:alpha:]')"
    		adduser ${RealUserName} || reboot
    		usermod -aG sudo ${RealUserName}
    		sed -i "s/NODM_USER=root/NODM_USER=${RealUserName}/" /etc/default/nodm
            rm -f "$HOME/.not_logged_in_yet"
            update-rc.d nodm defaults 90 10
            echo "reboot necessary..." ; sleep 1
            reboot
        fi
    fi

WereCatf, 2016-02-26:
I was just thinking yesterday that it's Bad Hygiene(TM) to even teach users to use root as their user-account; users should, indeed, be guided to create a proper user-account for themselves. This is also one of the reasons I added those udev-rules in my mali-packages -- they allow non-root users to also access H/W-acceleration. I was going to mention this issue at one point or another.

There are a lot of things, to be honest, that I think should be improved upon regarding desktop-usage, like e.g. fbturbo and libvdpau should also be installed as debs so updates to them can be distributed to users much more readily and on the images that have been built without desktop it'd be beneficial to still include a script that helps with installing mali-libs, fbturbo, vdpau etc. and gives a selection of login-managers and desktops to choose from and proceeds with the installation at user's discretion. I had a lot more stuff in mind, but I can remember what at the moment.

Anyways, I am touching things here and there, even if I am being a tad slow about it. I'm a novice coder so I'm no good with low-level stuff, but things like these I can still handle, it's just going to take me a while.

igorpecovnik, 2016-02-26:
Agree, we should build debs, also for hostapd which is also some left over from old days. If you install debian version you get conflict error ...  

Desktop support came late in the project since it was not meant to be here at all. I would be glad if you can take over this part. There is no rush for all this.

If we go for adding a user - shell we done "a must" in first login, like Thomas proposed?

ThomasKaiser, 2016-02-26:
I agree to all of the above. But we have to keep two things in mind:

- Armbian's roots are somewhere else (stable release for servers, main devs not interested in desktop/GUI that much or even at all)
- Shift in user base now that we went the Orange Pi route

Therefore I think it's necessary to apply the aforementioned fix before releasing 5.04 (the script code is confirmed to work with jessie and trusty, but it has to be added to our build system that the above code only gets called for desktop images (check_first_login_desktop.sh?) and that nodm will be installed but disabled so users will never log into the desktop as root). Maybe it's a good idea to force user creation anyway, then check if nodm is installed and start it?

I also agree that this stuff should be available as debs but this can come later. IMO we should release 5.04 rather soon even if it's not perfectly upgradeable and then let all the improvements go into 5.05. Since the official available images from a week ago all really suck. BTW: Your work is highly appreciated, please keep on rocking! :)

I'm out of the game for now. My server hosting the preliminary images the last days draw too much attention, was hacked and shut down (I would suspect this is a warning from a white hat). Now busy setting another server up.

BTW: runlevels for nodm are wrong:

    macbookpro-tk:Orange Pi PC tk$ slogin -l root 192.168.83.118
    The authenticity of host '192.168.83.118 (192.168.83.118)' can't be established.
    RSA key fingerprint is 1b:5d:03:7e:e8:ff:20:31:41:8b:47:97:60:bd:65:6d.
    Are you sure you want to continue connecting (yes/no)? yes
    Warning: Permanently added '192.168.83.118' (RSA) to the list of known hosts.
    root@192.168.83.118's password: 
    You are required to change your password immediately (root enforced)
    
    Welcome to ARMBIAN Ubuntu 14.04 LTS 3.4.110-sun8i 
    
    System load:   0.00            	Up time:       2 sec		
    Memory usage:  3 % of 1001Mb 	IP:            
    CPU temp:      30°C           	
    Usage of /:    100% of 2.4G   	
    
    [ 73 updates to install: apt-get upgrade ]
    
    
    The programs included with the Ubuntu system are free software;
    the exact distribution terms for each program are described in the
    individual files in /usr/share/doc/*/copyright.
    
    Ubuntu comes with ABSOLUTELY NO WARRANTY, to the extent permitted by
    applicable law.
    
    Changing password for root.
    (current) UNIX password: 
    Enter new UNIX password: 
    Retype new UNIX password: 
    
    Thank you for choosing Armbian! Support: www.armbian.com
    
    Creating new account. Please provide a username (eg. igor): tk
    Adding user `tk' ...
    Adding new group `tk' (1000) ...
    Adding new user `tk' (1000) with group `tk' ...
    Creating home directory `/home/tk' ...
    Copying files from `/etc/skel' ...
    Enter new UNIX password: 
    Retype new UNIX password: 
    passwd: password updated successfully
    Changing the user information for tk
    Enter the new value, or press ENTER for the default
    	Full Name []: Thomas Kaiser
    	Room Number []: 
    	Work Phone []: 
    	Home Phone []: 
    	Other []: 
    Is the information correct? [Y/n] 
    update-rc.d: warning: default stop runlevel arguments (0 1 6) do not match nodm Default-Stop values (none)
     System start/stop links for /etc/init.d/nodm already exist.
    
    Reboot necessary...
    
    Broadcast message from root@orangepipc
    	(/dev/pts/0) at 11:08 ...
    
    The system is going down for reboot NOW!
    root@orangepipc:~# Connection to 192.168.83.118 closed by remote host.
    Connection to 192.168.83.118 closed.

WereCatf, 2016-02-26:
I can take a look at setting up scripts for building fbturbo and libvdpau into debs, I'd be uploading them in my armbian-debs repo.

But yeah, I agree on enforcing creation of a proper user-account.

ThomasKaiser, 2016-02-26:
User creation is part of #185

igorpecovnik, 2016-02-26:
Video playing ... I have some problems. Is it working for you?

ThomasKaiser, 2016-02-26:
Nope, I would suspect CedarX support still missing?

BTW: Next PR shows expected behaviour, no desktop at 1st login, forced user creation, then

    sed -i "s/NODM_ENABLED=\(.*\)/NODM_ENABLED=true/g" /etc/default/nodm

followed by reboot. BTW: With our image suspend to RAM and resume through power button works flawlessly :)

And I realised that zador patched sunxi-tools installation so that I revert my stuff again.

WereCatf, 2016-02-26:
I haven't even tried playing video yet, are you talking about libvdpau-assisted video or just software-decoded? I do notice that you're pulling some specific commit from libvdpau instead of the latest from their git, maybe try building latest git instead?

igorpecovnik, 2016-02-26:
yes, libvdpau assisted ... The reason of using that specific commit because it breaks from latest commit. It used to. I need to check if it's working now.

WereCatf, 2016-02-26:
Did a newer libvdpau fix your issue?

ThomasKaiser, 2016-02-26:
BTW: If we're already talking about video (IMO stuff for 5.05 ;) then a lot of useful code is contained in Jernej's patch 0000 [here](https://github.com/jernejsk/OpenELEC-OPi2/tree/master/projects/H3/patches/kodi). Also 0001 and 0002 should fix the audio issues users already reported.

igorpecovnik, 2016-02-26:
If video is not done yet, we can wait with it ... sure. BTW. OpenGLES works on A20 install out of the box.

igorpecovnik, 2016-02-26:
One more think. If we are tuning deskop ... where is default new profile for new user, that we have the same desktop as it was designed for root, icons and wallpaper.

ThomasKaiser, 2016-02-26:
Seems like it works out of the box after I exchanged the untar location from /root to /etc/skel/. The only thing to think about is the contents of trusty-desktop.tgz (and the two others). I would prefer to link not to the forum but armbian.com/documentation instead and link from there to the H3 Mini FAQ on the top (since we're generating a _few_ more users the next days).

[image]

WereCatf, 2016-02-26:
@ThomasKaiser That patch you mentioned is for Kodi, it's not applicable to Armbian, as far as I can see.

ThomasKaiser, 2016-02-26:
Sure but I (naive) thought the code could be re-used. I've to admit that I don't have the slightest idea about this CedarX stuff :)

ThomasKaiser, 2016-02-26:
Since we now ensure also through #187 that H3 users when running 3.4 kernel are directly pointed to _h3disp_ and we resolved the other problems I close the issue now. Video acceleration is IMO stuff for later.

IMHO we're ready to release 5.04 images for Oranges now to get a flood of feedback.

BTW: Any chance we could add both _htc_9271.fw_ and _htc_7010.fw_ to /lib/firmware/?

ThomasKaiser, 2016-02-26:
@igorpecovnik Oops, I've been wrong. Desktop items are missing now, just tried it again with an Ubuntu desktop image and it comes up without Armbian wallpaper.

igorpecovnik, 2016-02-26:
@ThomasKaiser 
BTW: Any chance we could add both htc_9271.fw and htc_7010.fw to /lib/firmware/?

I assume those files?
http://linuxwireless.org/download/htc_fw/1.3/

ThomasKaiser, 2016-02-26:
Yes, since the driver is already included it would be good to support [AR9271 WiFi dongles](https://olimex.wordpress.com/2015/12/20/new-product-soon-to-be-in-stock-mod-wifi-ar9271-ant-wifi-usb-dongle-with-open-source-firmware-and-linux-drivers/) out of the box

closed 2016-02-26 by igorpecovnik