armbian/build #188: common.sh conditional broken

WereCatf, 2016-02-26:
Line 316, conditional broken. I'm not sure what the point there is: is there no longer sunxi-kernel, only sun8i? Or was the conditional intended to check for either of those? Since I'm not sure of the point I ain't trying to fix it.

zador-blood-stained, 2016-02-26:
"sunxi" part comes from here [(main.sh, line 259)](https://github.com/igorpecovnik/lib/blob/master/main.sh#L259). Conditional should check for either of those (which is not necessary I believe), but "-o" operator is specific to single bracket test, not for double bracket test.

ThomasKaiser, 2016-02-26:
Should I switch to single brackets or split it into two parts? I believe the latter is more safe.

ThomasKaiser, 2016-02-26:
Should be fixed if #189 is merged. Sorry.

zador-blood-stained, 2016-02-26:
```
if [[ -n "$MISC5_DIR" && $BRANCH != "next" && $LINUXSOURCEDIR == *sun[x8]i* ]]; then
```
This should work. At least the syntax is correct.

BTW, you can use `bash -n file.sh` to check for syntax errors without executing any commands in file. Works in Git for Windows shell too.

igorpecovnik, 2016-02-26:
OK, I am building 5.04 ... got enough for today ;)

closed 2016-02-26 by igorpecovnik

ThomasKaiser, 2016-02-26:
I just applied Jernej's backported firmware loader patch, will create an image and send Martin Ayotte a link to the image to let him test. If everything's alright it goes into 5.05 then.