armbian/build #194: Missing udev rules for Mali

ThomasKaiser, 2016-02-29:
According to [this post](http://forum.armbian.com/index.php/topic/617-wip-orange-pi-one-support-for-the-upcoming-orange-pi-one/?p=5719) we need udev rules that allow users in group video to make use of Mali. Since I don't know where to start I just opened an issue.

User creation should already take care of the correct [group membership](https://github.com/igorpecovnik/lib/blob/master/scripts/check_first_login.sh#L10)

igorpecovnik, 2016-02-29:
We need to create udev rules file as written in docs. I think that's all ... user is getting into video group within your script.

igorpecovnik, 2016-02-29:
I am doing it.

igorpecovnik, 2016-02-29:
It's already prepared but included.
https://github.com/igorpecovnik/lib/tree/master/config/sunxi-udev

WereCatf, 2016-02-29:
@igorpecovnik I had forgotten to actually make use of the files, it's fixed now.

closed 2016-02-29 by igorpecovnik