armbian/build #200: Increased kernel logging at 1st boot

ThomasKaiser, 2016-03-01:
What about changing loglevel to 8 in default boot.cmd and then

    sed -i 's/loglevel=1/loglevel=8/' /boot/boot.cmd
    mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr

from within _firstrun_ or even better _check_first_login.sh_? Background [here](http://forum.armbian.com/index.php/topic/617-wip-orange-pi-one-support-for-the-upcoming-orange-pi-one/?p=5764)

And maybe also ship _/usr/local/bin/toggle-verbosity_ (changing also _ console_)?

igorpecovnik, 2016-03-01:
First boot is actually not critical since we test all the images and they are booting / must boot ;) I support the idea of having a command for enabling / disabling. That would be something nice for end users. Type: toogle-verbosity and copy / paste log. Console could be as optional parameter ?

zador-blood-stained, 2016-03-01:
@ThomasKaiser 
Or even better - something like this
```
if ext4load mmc 0 0x00000000 /boot/.verbose
then
setenv verbosity 7
else
setenv verbosity 1
fi
setenv bootargs "... loglevel=${verbosity}"
```
Then you can ship images with /boot/.verbose file present and delete it after successfull first boot.

ThomasKaiser, 2016-03-01:
Igor's right: The 1st boot isn't the problem but subsequent (mostly the 2nd due to auto detection woes).

@zador-blood-stained the idea is great. What about combining it in the following way:

- armhwinfo creates /boot/.verbose on every boot
- toogle-verbosity creates/deleted /boot/.force-verbose
- armhwinfo gets a stop case and deletes /boot/.verbose if /boot/.force-verbose doesn't exist

This way we enhance verbosity automagically in case a board crashes and enable the user to control verbosity too. 

Regarding console: In case we adopt this [/etc/armbian-boards.conf approach](http://forum.armbian.com/index.php/topic/751-rfc-support-cortex-a53arm64/?p=5670) we could write the correct console settings into /boot/.verbose and let u-boot read them? Possible?

zador-blood-stained, 2016-03-01:
I believe u-boot can read environment variables from specially prepared files (plaintext won't work). Loading console configuration from a file is a good idea, but it's just a different approach to a same problem.

ThomasKaiser, 2016-03-04:
I commited the skeletons for armbianmonitor's 2 main components with #204 

It's not very useful for the original idea but maybe we can add armbianmonitor to /usr/bin (after some cleanups/sprucing) to provide at least

- armbianmonitor -b (toggle boot verbosity already)
- armbianmonitor -u (upload armhwinfo.log -- [the 1st request already worked](http://forum.armbian.com/index.php/topic/71-armhwinfo-can-take-a-long-time-to-finish/?view=getlastpost)
- enable primitive monitoring on the command line (was inspired by [longsleep](https://github.com/longsleep/build-pine64-image/issues/4#issuecomment-190904404) and could be enhanced a bit but should already be somewhat useful when output is formatted better and a few more datasources that are not available using the standard tools might be included, scaling_cur_cpufreq being the one I definitely forgot)

Regarding the original _armbianmonitoring_ approach I still want to follow @zador-blood-stained suggestion to switch to a template system (good idea!) but hadn't had the time. Idea is to use templates that RPi-Monitor can read but where the armbianmonitor specific stuff (which sysfs to choose for what) is also contained in comments so that we move from [code blocks](https://github.com/ThomasKaiser/lib/blob/cf07fb0ad821e9eec583541fc452cc45c0fbba18/scripts/armbianmonitor/armbianmonitor-daemon#L170-L216) to templates.

I won't look into this at least the next two weeks so in case anyone wants to have a look or tweak something: feel free to do so. IMO two showstoppers exist :

- naming conventions (related to @zador-blood-stained new configuration templates): it would be great if we could use $BOARD/$NAME in the build system in a way that it's always consistent ($ID generation in armhwinfo and also armbianmonitor template names. If names are consistent no special logic is needed, simply parse /var/run/machine.id, remove whitespace and convert to lower case and look whether an armbianmonitor template exists, if not choose unknown_device.template instead)

- disk detection (IMO this is a really nice killer feature but we will need extensive feedback from our users for USB disks, so I thought about preparing the part to detect disk infos from within armbianmonitor with the upload functionality and to provide our users with this to get at least a bit more feedback -- not done yet)

ThomasKaiser, 2016-03-04:
Sorry, two more remarks regarding a potential inclusion of armbianmonitor for now

- armbianmonitor -U could also add the output of _zcat /proc/config.gz_ to the uploaded stuff and in case smartctl is on the system the output of _smartctl -q noserial -a_ for every block device ("-q noserial" omits the drive's serial numbers, something that might prevent paranoid people to upload debug stuff -- same applies to hostname and IP addresses in the output too)
- And the _armbianmonitor_ -m idea was to display detailed CPU statistics (especially %sys and %iowait) since this can help identify bottlenecks more easily when used together with other data sources -- but since tools like top do it display also it's not that important

I tried to implement the latter yesterday but clearly lost against _printf_ ;)

ThomasKaiser, 2016-03-22:
Since everything (except of packaging armbianmonitor and deploying it through apt.armbian.com) is done we can close this.

closed 2016-03-22 by ThomasKaiser