armbian/build #201: Build/provide mpv package with vdpau acceleration

ThomasKaiser, 2016-03-01:
Disclaimer: I've not the slightest idea about desktop/GUI/video issues but fear that questions regarding this will come soon in the forum.

Are we able to build/provide a mpv package [built like this](http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=878&extra=page%3D1) for A10/A20/H3?

igorpecovnik, 2016-03-01:
video playing is working with stock mpv ... but there are some issues. sound is not synced and what else?

ThomasKaiser, 2016-03-01:
No idea, i've not even tried it yet. But since @WereCatf  already provided Mali acceleration I thought maybe he is already preparing something is this area.

I've really no clues regarding video stuff but just thought it would/might be nice to be able to add 'use the included mpv package on our desktop images to enjoy HW accelerated video where possible' to the user documentation in advance instead of answering the question multiple times in the forum :)

igorpecovnik, 2016-03-01:
That's true. I was already thinking to provide two icons on the desktop:

1. Play video
2. Test 3D acceleration

Both things works out of the box on A20 (H3 video not yet), but we need to add mpv to the desktop package base. That far is where I am willing to go. Don't want to play with further optimizations - if someone take it from here, yes I'll support and include.

WereCatf, 2016-03-01:
I just happened to play with video a little yesterday, compiled the latest libvdpau-sunxi from their git - repo and such and while playing H/W-accelerated H.264 seemed to work otherwise fine I didn't test with audio. Couldn't test HEVC, mpv always insisted on using software-decoding for that and it resulted in corrupted display -- would probably require recompiling mpv and possibly some patching to get it to play HEVC with H/W-decoding.

I can test in a few hours if I'm seeing out-of-sync audio with latest libvdpau-sunxi, but if I am I wouldn't know what to do about it and as such I wouldn't be able to fix it. It'll take some more hours to try and see if I can get HEVC-decoding to work.

igorpecovnik, 2016-03-01:
Take your time. Video / desktop is not our primary goal but if it can work good, why not prepare it.

WereCatf, 2016-03-01:
Tbh, Jernej is also having troubles with his OpenElec - build using libvdpau-sunxi, so I wouldn't expect this stuff to work well for a while longer.

WereCatf, 2016-03-02:
Yeah, no idea what to do about the timing-issues for now other than wait for them to patch libvdpau-sunxi up some more.

ThomasKaiser, 2016-03-02:
@WereCatf what you're currently trying out is exactly what [Jernej now suggests?](http://forum.armbian.com/index.php/topic/617-wip-orange-pi-one-support-for-the-upcoming-orange-pi-one/?view=getlastpost) isn't it? The confused video NOOB is asking ;)

WereCatf, 2016-03-02:
@ThomasKaiser Um, yes and no: the Armbian - images already pack libvdpau-sunxi, so he's not talking about anything new per se. The version used by Armbian - images is just quite old and I, instead, compiled the latest stuff from github -- I'm still getting timing - issues, so using newer libs didn't fix that. (I can still submit a commit for Armbian that builds newer libs, if you guys want that; I just haven't seen the point, because it's not quite useable anyways until someone figures out how to fix the timing - stuff.)

Background: CedarX is binary-only, Allwinner-specific stuff and software has to be specifically designed to support that, whereas vdpau is used by e.g. NVIDIA's drivers on x86 and is thus much more widely supported. Most Linux video-players support vdpau already, so obviously it makes more sense to just ignore CedarX and use vdpau.

This applies only to playback, though: vdpau can't be used for encoding. CedarX supports encoding, too, but there's nothing out there that uses it, as far as I can tell.

ThomasKaiser, 2016-03-02:
Thx for the explanation. If it doesn't matter I've no opinion on that (and also no knowledge whether this vdpau stuff already works identical on A20 and on H3?)

WereCatf, 2016-03-02:
@ThomasKaiser It should work just the same on A20, except on the H3 it also supports HEVC. I don't own an A20, though, so I have no idea how well or how badly it works.

ThomasKaiser, 2016-03-02:
If there aren't real improvements in bundling a newer version we could skip this. IIRC anyone (you?) wanted to repackage the desktop stuff so that we can also ship the whole thing as debs and do not have to rebuild it everytime we create a desktop build?

And Igor wanted to bundle a mpv package linked to the desktop, right? But we're not in a hurry at all so better think twice about that and do it right :)

WereCatf, 2016-03-03:
Xfwm4, the window-manager we install by default, uses software-compositing by default -- try playing some H.264 - video with mpv using vdpau and you'll see mpv doesn't use much CPU, but X is using ~66% CPU! Then, disable compositing with "xfconf-query -c xfwm4 -p /general/use_compositing -t bool -s false" and try it again: you'll notice X's CPU-usage has dropped to ~14%.

That is quite significant, especially on such a low-performance device. Using kwin_gles allows for utilizing eGL/GLESv2 for hardware-accelerated compositing and drops CPU-usage by a very large amount, both when playing video and in general, but it carries a lot of dependencies with it.

WereCatf, 2016-03-05:
Are there any plans to move the desktop to use Ubuntu 16.04 instead of 14.04 once it's released? Building mpv is quite a mess and finding out which commits work and which ones don't work on Debian Wheezy and Ubuntu 14.04 is, well, a lot of work, whereas on 16.04 there is a reasonably recent build of mpv already in the repos and we could just simply install it in the desktop - images. This would leave the other distros out, though.

Also, I have a crappy script at https://dl.dropboxusercontent.com/u/11811685/compile.tar.bz2 that builds fbturbo, libvdpau, sunxi-mali and friends into debs. I have not tested whether the script works under 14.04 or Wheezy, I've only run it under Jessie. I have also not yet tested whether the debs work under Wheezy or Ubuntu 14.04, but they should. The script, on the first run, builds all the debs, but on subsequent runs it only checks if there are new commits to the libraries/fbturbo and builds new debs only if new commits were found. Feel free to modify it, I'm not going to be uploading more debs to github as this is definitely not the right place for that stuff.

igorpecovnik, 2016-03-05:
Thanks for update, I'll check and test the script asap ... Ubuntu 16.04 is planned.

WereCatf, 2016-03-05:
Here ( https://dl.dropboxusercontent.com/u/11811685/mpv_0.16.0_armhf.deb ) is latest git-build of mpv that is capable of using vdpau for decoding of HEVC, too. I am pretty sure it'll only work on Jessie or newer distros, like e.g. Ubuntu 16.04, due to many of the dependencies being terribly out-of-date on Wheezy and Ubuntu 14.04. I just tested it on my OPi PC running Jessie and H/W-decoding of both H.264 and non-10-bit HEVC works beautifully -- 10-bit HEVC is not supported by hardware.

Grab files http://jell.yfish.us/ and use `mpv --vo=vdpau --hwdec=vdpau --hwdec-codecs=all [filename]` if you wish to test.

ThomasKaiser, 2016-03-05:
Great efforts! Hmm... what about the following (if it complicates our build process not too much)? We use the H3 platform for testing 16.04 since currently no OPi user really expects 100% stable images and they know that we need their feedback for the whole situation to improve.

So we add Xenial now and switch for sun8i already to it. This might allow to ship with this mpv package included in the desktop image (and a link on the desktop to mpv) to collect feedback from OPi users the next few weeks and then when 16.04 will be released switch on all other platforms to it without any hassles (same with u-boot, we can then skip the special behaviour for sun8i and update to 2016.03 for all platforms?).

Disclaimer: I'm still not 100% familiar with our build process so maybe I underestimate the whole idea totally.

WereCatf, 2016-03-05:
@ThomasKaiser There are no armhf-binaries built in Xenial - repos yet, I assume they'll build those only for final release. I just tried this a few days ago and would've submitted a commit otherwise. So no, moving to 16.04 already now isn't going to work, unless you want to build everything manually!

ThomasKaiser, 2016-03-05:
@WereCatf WTF? Ok, I played with Xenial around the last few days but that's arm64 (and there are some packages missing or not adopted already to arm64). But what about https://launchpad.net/ubuntu/xenial/armhf then?

WereCatf, 2016-03-05:
@ThomasKaiser Well, I dunno. Maybe I did something wrong or maybe the packages weren't there when I tried it? If you can build Armbian based on arm-binaries by all means, submit a commit. If we're going to be moving to 16.04 anyways we may as well test it out already.

WereCatf, 2016-03-05:
By the way, vdpau works with VLC, too, from 2.2.0 onwards. Decoding HEVC didn't work, I'm just trying to find out if newer builds of VLC support using vdpau for HEVC.

zador-blood-stained, 2016-03-05:
@WereCatf http://ports.ubuntu.com/dists/xenial/main/binary-armhf/
http://ports.ubuntu.com is used for older armhf debootstrap too AFAIK.

ThomasKaiser, 2016-03-05:
Attention! NOOB question: How does http://apt.armbian.com/dists/xenial/ gets populated?

zador-blood-stained, 2016-03-05:
> Attention! NOOB question: How does http://apt.armbian.com/dists/xenial/ gets populated? Is this a manual step or can this be automated too?

repo-update.sh & addtorepo() in general.sh?

igorpecovnik, 2016-03-05:
> repo-update.sh & addtorepo() in general.sh?

yes. only extra "script" is rsync to the server

ThomasKaiser, 2016-03-05:
Ok, the last one puzzled me a bit. So it's some sort of chicken&egg problem? Unless the packages are already rsynced the build will fail with 

    Failed to fetch http://apt.armbian.com/dists/xenial/main/binary-armhf/Packages

when trying to add Xenial?

igorpecovnik, 2016-03-05:
Directory / distro xenial does not exists on server yet. It will once packages will be added ...

WereCatf, 2016-03-06:
I am preparing a commit that will install updated debs, including mpv, and setting defaults for mpv so that it'll use VDPAU for decoding if possible. That should make it quite plug-and-play for end-users. 

Alas, do I submit the debs with the commit or do you want to host them somewhere else? If so I'll post a link to a zip-file and then you'll have to tell me where you're hosting them or if you're adding them to Armbian-repos so they can be installed simply with apt-get -- your decision. And one of the problems with playing video is that xfwm4 in our images uses compositing by default and that eats a helluva lot of CPU -- should we disable compositing by default on sunxi so as to enable a bit smoother playback, or just stick with it and let users disable it if they wish?

ThomasKaiser, 2016-03-06:
IMO we should disable it. The average user of desktop images won't touch defaults.

WereCatf, 2016-03-06:
@ThomasKaiser Personally I lean towards that myself, too, but I wanted to ask your opinions. Users could, theoretically, expect to see pretty transparency-effects and stuff like they see on the other boards and be a bit taken aback to find out sunxi-boards lack those (by default, that is), but on the other hand, good performance is more useful and I would imagine that's important to a larger group of people than compositing.

ThomasKaiser, 2016-03-06:
@WereCatf I just played around with Armbian 5.05 Ubuntu Trusty desktop and trying to use the GUI is close to impossible (at least for me who is lacking patience). The SD card currently used is a bit slow especially regarding IOPS and maybe that's also responsible for the mess. I already searched around how to disable stuff and am not even able to disable the screensaver that starts to eat one CPU core after a while (so clearly I'm the wrong person to judge about Linux with GUI ;) )

So in my very personal opinion everything that speeds up the desktop experience might be worth a try even if transparency-effects are affected.

WereCatf, 2016-03-06:
@ThomasKaiser Play around with https://dl.dropboxusercontent.com/u/11811685/Armbian_5.05_Orangepih3_Debian_jessie_3.4.110_desktop.zip instead, much more fun -- includes all the stuff I mentioned above, including VDPAU-support for HEVC, mpv that uses VDPAU by default, compositing disabled by default and working 3D, too. Decoding of HEVC doesn't work with VLC using VDPAU, though -- I couldn't make it work. I think VLC just doesn't support that yet.

WereCatf, 2016-03-06:
@igorpecovnik is busy somewhere else, so I'll just leave these links here for him: https://dl.dropboxusercontent.com/u/11811685/compile.zip has updated the script that builds those debs, there were a bunch of bugs in the earlier one. https://dl.dropboxusercontent.com/u/11811685/debs.zip has already-built debs for now.

igorpecovnik, 2016-03-07:
https://github.com/igorpecovnik/lib/pull/215

closed 2016-03-07 by igorpecovnik

ThomasKaiser, 2016-03-07:
BTW: @WereCatf this mpv stuff on Debian with the new default settings works way better than before. But am I'm the only one not being able to shutdown / reboot / suspend the board from an unprivileged user account in the GUI ('sudo reboot' in an xterm works of course)? There seems to be something different between Trusty and Jessie in this area?

WereCatf, 2016-03-07:
I dunno, I haven't tried using the shutdown/reboot/suspend - stuff from the menu. It's certainly nothing I have changed. I did notice that for a normal user it said "Command not found" when trying to issue "poweroff" from CLI, so it's either a path-issue or permission-issue. Or maybe even both?

You could try seeing where "poweroff" is located, adjusting the user's path to include the location and seeing if things work better. And maybe report back here about your findings.

ThomasKaiser, 2016-03-07:
(Googling and) Installing gksu package solved the problem partially (so we should add it for the desktop builds). The shutdown GUI link in the upper right corner is available and working but suspend is still greyed out (a bit sad, that was also one very nice feature but maybe it's good to promote it later when we switched the meaning of the leds on sun8i so we use green for power and switch to red before the boards starts to suspend to RAM)

ThomasKaiser, 2016-03-07:
Update: the 'Shutdown' menu item doesn't work (GUI disappears and gets back after some time, since I left the room I thought 'reboot? strange'). But clicking on 'Log out' showed then 3 choices: Logout, reboot and shutdown. Reboot worked and now the systems seems somewhat broken (can't login through SSH, not able to start local programs, one CPU core busy doing $something according to RPi-Monitor). Trying to shutdown/reboot ends up with

[image]

After a forced power cycle now everything's back to normal. Hmm... needs a bit more testing.

ThomasKaiser, 2016-03-09:
BTW: http://linux-sunxi.org/User:Rellla/Armbian

WereCatf, 2016-03-10:
The two patches from https://github.com/rellla/lib/commit/622a9b075661f95a8bdd457b652c5fb5b0f41ce3 may be worth adding to Armbian. They do apply cleanly and the kernel compiles fine with them, but since I don't have any A20-hardware I cannot test any further than that. They do not apply to the H3-kernel, though, only sun7i.

ThomasKaiser, 2016-03-10:
I would suspect he will send a PR later. And I already thought about how to deal with the situation (different legacy kernel configs for desktop and 'server' builds? Trying to provide up-to-date mpv packages on apt.armbian.com?)

But no idea. Let's wait and see :)

WereCatf, 2016-03-10:
Might as well add pcsx-rearmed to the list of stuff to build, then. Neither Ubuntu or Debian provide pcsx-rearmed in their repos, they use pcsx-reloaded, but pcsx-rearmed performs a whole lot better on ARM-devices. I built a temporary version of it just for testing and it actually works impressively well, all things considered.

Personally, I have no interest in using my OPi as a desktop, as a media-player or for playing games on or such, but a lot of people do seem to like such stuff and including stuff like this in the repos might make Armbian more appealing for them.

igorpecovnik, 2016-03-23:
Youtube playing is no more with this version ... do we need some special compilation procedure?

WereCatf, 2016-03-23:
Youtube-playing?

ThomasKaiser, 2016-03-23:
Just tested it with a desktop Jessie image I built 2 days ago and with 'Linux version 3.4.110-sun8i (root@armbian) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11-1) ) #2 SMP PREEMPT Mon Mar 21 18:44:33 CET 2016' youtube still works

WereCatf, 2016-03-23:
I hadn't even realized you could use mpv for Youtube. Well, I don't know. As far as I know the version I built was compiled with pretty much everything possible enabled, but it is entirely possible that something is missing. On the other hand, it's also possible they've changed something on Google's ending, I really don't know.

ThomasKaiser, 2016-03-23:
@igorpecovnik what are you talking about? Playing youtube videos in Iceweasel (that's what I tested -- not related to mpv at all ;) ) or something else?

igorpecovnik, 2016-03-23:
`mpv --vo=vdpau --hwdec=vdpau --framedrop=vo --hwdec-codecs=all https://www.youtube.com/watch?v=sqqvQRHScgY`

This used to work. The error is:
[ytdl_hook] youtube-dl failed, trying to play URL directly ... 
[ffmpeg] tls: The TLS connection was non-properly terminated.
Failed to recognize file format.

Which might lead to:
https://github.com/mpv-player/mpv/issues/1291

WereCatf, 2016-03-23:
Oh, I think the mpv I provided doesn't have lua enabled

zador-blood-stained, 2016-03-23:
youtube-dl is an [external program](https://rg3.github.io/youtube-dl/download.html), so even if LUA was enabled in mpv, there is nothing for mpv to launch.

igorpecovnik, 2016-03-23:
So we miss a package and a LUA support, right?

WereCatf, 2016-03-23:
Looks like it. As far as I know, youtube-dl is available in the repos already, though I don't know how outdated it is.

zador-blood-stained, 2016-03-23:
It's not that easy with youtube-dl. It is present in Debian/Ubuntu repositories, but it may not be updated as frequently as it needs to be (on stable branches like Jessie). For end users it's better to install it manually and update if something stopped working.

igorpecovnik, 2016-03-23:
In Jessie we got 2014.08.05 ... huh. I guess this can be added to the build / packaging process. The functionality is not top urgent but nice to have.

WereCatf, 2016-03-23:
By the way, since libvdpau-sunxi now depends on pixman it should be possible to enable G2D for subtitles as it uses pixman to emulate the G2D-hardware on H3, as far as I've understood. Anyone of you wanna test doing export VDPAU_OSD=1 and testing subtitles with mpv? If it works then that should probably be exported by default, too.

ThomasKaiser, 2016-03-25:
BTW: I applied the 2 patches from Rellla to get UMP, enabled CMA with 192 MB size and [provided a test package](http://forum.armbian.com/index.php/topic/728-kodixbmc-on-armbian-with-hw-accleration-possible/?p=6845). Maybe we get some feedback.

But then I wonder how we should deal with this? Different kernel builds for desktop and CLI images? Shipping with active CMA and disabling it by setting `cma=0` in kernel command line for CLI images (we still don't understand whether there's a relationship with the kswapd issue and maybe enabling/using CMA breaks other stuff as well, at least less available memory isn't something we want for 'server images').

@WereCatf can you estimate the amount of work needed to put [all the software Rellla provides an installation log for](http://linux-sunxi.org/User:Rellla/Armbian) into debs hosted on apt.armbian.com?

WereCatf, 2016-03-25:
@ThomasKaiser Using kernel command-line to disable CMA for CLI-images sounds the saner approach, shipping multiple builds of the same kernel just to enable/disable something like that would be a silly thing to do. Maybe consolidate h3disp and all the other various scripts and stuff into an armbian-config, similar to raspi-config on Raspbian, for easy toggling of settings -- including CMA?

As for those debs: if one was just going to do a quick-and-dirty one-off build of them all it would take like a day or two per architecture per distro, I guess? Are you asking me to do that?

igorpecovnik, 2016-03-25:
Speaking of debs - I would say we need some more generic approach - we need to pack other debs as well and we can do it in one streak. Now things are scattered around or stored binary. Since arm64 introduction all binary packages become deprecated ... Ideas for generic approach - I'll also work on this.

WereCatf, 2016-03-25:
Well, when my OPi died it destroyed the SD-card with it and took the build-script I had made with it, so I took the chance and redid a much more generic one from scratch. It can now build for multiple distros and multiple archs and marks the resulting debs appropriately. If you want to I can post a link here in a week or so, I guess, so you can take a look and provide feedback on it.

igorpecovnik, 2016-03-25:
Sure, take your time and just push it to the scripts -> lib/packaging.sh or similar, I'll join at once with other necessary things.

[1 more reply]