armbian/build #205: RAID 6 calculations with vanilla kernel

ThomasKaiser, 2016-03-04:
With vanilla kernel on every boot in a very early stage the mdraid code tries to detect the most efficient XOR algorithm (for example starting at [line 95 here](http://sprunge.us/dFBB)). This does not only delay the boot process by a few seconds but might also put significant load on the machine in an early state where no monitoring is already up and running (might be a problem on boards with weak power supplies, endless reboot loops and something like that).

Are we able to change the config so this gets only executed if the user really wants to use RAID (IMO a not so great idea with SBC at all, the Clearfog currently being the only exception)?

zador-blood-stained, 2016-03-04:
MD_RAID456 and **BTRFS_FS** depend on RAID6_PQ, which causes [RAID algos checks](https://github.com/torvalds/linux/blob/master/lib/raid6/algos.c#L131)
So unless you want to modularize btrfs (you won't be able to use it as root filesystem without initrd), these checks can't be disabled as far as I see.

ThomasKaiser, 2016-03-04:
Thx for the insights! Given that RAID5/6 is still considered somewhat experimental AFAIK and especially that RAID6 is absolutely useless on all the devices we support (even on the Clearfog)... what about patching the dependency away?

zador-blood-stained, 2016-03-04:
> what about patching the dependency away?

I don't think it's a good idea. RAID6_PQ is also responsible for other library functions.

ThomasKaiser, 2016-03-04:
Ok, agreed. But we should keep this in mind when users get back to us reporting early boot problems with mainline kernel since this _might_ have an influence when DC-IN is insufficient. Thx again!

closed 2016-03-04 by ThomasKaiser