armbian/build #209: Provide .7z instead of .zip for end users?

ThomasKaiser, 2016-03-05:
Armbian images are already pretty small since we're doing it right (building an image automated from scratch and not hacking around manually so we do end up with a rootfs as small as possible that's not heavily fragmented already). But it could be always better:

    449M	Armbian_5.05_Orangepih3_Ubuntu_trusty_3.4.110_desktop.7z
    648M	Armbian_5.05_Orangepih3_Ubuntu_trusty_3.4.110_desktop.zip

Thats _7za a -t7z -m0=lzma -mx=9 -mfb=64 -md=32m -ms=on_ saving another 30 percent which might be interesting for users with small Internet bandwidths. Compression is heavy but regarding decompression times it's 38 seconds for .7z vs. 16.5 for .zip (2.7 GHz dual core Core i5 running OS X 10.10):

    macbookpro-tk:~ tk$ time 7za e Armbian_5.05_Orangepih3_Ubuntu_trusty_3.4.110_desktop.7z 
    
    7-Zip (A) [64] 9.20  Copyright (c) 1999-2010 Igor Pavlov  2010-11-18
    p7zip Version 9.20 (locale=utf8,Utf16=on,HugeFiles=on,4 CPUs)
    
    Processing archive: Armbian_5.05_Orangepih3_Ubuntu_trusty_3.4.110_desktop.7z
    
    Extracting  Armbian_5.05_Orangepih3_Ubuntu_trusty_3.4.110_desktop.raw
    
    Everything is Ok
    
    Size:       2509484032
    Compressed: 470786316
    
    real	0m37.682s
    user	0m34.930s
    sys	0m1.644s

vs.

    macbookpro-tk:~ tk$ time unzip Armbian_5.05_Orangepih3_Ubuntu_trusty_3.4.110_desktop.zip 
    Archive:  Armbian_5.05_Orangepih3_Ubuntu_trusty_3.4.110_desktop.zip
      inflating: Armbian_5.05_Orangepih3_Ubuntu_trusty_3.4.110_desktop.raw  
      inflating: armbian.txt             
      inflating: imagewriter.exe         
    
    real	0m16.519s
    user	0m15.028s
    sys	0m1.359s

(7za needs ~22 seconds for the ZIP and OS X' internal _ditto_ command needs just 8 seconds uncompressing the same ZIP but I would suspect the majority of users will use Windows so the latter is irrelevant). So it might take over twice as long decompressing .7z compared to .zip. But I would suspect that's wort the time since downloading another 200MB in just 21.5 seconds will not be that easy for most of our users.

I didn't send a commit since this topic might be discussed already and there might be also major concers providing .7z (and to be honest I almost never touch Windows so I've no idea what's best there). Since this might mostly be of interest for Igor's build host providing the official images I simply leave it up to him to decide but would be interested why refraining in case we're staying with ZIP. :)

zador-blood-stained, 2016-03-05:
I though about this, one of my concerns is outdated 7-zip in Ubuntu (even in Xenial), based on 7-zip 9.20.
No native extraction tool on Windows is an issue too (and providing SFX archives would either duplicate amount of images on server or make life on non-Windows systems harder).

ThomasKaiser, 2016-03-05:
I just got a mail (scary, we get a lot of attention these days and people seem to monitor our github issues as well) from someone recommending to provide the image type based on the 'user agent' in question. They would do this since a while and save a lot of ressources. Should be possible with all modern web servers. Hmm... too advanced stuff for me (now)

zador-blood-stained, 2016-03-05:
user-agent won't tell the server if you have any means of extracting .7z in your OS. If Igor doesn't mind keeping and maintaining 2 versions of each image, you can just show 2 download links (well, maybe 3rd link to .torrent sometimes in future) like
`Armbian_5.05_Cubietruck_Debian_jessie_4.4.4` [[306MB .zip]]() [[184MB .7z]]()
^ real results of packing 1.21GB image, tested with 7-zip 15.14, LZMA2, 64MB dictionary.

igorpecovnik, 2016-03-06:
.7z for windows users .gz for *nix ?

ThomasKaiser, 2016-03-06:
I would adopt @zador-blood-stained 's idea and provide .zip and .7z (and take care that 7-zip version on your build host remains at 9.20 for compatibility reasons... (BTW: I would like to stay with ZIP since on OS X it will be decompressed way faster -- maybe OpenCL/GPGPU acceleration is used there since it's more than twice as fast as unzip and top shows just 25% CPU utilisation)

igorpecovnik, 2016-03-06:
I also prefer zip, even might not be perfect ... we need to use a multi threaded zip creation with pigz: 
> pigz -K -k archive.zip image.raw 

My hosting has limit on space usage  - ssd storage ;) so .7z realisation must wait until I made some cleaning.

ThomasKaiser, 2016-03-22:
Thx for adding that. When I read through the commit I immediately thought about allowing $SEVENZIP to also contain a number between 1 and 9 to be used for `-mx=$SEVENZIP`

But since the main problem is that 7-zip runs single threaded per file further tweaking options would only make some sense if we manage to parallelize OS image creation (one image for all A20 boards created on btrfs, then one snapshot per board, 'image personalisation' on top, closing all images and then starting one 7za run per image).

BTW: Depending on whether you installed 7-zip or 7-zip-full package the available binary is either called 7za or 7zr. Don't know how of a problem that will be for users...

closed 2016-03-22 by ThomasKaiser

zador-blood-stained, 2016-03-22:
@ThomasKaiser 

> But since the main problem is that 7-zip runs single threaded per file

Hm, does it? From 7-zip help:

> LZMA2 | Default compression method of 7z format. LZMA2 is LZMA-based compression method. It provides better multithreading support than LZMA. But compression ratio can be worse in some cases. For best compression ratio with LZMA2 use 1 or 2 CPU threads. If you use LZMA2 with more than 2 threads, 7-zip splits data to chunks and compresses these chunks independently (2 threads per each chunk).

ThomasKaiser, 2016-03-22:
Ok, I was judging more or less by (no) fan noise. At least not all cores are used and I will only take a look into it when it's about improving build parallelism sometimes in the future. :)

igorpecovnik, 2016-03-23:
If this is will be primary used for automated build, than we set for maximum compression and voila, now is somewhere in between. Anyway, feel free to further tune up the parameters.

ThomasKaiser, 2016-03-30:
@zador-blood-stained Today I had a closer look and you were right: Compression happened in multithreaded mode and CPU utilization was between 400% and (most of the times) 1000% so most/all cores are used while compressing :)