armbian/build #210: Increase usability of web site

ThomasKaiser, 2016-03-05:
My last one for today (and if everything goes well the next 2 weeks also ;) )

Based on feedback from forum users and a nice young lady (art director for a living and just awarded for her work by german ADC):

We should try to improve useablity of the website a bit. Menu bar titles as menu item is not intuitive. It should better look like this and the first menu item shouldn't be named like the title but better "Armbian basics" or something like that. You've to push users to read documentation (I know it's hard but it works)

[image]

The symbols to show the difference between CLI and GUI images are really nice. But adding 3 characters in brackets doesn't break anything and helps the sort of people that are focused on text only (woohoo, just got 15 minutes explanation about different types of humans and their perception of the world!)

[image]

Identify the most important link easily: I really doubt that much people click on the light bar where 'Questions?' is written. And people that click on the link to the most read forum thread ever are simply lost because they would have to read through hundreds of forum posts _now_ (most people don't even read 2 warning sentences so it would be smarter to highlight the FAQ contents instead)

[image]

WereCatf, 2016-03-05:
I agree, especially on the dropdown-menu-acts-like-a-link -- it makes no sense, it's not intuitive. Even I hadn't realized it was actually a clickable link before you mentioned this!

Nitpick: "information" is an arbitrary amount, it cannot be counted, so there is no plural for it and you can't spell it as "informations." :)

igorpecovnik, 2016-03-06:
Updated.

ThomasKaiser, 2016-03-06:
Yes, that's a great improvement! Thx!

closed 2016-03-06 by ThomasKaiser