armbian/build #214: Replace wicd with something else in desktop-images

WereCatf, 2016-03-06:
That wicd is, quite frankly, downright terrifying in how clunky and un-user-friendly it is. Especially worrying is that it doesn't even try to detect the network-interfaces in the system, the user has to go into preferences and change them manually if whatever USB WiFi-dongle they are using doesn't use wext or if the WiFi-dongle isn't on wlan0 -- my mt7601u, for example, is on ra0. In addition, wicd does not ask for passkeys when trying to connect to protected networks, you have go into settings and find out where to put the key.

Oh, and connecting to protected networks doesn't even seem to work with wicd -- It just complains about wrong password. Connecting manually with wpa_supplicant, for example, works, but the whole point with a desktop-image is to be a bit more user-friendlier and allow for such things without having to drop to CLI.

igorpecovnik, 2016-03-06:
I am afraid Network Manager, as first alternative, is even worse.

Regarding password. I don't think that's a problem of wicd ... I have made several tests on WPA2 network, 2.4 and 5G ... works like charm. I think it's better to fix a workaround for device naming.

zador-blood-stained, 2016-03-06:
> I think it's better to fix a workaround for device naming.

udev rules allow renaming network interfaces, but I don't know if it's worth to automate it.

WereCatf, 2016-03-06:
I still think it'd be better to use something that doesn't require renaming interfaces at all. As for the password: if the problem isn't with wicd then I don't know where the problem is, I just know it works from nmcli and with wpa_supplicant, but not with wicd.

igorpecovnik, 2016-03-06:
Then let's find out where the problems are. You are using Jessie, right?

WereCatf, 2016-03-06:
No, I was testing out desktop-images. I've been receiving complaints about them, so I took the latest images we have on our website out for a ride. A lot of things seem broken with it compared to the Jessie - images I've built myself, like e.g. if I install lightdm it won't start -- permissions are wrong somewhere, I would assume, whereas it'd work fine under Jessie.

WereCatf, 2016-03-06:
Yup, works in Jessie. That means our Ubuntu - builds are fucked.

igorpecovnik, 2016-03-06:
So we build Jessie CLI and Jessie desktop in the future on all boards with desktop potential ;)

WereCatf, 2016-03-06:
That, or we move to testing Ubuntu 16.04 already. There's little point in fixing 14.04 at this point, IMHO.

igorpecovnik, 2016-03-06:
I don't mind but we need to prepare 16.04 and start do some testing while we can put out Jessie right away.

WereCatf, 2016-03-06:
True enough. I still retain my opinion that wicd is atrocious, though, even if the problem with encrypted passwords doesn't apply to Jessie.

ThomasKaiser, 2016-03-07:
@WereCatf did you already gave wifi-radar a try?

WereCatf, 2016-03-07:
@ThomasKaiser Nope

ThomasKaiser, 2016-03-07:
I [added wifi-radar](https://github.com/igorpecovnik/lib/commit/e518304e5a4141706f9c0bbcf096ae590487991c) to the list of desktop packages since it seems to work better than wicd. But due to lack of hardware (the only WiFi dongle here seems to be broken) I can not really test. At least wifi-radar scans automagically for the name of WiFi devices and seems to be highly customisable so in case adjustments are necessary we could ship with a default _/etc/wifi-radar/wifi-radar.conf_

zador-blood-stained, 2016-09-10:
Closing this since new images will have NetworkManager by default.

closed 2016-09-10 by zador-blood-stained