armbian/build #216: motd showing values of historical interest

ThomasKaiser, 2016-03-07:
Something I also discovered but didn't looked into yet:

    System load:   0.99            	Up time:       45 sec		
    Memory usage:  14 % of 1001Mb 	IP:            192.168.83.118
    CPU temp:      30°C           	
    Usage of /:    47% of 15G    	

vs.

    tk@orangepipc:~$ uptime
     12:16:20 up 18:13,  1 user,  load average: 0.09, 0.09, 0.09

A suggestion to fix/improve this has already been made by [forum user Baos](http://forum.armbian.com/index.php/topic/791-armbian-ssh-login-motd/)

igorpecovnik, 2016-03-07:
Yes. I am trying to recall why I created my own function for this. I only comment that "we need this" but forgot to write a reason. One is clear, it's not working properly.

ThomasKaiser, 2016-03-22:
Sorry, I neither wanted to refer to the format of uptime (-p doesn't work since it's output can get rather lengthy so your function should be preferred) nor Baos but [Vlad's fix for the generation of motd](http://forum.armbian.com/index.php/topic/791-armbian-ssh-login-motd/?p=6013) instead.

This should be fixed.

And I also wondered whether the display of packages to be updated can't be split in one daemon like approach that runs earlier at login (so the information will already be present when the login occurs and it doesn't delay the login any more) and the motd mechanism simply reading out `/var/cache/apt/archives/updates.number` later. But this would also require the the script that creates the contents of `/var/cache/apt/archives/updates.number` will be run on a daily or hourly base.

What about splitting [40-updates](https://github.com/igorpecovnik/lib/blob/master/scripts/update-motd.d/40-updates) into two parts. The data gathering part storing as `/etc/cron.daily/get-updates` and reducing `/etc/update-motd.d/40-updates ` to the last few lines and calling `/etc/cron.daily/get-updates &` from within `/etc/init.d/armhwinfo`?

igorpecovnik, 2016-03-22:
This vlad fix disables caching if I understand properly ... and that's not what we want on our slow boards. 40-updates we need leave it as is with one exception - we run update if no update.number exists, so if cron weren't started yet ...

ThomasKaiser, 2016-03-22:
> his vlad fix disables caching if I understand properly ... and that's not what we want on our slow boards. 

Not really. With your or his version the scripts get called everytime a user logs in. With your version the user gets the values displayed from his last successful login, with Vlad's from the current. Displaying historical data isn't what this motd approach was intended for, right? ;)

igorpecovnik, 2016-03-22:
Strange. I don't have this line, neither in Jessie nor Wheezy:
session    optional   pam_motd.so  motd=/run/motd.dynamic noupdate

ThomasKaiser, 2016-03-22:
On Jessie I have

    session optional pam_motd.so motd=/run/motd.dynamic
    session optional pam_motd.so noupdate

but if I uncomment it breaks completely. And after reverting everything back to Armbian defaults it now works always. So we still need to find the root cause for this behaviour it seems :\

closed 2016-06-12 by ThomasKaiser