armbian/build #220: linux-headers with scripts

gklipel, 2016-03-09:
Hi!

I'm trying to compile xtables modules but not seems to be working:

`/usr/src/modules/xtables-addons# make
make  all-recursive
make[1]: Entering directory '/usr/src/modules/xtables-addons'
Making all in extensions
make[2]: Entering directory '/usr/src/modules/xtables-addons/extensions'
Xtables-addons 2.6 - Linux 3.4.110
if [ -n "/lib/modules/3.4.110-sun7i/build" ]; then make -C /lib/modules/3.4.110-sun7i/build M=/usr/src/modules/xtables-addons/extensions modules; fi;
make[3]: Entering directory '/usr/src/linux-headers-3.4.110-sun7i'
  CC [M]  /usr/src/modules/xtables-addons/extensions/compat_xtables.o
/bin/sh: 1: scripts/genksyms/genksyms: not found
scripts/Makefile.build:313: recipe for target '/usr/src/modules/xtables-addons/extensions/compat_xtables.o' failed
make[4]: *** [/usr/src/modules/xtables-addons/extensions/compat_xtables.o] Error 127
Makefile:1367: recipe for target '_module_/usr/src/modules/xtables-addons/extensions' failed
make[3]: *** [_module_/usr/src/modules/xtables-addons/extensions] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-3.4.110-sun7i'
Makefile:448: recipe for target 'modules' failed
make[2]: *** [modules] Error 2
make[2]: Leaving directory '/usr/src/modules/xtables-addons/extensions'
Makefile:485: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/usr/src/modules/xtables-addons'
Makefile:369: recipe for target 'all' failed
make: *** [all] Error 2`

Error on make mrproper:

`# make mrproper
/usr/src/linux-headers-3.4.110-sun7i/drivers/gpu/mali/mali/Makefile:32: *** "KDIR or CPU must be specified.".  Stop.
scripts/Makefile.clean:94: recipe for target 'drivers/gpu/mali/mali' failed
make[3]: *** [drivers/gpu/mali/mali] Error 2
scripts/Makefile.clean:94: recipe for target 'drivers/gpu/mali' failed
make[2]: *** [drivers/gpu/mali] Error 2
scripts/Makefile.clean:94: recipe for target 'drivers/gpu' failed
make[1]: *** [drivers/gpu] Error 2
Makefile:1182: recipe for target '_clean_drivers' failed
make: *** [_clean_drivers] Error 2`

More yet, error on make prepare scripts:

`# make prepare scripts
   HOSTCC  scripts/basic/fixdep
   HOSTCC  scripts/kconfig/conf.o
   SHIPPED scripts/kconfig/zconf.tab.c
   SHIPPED scripts/kconfig/zconf.lex.c
   SHIPPED scripts/kconfig/zconf.hash.c
   HOSTCC  scripts/kconfig/zconf.tab.o
   HOSTLD  scripts/kconfig/conf
 scripts/kconfig/conf --silentoldconfig Kconfig
 drivers/net/wireless/ap6210/Kconfig:42:warning: defaults for choice values not supported
   CHK     include/linux/version.h
   CHK     include/generated/utsrelease.h
   UPD     include/generated/utsrelease.h
 make[1]: *** No rule to make target 'arch/arm/tools/gen-mach-types', needed by 'include/generated/mach-types.h'.  Stop.
 /usr/src/linux-headers-3.4.110-sun7i/arch/arm/Makefile:282: recipe for target 'archprepare' failed
 make: *** [archprepare] Error 2`

So, scripts on kernel headers are not present. Can you please provide an linux-headers-sun7i with this as an update or a guide on how achive it?

Thanks in advance!

igorpecovnik, 2016-03-09:
Kernel headers are working fine,  I compiled few drivers with them. Here is a make scripts output:

```
root@orangepiplus:/usr/src/linux-headers-3.4.110-sun8i# make scripts
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --silentoldconfig Kconfig
  HOSTCC  scripts/genksyms/genksyms.o
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/keywords.hash.c
  SHIPPED scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/lex.lex.o
  SHIPPED scripts/genksyms/parse.tab.c
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTLD  scripts/genksyms/genksyms
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTLD  scripts/mod/modpost
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/conmakehash
  HOSTCC  scripts/bin2c
  HOSTCC  scripts/recordmcount
```

WereCatf, 2016-03-09:
`make prepare scripts` fails at one point for me too, but just simply doing `make scripts` instead works fine on sun8i-headers.

gklipel, 2016-03-10:
Thanks Igor, and in time, your work is wonderful!

igorpecovnik, 2016-03-10:
Oh, just my two cents. Thanks!

closed 2016-03-10 by igorpecovnik