armbian/build #221: Set default compiler settings when creating first user account?

ThomasKaiser, 2016-03-09:
Again not an issue but more a request for comments.

I recently did some benchmarking just to discover how moronic/broken the most used is. Part of my build customisation is to set CFLAGS/CXXFLAGS automatically (I use an adapted /etc/skel/ and .profile therein) and therefore my benchmarks scores explode automagically when running the Phoronix test suite: http://openbenchmarking.org/result/1603099-GA-1603083GA01&obr_sor=y&obr_hgv=Orange+Pi+PC+Armbian

As a matter of fact choosing Armbian instead of a manufacturer provided OS image already improves performance on most boards that use a modern SoC that is a throttling candidate. That's simply because the linux-sunxi community and we spend an insane amount of time in optimising thermal/throttling/dvfs settings and come up with defaults that are more reasonable than the stuff board vendors provide. An extreme example is the link above. Orange Pi PC/Plus when tested by Michael show bad scores not only since the whole Phoronix testing methodology is questionable (not taking throttling into account) but most importantly caused by [these 5 lines](https://github.com/linux-sunxi/sunxi-boards/blob/88b268e45e43ba05515aba142b74a55191053759/sys_config/h3/xunlong_orange_pi_pc.fex#L287-L291) leading to killed CPU cores in case the thermal tresholds were reached.

So when we already improve 'real world performance' (not only by sane dvfs/thermal defaults but also by distributing IRQs accross CPU cores and many tweaks here and there) why not improve also irrelevant benchmark scores when running Armbian?

Please compare the last three listed results here: http://openbenchmarking.org/result/1603090-GA-1603090GA57

This is someone running the Phoronix benchmarks on a BPi M3 using SinoVoip's Raspbian (ARMv6 userland, old GCC with unknown settings used, horrible thermal trip points defined that throttle way too early) vs. my Armbian test image for A83T I thought I threw away (still present on the eMMC, I booted accidentally into it). First run with these settings (not active for the 'John The Ripper' bench and responsible for worse results with some tests!):

    export CXXFLAGS="-O3 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -ftree-vectorize"
    export CFLAGS="-O3 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -ftree-vectorize"
    export EXTRAOPTFLAGS="-O3 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -ftree-vectorize"

3rd run without these variables exported (only sane cooler state and heatsink + fan). It's obvious that Michael's test methodology pretty much sucks (some tests choose good compiler flags suitable for the board others not) and that throttling is a very important factor (I will add another run of tests when I have physical access to the BPi M3 to disable the fan and also thought about enabling 2.2 GHz _scaling_max_cpufreq_ in the kernel sources to pimp the Phoronix stuff that runs single-threaded -- close to unbelievable but he lets some tests run only on a single core. This is braindead and also typical for the whole test suite since throttling or strategies like Turbo Boost are ignored completely)

If Armbian would export CFLAGS/CFLAGS/EXTRAOPTFLAGS="-O3" from the user's profile then when someone starts to explore his board with Armbian and submits test results to openbenchmarking.org they're always way better than those made with manufacturer provided OS images. It's a bit of manipulation but nothing compared to the manipulation caused by Phoronix using all these meaningless numbers as a proof of hardware performance differences. But I'm not skilled enough to decide whether we would do any harm when setting these defaults.

I would suspect that real devs define CFLAGS/CFLAGS/EXTRAOPTFLAGS always on their own. So the only side effect would be that hardware benchmarked running Armbian looks way faster and that might put some pressure on Phoronix to overthink their questionable testing methodology.

Any thoughts?

WereCatf, 2016-03-09:
I'm not sure I understand you: you're asking to define some default CFLAGS for... getting better results in a benchmark? No actual benefit to the user or anything, just bragging-rights on a benchmark?

WereCatf, 2016-03-10:
Ps. -O3 already enables auto-vectorization, -ftree-vectorize is redundant.

ThomasKaiser, 2016-03-10:
Since Armbian contains everything needed to compile something it should help novices also to get faster binaries? What are the potential drawbacks of setting -O3 as default?

And 'no actual benefit'... people that trust in numbers get better results and have better feelings. This can drive whole communities if you've a look at openbenchmarking.org for example ;)

WereCatf, 2016-03-10:
Novices are unlikely to do anything that needs such drastic optimizations, and for novices compatibility is more beneficial; compiling with those options would prevent the novice from trying that binary out on other devices that may lack full vfpv4 or neon, for example, or from sharing those binaries with their buddies whose devices may lack those. It's better to learn about and use CFLAGS when one is ready for that.

-O3 provides little benefit over -O2, especially if you use -O2 -ftree-vectorize, but drastically increases code-size. In general it is not worth using -O3, it should only be used for special-cases.

> And 'no actual benefit'... people that trust in numbers get better results and have better feelings.

The benchmark is broken and ugly, and if the benchmark doesn't already enforce such things then that just goes to show even better exactly how broken it is. If there are results compiled without any special CFLAGS and results compiled with tuned CFLAGS all mixed up, without the CFLAGS themselves being defined as part of the comparison-criteria, that makes **all those results** suspect and any comparisons invalid -- the comparisons would only be valid if all the results were compiled with similar CFLAGS!

And no, that is no actual benefit; "better results" is still just about bragging-rights in this case and "better feelings" is a result of egotism and ignorance.

ThomasKaiser, 2016-03-10:
Ok, agreed. Then we leave it as it is now and Armbian will still look better when people submit results since we ship with sane throttling/thermal settings by default.

Hmm... still thinking about 

    alias phoronix-test-suite="CLFLAGS =\"-O3\" /usr/bin/phoronix-test-suite \"$@\""

since this will produce drastically better looking numbers without meaning ;)

Just kidding, we stop here and I will ad some documentation later. BTW: The first link above where Orange Pi PC running with Armbian is twice as fast as when Michael measured was without any tweaked CFLAGS. Simply other cooler state table settings, 1296MHz enabled, heatsink and fan.

closed 2016-03-10 by ThomasKaiser