armbian/build #222: 2560x1080 screen resolution support

bandesznet, 2016-03-10:
It will be possible to set 2560x1080 screen resolution in the future? (H3 boards)

Tomvictor, 2016-07-04:
H3 processor supports 4K resolution on HDMI, But no way to power up 2K ( 1440x2560 ) display. Is that possible?

shahidali55, 2016-07-15:
Is it possible to add a new mode for 1024x600 60Hz ? [This display](http://www.waveshare.com/wiki/7inch_HDMI_LCD_%28C%29) works fine with the Raspberry Pi and I really wish I can use it with a Orange Pi One

shahidali55, 2016-07-18:
The [waveshare display](http://www.waveshare.com/wiki/7inch_HDMI_LCD_%28C%29) seems to be working with "h3disp -m 720p60 -d" on the Orange Pi One running Armbian 5.14 with legacy kernel. It is just that the image is too large for the screen and the edges get chopped up. Can someone please guide me with patching the kernel to get a custom mode 1024x600 60Hz over DVI ? I have setup the build environment following [these steps](http://docs.armbian.com/Developer-Guide_Build-Preparation/) and was successfully able to build Armbian 5.17 for Orange Pi One with legacy kernel

ThomasKaiser, 2016-07-18:
We do not support currently any _custom_ HDMI modes with legacy sun8i kernel (unlike sun4i/sun5i/sun7i where HDMI code is better and [custom resolution can be configured using fbset](http://forum.lemaker.org/thread-15430-1-1.html). There exist some patches to add additional modes already (replacing existing ones which is not that good) so a good starting point would be to look at them (see [post #2 here](http://forum.armbian.com/index.php/topic/881-prepare-v51-v201604/?p=6777), then figure out how parameters have to be adjusted for your 1024x600 display and get back to us with a patch.

BTW: The whole idea to file github issues for feature requests is somewhat pointless. If I were you and interested in custom HDMI resolutions I would start a new thread in H3 section in Armbian forum, collect all available information and ask for help (in the hope that knowledgeable people might jump in and provide help with patching or settings)

blackandwhitetux, 2016-08-11:
shahidali55, 

I have the same screen as you. I am working on the same issue - Orange Pi with custom 1024 x 600 resolution.  I will let you know if I get it working and upload the kernel hopefully.

zador-blood-stained, 2016-11-30:
Custom HDMI modes will be supported with mainline kernel and simplefb, and with DRM driver too (eventually). If these changes are backported to the legacy kernel it will work too. Progress should be tracked in #443

closed 2016-11-30 by zador-blood-stained