armbian/build #224: Race condition between firstrun and check_first_login.sh

zador-blood-stained, 2016-03-12:
OS release is Xenial Desktop.

If you were able to log in with "root/1234" and `firstrun` was started before you entered new user name, this happens (followed by reboot):
```
Creating new account. Please provide a username (eg. your forename): zador
Adding user `zador' ...
Adding new group `zador' (1000) ...
Adding new user `zador' (1000) with group `zador' ...
Creating home directory `/home/zador' ...
Copying files from `/etc/skel' ...
Enter new UNIX password:
Retype new UNIX password:
passwd: password updated successfully
You are required to change your password immediately (root enforced)
chfn: PAM: Authentication token is no longer valid; new one required
adduser: `/usr/bin/chfn zador' returned error code 1. Exiting.
```
because `.not_logged_in` wasn't removed, second login with root account will again prompt for new user creation, but using same name will fail and result in reboot because user **was** created, but probably wasn't added to groups, nodm wasn't enabled and so on.

So do we change root password in check_first_login.sh after successfull creation of "normal" user or do we force root password change only after reboot?

ThomasKaiser, 2016-03-12:
Hmm... I think I understand now. The timing is somewhat different when using Xenial and therefore this could happen?

What about making the 1st reboot mandatory again and trying to prevent any login at this stage (setting the root passwd from within firstrun)?

We now have a couple of users that lack both patience and reading skills so they will try to create all sorts of trouble in situations where they just should wait a few minutes to let all the magic happen...

zador-blood-stained, 2016-03-12:
Timings may be off because I'm testing with FEL/NFS, and problem may have appeared due to different PAM features in new release, plus force user creation feature is relatively new.

Mandatory first reboot could solve the problem, but it needs more red text messages like "**Please wait and don't do anything, system will be rebooted automatically**", and even this won't help much when you have users with poor reading skills.

Another solution - changing check_first_login script logic, so if "normal" user was created successfully, run `chage -d 0 root` or `passwd`, or at least disable SSH root login to improve security in case user aborts password change or never logs in as root again.

igorpecovnik, 2016-03-12:
or disable user creation if reboot is ongoing?

zador-blood-stained, 2016-03-12:
> or disable user creation if reboot is ongoing?

it's not the problem here, problem is that firstrun script "expires" root password, after that any already **existing** root sessions won't be able to perform actions that require PAM authentication.

zador-blood-stained, 2016-03-12:
Another possible solution that I forgot about - building images with root password already expired (set in chroot from build script), so you can't login as root without changing password first.

igorpecovnik, 2016-03-12:
OK, but the problem is actually here to. If the user creation is broken by reboot when user is create and the process not finished your user will be created and you will be force to create it again, but you won't be able to use the same name ... it can happen.

Yes, let's fix expired root at build process.

zador-blood-stained, 2016-03-12:
> Yes, let's fix expired root at build process.

OK, I'll run a few tests to make sure that it works like intended (probably tomorrow) and push necessary changes.

ThomasKaiser, 2016-03-12:
> I'll run a few tests to make sure that it works like intended (probably tomorrow) and push necessary changes.

Please be aware that I just recently sent a PR to better deal with Banana Pi that affects armhwinfo and firstrun (trying to avoid merge conflicts)

WereCatf, 2016-03-13:
Why not simply delay check_first_login.sh until firstrun has been executed and the board rebooted? Disabling login altogether is a bad idea, just add a check to check_first_login.sh so that if firstrun hasn't completed and board rebooted it will only spill a warning, like, say, "Initial preparations and the subsequent automatic reboot haven't yet been completed, you are recommended to be patient and wait until an automatic reboot has been completed (may take several minutes.)" When the firstrun-reboot has finished then allow check_first_login.sh to do the usual stuff.

Doing it like that should fix all of the mentioned issues.

igorpecovnik, 2016-03-13:
Just two remarks:
- on newer kernels expand root fs works on the fly therefore we necessarily don't need a second boot. 
- If first boot results into read only fs ... what then.

WereCatf, 2016-03-13:
Well, why not just move `chage -d 0 root` above `check_prerequisits` in firstrun, and, if running on an older kernel that needs reboot, make check_first_login.sh spit out an error if /tmp/firstrun_running exists and log back out? I did take a look and I cannot see any reason for why chage couldn't just be moved up.

zador-blood-stained, 2016-03-13:
`chage` was moved to build time, so first part of the issue should be solved.

zador-blood-stained, 2016-03-22:
What about disabling updating packages list on first boot (and leaving it up to `unattended-upgrades` or manual update by user). This will speed up `firstrun` script, so if reboot is needed, it will happen without long delay.

> If first boot results into read only fs ... what then.

We can try creating a file (i.e. in /root), if it failed - display a message, abort auto reboot and maybe change LED triggers on some boards. Since filesystem is read-only, you can't do much without trying to remount it to r/w (i.e. you could ask and upload `dmesg` output to sprunge.us, but user probably won't be able to log in and change root password to do it).

igorpecovnik, 2016-03-22:
Then we won't have this fancy "xx - updates are waiting to install" at first boot :) ... reboot is needed only on legacy kernels and on wheezy images. I guess most people will go for mainline Jessie anyway, so we might don't need to bother with this?

zador-blood-stained, 2016-03-22:
> Then we won't have this fancy "xx - updates are waiting to install" at first boot :)

If we execute `apt-get -y -qq upgrade` somewhere before packing rootfs image, then more or less fresh images won't have any updates to install at first boot anyway :smile: 

> If we disable update - do we have an package list from build process? we don't want error from apt-get install xxxx

Of course, but even if some packages were updated `apt-get install` should print nice error message like "try updating package list"

igorpecovnik, 2016-03-22:
OK, then let's disable update at first boot.

zador-blood-stained, 2016-03-29:
I believe original problem is solved now, so this can be closed.

closed 2016-03-29 by zador-blood-stained