armbian/build #228: Pimping H3 support for mainline _now_?

ThomasKaiser, 2016-03-15:
I would suspect it's just matter of days until Ethernet works _really_ with mainline: http://irclog.whitequark.org/linux-sunxi/2016-03-15#15783663;

I tried it out and needed an additional 5th [0000-sun8i-h3.patch](http://sprunge.us/QFXd) and the 4 from [here](http://sunxi.montjoie.ovh/ethernet/)

Then added 

    CONFIG_SUN4I_EMAC=y
    CONFIG_SUN8I_EMAC=y

to config/linux-sunxi-next.config, set KERNELBRANCH="v4.5" in userpatches/lib.config, disabled all other mainline H3 patches and gave it a try: Nope, no network. Since this was such a frustrating day I simply powered every board off.

Montjoie has collected/created quite some useful stuff for H3 here: http://moinejf.free.fr/opi2/ (HDMI, Audio, WiFi for mainline). And we already have patches for SPI/I2C from martinayotte :)

Worth to pick it up now being the 1st with a mainline distro for H3 quite useable?

(I quit here for some time and try to focus on vacation again :) )

WereCatf, 2016-03-16:
My Orange Pi died a few days ago, so I can't test anything, but how's the state of mainline: can ethernet be made to work, and how about HDMI framebuffer?

ThomasKaiser, 2016-03-16:
At least patches apply properly when [creating one additional patch against 4.5](http://forum.armbian.com/index.php/topic/617-wip-orange-pi-one-support-for-the-upcoming-orange-pi-one/page-12) but still no network (but [wens starts to look into and maybe that improves speed a lot](http://irclog.whitequark.org/linux-sunxi/2016-03-16))

Regarding HDMI please see the link above. I've not tested anything, just a quick try to get Ethernet running yesterday.

ThomasKaiser, 2016-03-20:
JFTR: Hans de Goede submitted a new/consolidated set of patches: https://groups.google.com/forum/#!topic/linux-sunxi/EIeU32D5kpM (even a mainline Wi-Fi driver referenced)

igorpecovnik, 2016-03-21:
OK

https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sunxi-dev/0005-ethernet-add-sun8i-emac-driver.patch

We got working EMAC with this patch?

ThomasKaiser, 2016-03-21:
> We got working EMAC with this patch?

Not really, it works somehow according to yesterday's forum posts (for me not at all, maybe due to GbE switches here) but at least his patches apply against a clean 4.5 when adding the necessary [sun8i-h3.dtsi stuff](https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sunxi-dev/0000-sun8i-h3.patch) before.

Since a few others are now also looking into EMAC (apritzel/wens) maybe we get working Ethernet with 4.6. And based on Hans new/refactored patches this seems like something we could rebase the whole patches on?

faddat, 2016-04-03:
Testing 4.6-rc1 now....

zador-blood-stained, 2016-04-18:
Since current H3 patches for `sunxi-dev` break compilation of `dev` branch for non-H3 devices, maybe it's better to separate H3 branch into `sun8i-dev` LINUXFAMILY?

igorpecovnik, 2016-04-18:
Yes, that would be the best way i guess.

closed 2016-06-12 by ThomasKaiser