armbian/build #239: Banana Pi M1 doesn't boot

Hiradur, 2016-03-22:
#### Setup:
- Ubuntu 14.04
- lib git commit 8f651897094a423fb62ba2740e20b363a90129d9
- Banana Pi M1
- options chosen in compile.sh:
  - bananapipro system
  - "next" kernel
  - no GUI
  - Debian Jessie
  - otherwise untouched sources/scripts

#### Problem:
After writing the image to the SD card with `sudo dd if=Armbian_5.06_Bananapipro_Debian_jessie_4.4.6.raw of=/dev/sdb bs=1M`, putting it into the Banana Pi M1 and powering the Pi the red LED lights up but nothing else happens. No output to the display connected via HDMI, no reaction to keyboard input, not showing up in the network, no other LEDs lighting up.

I think it's not the SD card. I checked it with some tools, tried another card and the official image from Armbian.com runs just fine.

This is my first time using the Armbian build tools, did I miss something?

ThomasKaiser, 2016-03-22:
How does 'gcc --version' looks like?

igorpecovnik, 2016-03-22:
I can confirm it's working - host = Ubuntu 14.04, GCC is default 4.8.4 ... boots fine, there is permanent red light and green is flashing on SD activity.

igorpecovnik, 2016-03-22:
I use those extra switches (userpatches/lib.confgih) but also without them it should work.
EXTENDED_DEBOOTSTRAP="yes"
FORCE_USE_RAMDISK="yes"
PROGRESS_DISPLAY="none"
GPG_PASS="xxxxxxxxxxxxxxxxxxxx"

ThomasKaiser, 2016-03-22:
@igorpecovnik did you try it today? Since I just let a fel-boot build run and this one stops here:

    U-Boot SPL 2016.03-armbian (Mar 22 2016 - 16:39:05)
    DRAM: 1024 MiB
    CPU: 912000000Hz, AXI/AHB/APB: 3/2/2
    Trying to boot from 

The very same run worked a few days ago.

Hiradur, 2016-03-22:
> How does 'gcc --version' looks like?

``` 
~$ gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.1) 4.8.4
```

igorpecovnik, 2016-03-22:
@ThomasKaiser

This actually happen to me the other day but I changed to another SD card and it was booting. Perhaps something with u-boot environmental ? We are not cleaning it on all boards ...

Hiradur, 2016-03-22:
@ThomasKaiser Do you see this on the screen? Because I don't even see a screen from U-Boot.

igorpecovnik, 2016-03-22:
Allwinner displays Uboot messages to HDMI and serial ... currently there is a problem / bug if you attach 4K monitor, than you will have no screen. If you got nothing, than u-boot was not written properly or there is some problem reading it.

zador-blood-stained, 2016-03-22:
@Hiradur you always see a message like this from SPL when loading u-boot via FEL
If you have any kind of USB UART adapter (preferrably 3.3v TTL) you can check if u-boot outputs anything on serial console), otherwise it's possible that configuration for your board is broken, there are some problems with different boards in this u-boot release (2016.03).

Hiradur, 2016-03-22:
I have the BPi connected to a 1920x1200 screen. Unfortunately I don't have the equipment to read the serial console but I'll see if I can find something at my workplace.

ThomasKaiser, 2016-03-22:
I tried it with 2016.03-rc2 before just to notice that booting didn't work again. But since one u-boot patch failed this might be the expected result. Can't/won't look into it further since I'm away from the build host now.

igorpecovnik, 2016-03-23:
Yes, there was a problem, but I can't recreate this error. 

Try setting older u-boot manually - add BOOTBRANCH="v2016.01" to userpatches/lib.config prior to compilation start. This will force use older u-boot.

Hiradur, 2016-03-23:
> Try setting older u-boot manually - add BOOTBRANCH="v2016.01" to userpatches/lib.config prior to compilation start. This will force use older u-boot.

With this setting the Banana Pi M1 boots fine. So it appears to be a problem with U-Boot 2016.3. Thanks for your help!

ThomasKaiser, 2016-03-23:
Shouldn't we think about defining 'well known settings' (last u-boot commit that is known to work and such things) to prevent such surprises? It happened a few times in the last few weeks that I wasted hours since checked out versions from Github broke compilation (the usual 'yesterday it worked' experience)

zador-blood-stained, 2016-03-23:
> Shouldn't we think about defining 'well known settings'

yes, but it should be optional to allow testing new releases easily without editing any files controlled by git.

ThomasKaiser, 2016-03-23:
What about 'BLEEDING_EDGE=yes' in `userpatches/lib.config` or something like that? This way the core devs can enjoy surprises every other day but normal users are able to build Armbian from sources without them and us wasting time analysing 'just another u-boot bug'

Hiradur, 2016-03-23:
Actually, I just realized that my BPi sometimes doesn't detect SD cards and when I move them a bit it does. Maybe this was not a software problem but a problem with the BPi's SD card reader after all. :/

igorpecovnik, 2016-03-23:
It's also possible. In any case if we add some kind of "last known good commit" to the configuration we just made an added value to the stability of the whole process.

Hiradur, 2016-03-24:
I investigated a bit further over the last days. 

> Actually, I just realized that my BPi sometimes doesn't detect SD cards and when I move them a bit it does. Maybe this was not a software problem but a problem with the BPi's SD card reader after all. :/

This seems to be wrong. The BPi only very rarely does not detect the SD card. 
The real problem seems to be that the filesystem created by lib is broken. Immediately after writing the image to the SD card and mounting it I get this error:
```
Error mounting /dev/sdb1 at /media/hiradur/55adbf46-8610-40e2-8916-33bb60d73557: Command-line `mount -t "ext4" -o "uhelper=udisks2,nodev,nosuid" "/dev/sdb1" "/media/hiradur/55adbf46-8610-40e2-8916-33bb60d73557"' exited with non-zero exit status 32: mount: wrong fs type, bad option, bad superblock on /dev/sdb1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
.
```

``` dmesg | tail ``` outputs:
```
dmesg | tail
[ 6028.970866] sd 13:0:0:0: [sdb] 15523840 512-byte logical blocks: (7.94 GB/7.40 GiB)
[ 6028.971966] sd 13:0:0:0: [sdb] Write Protect is off
[ 6028.971974] sd 13:0:0:0: [sdb] Mode Sense: 03 00 00 00
[ 6028.973029] sd 13:0:0:0: [sdb] No Caching mode page found
[ 6028.973042] sd 13:0:0:0: [sdb] Assuming drive cache: write through
[ 6028.995421]  sdb: sdb1 sdb2
[ 6029.001028] sd 13:0:0:0: [sdb] Attached SCSI removable disk
[ 6250.675151]  sdb: sdb1
[ 6292.984442] EXT4-fs (sdb1): bad geometry: block count 302085 exceeds size of device (302080 blocks)
[ 6565.595546] EXT4-fs (sdb1): bad geometry: block count 302085 exceeds size of device (302080 blocks)
```

The weird thing is that sometimes I can mount the SD card without errors and sometimes the BPi boots from it but usually it doesn't. Now I thought the problem may be the SD Card or the SD Card reader in my laptop. However, I wrote several different images, including Armbian official, multiple times to different SD cards and they never gave me a single error. The BPi always boots from them.
The only image which gives me trouble is the one created by the Armbian build tools.

Did somebody else encounter something like this? Any ideas what could be wrong? I already edited the partition table manually. Then the laptop has no trouble mounting it but the BPi still refuses to boot.

zador-blood-stained, 2016-03-24:
This partition table problem happens sometimes (randomly), we can't reproduce it reliably yet. It's related to image size calculations or shrinking procedure.

Hiradur, 2016-03-24:
Thanks for the info. So the solution would be to run the build tools again until they created a working image?

zador-blood-stained, 2016-03-24:
I would try running build with additional options (if your SD card is 4GB or bigger)
```
EXTENDED_DEBOOTSTRAP=yes
FIXED_IMAGE_SIZE=4000
```

Hiradur, 2016-03-24:
I tried this. Even though the filesystem appears to be ok now the BPi still refuses to boot with the created image.

igorpecovnik, 2016-03-28:
https://github.com/OLIMEX/OLINUXINO/issues/31#issuecomment-202271245

It seems that many boards could have problems - shell we rather set u-boot to 2016.01 after all?

zador-blood-stained, 2016-03-28:
I think yes, especially if you are planning new release for sunxi boards soon.

igorpecovnik, 2016-03-28:
It's planned in few weeks but the problem is more if people use build script - they end up with problems.

zador-blood-stained, 2016-03-28:
Since 2016.03 doesn't bring new features for A10 and A20 boards, it's safe to set 2016.01 for sun[457]i, while sun8i needs 2016.03 if I understand correctly.

igorpecovnik, 2016-03-28:
yes, I left sun8i out of this.

ThomasKaiser, 2016-03-28:
I tried to boot a Banana Pro 2 days ago with a new build, first with defaults then defining 2016.01 to be used in `userpatches/lib.config` and it got stuck like explained before: `Trying to boot from ` (but I suspect more and more that something's wrong with my build host anyway). Maybe the LDO patch is the culprit?

As far as I understood the solution for poor GbE throughput with Lime2 is unrelated to the LDO patch (master/slave mode). Can't test since I shut my build host down since the machine now boots from a different disk and serves as a demo machine for a workflow demonstration the next days (maybe even weeks).

igorpecovnik, 2016-03-29:
I'll do the tests later today to make sure.

igorpecovnik, 2016-03-29:
Banana boots without a problem with 2016.01, both kernels

Hiradur, 2016-03-29:
> Banana boots without a problem with 2016.01, both kernels

Would you mind sharing your exact settings?

igorpecovnik, 2016-03-29:
No extra settings - just make sure you sue latest version of build script.

Hiradur, 2016-03-29:
Still not working for me. Without any custom patches or changes in configuration I get a corrupt filesystem every time I tried and when I force an image size of 4G the BPi doesn't boot. This is with a fresh clone of lib.

jernejsk, 2016-04-12:
I had similar experience with building OpenELEC images, especialy under VM (VirtualBox). Someone solved everything with putting "sync" command after every important step while building file system / image.

I hope I'm not off topic

zador-blood-stained, 2016-06-12:
Closing since there weren't more reports about corrupted filesystem in images

closed 2016-06-12 by zador-blood-stained

Hiradur, 2016-07-18:
Recently I created a few more images and all of them worked fine. I didn't change my setup (except for a newer version of the scripts) and I don't really know why it didn't work beforehand but I'm glad it's working now.