armbian/build #243: Modify partition resizing behaviour

ThomasKaiser, 2016-03-26:
I'm currently playing around with different SD cards, partitioning and filesystem (settings). Being able to create a bunch of additional partitions on a card is therefore quite useable and Armbian's auto resizing gets in my way. I know that I can disable it completely (`touch /root/.no_rootfs_resize` as part of the customization) but my goal is to implement something a bit more fine grained.

Two questions:

1) Is there a specific reason we still create a MBR partition table and not GPT?

2) Same applies to the use of `fdisk` for the partition resizing in `firstrun` (most probably `parted` being outdated as hell in Wheezy ;) )?

I though about switching to GPT and replacing the fdisk call with `parted /dev/mmcblk0 resizepart 1 yes $VALUE` instead. $VALUE defaults to `100%` and can be adjusted by creating `/root/.rootfs_resize` containing a string that will the be used by parted (eg. `50%`, `5000MiB` or `7577599s`). For compatibility reasons the existence of `/root/.no_rootfs_resize` still has higher priority.

And I also thought about decreasing the default `100%` to `95%` when the size of the card in question is 4GB or below. What might sound absurd has a simple background. Using 2016 cards with 4GB or less is most probably a sign of 'o look, there's an old spare card lying around. Let's try Armbian with this'. Older cards are known to implement less superiour wear leveling and GC algorithms so by leaving a 5% spare area we help the card's controller in situations where users start to fill their cards to the max.

ThomasKaiser, 2016-03-26:
Playing a bit around with it looks already pretty discouraging since it depends on the parted version used. I didn't get it to work with the `yes` argument but had to use an undocumented switch and redirecting stdin:

    root@pine64hot:~# cat <<EOF | parted ---pretend-input-tty /dev/mmcblk0
    > resizepart 4
    > yes
    > 74792959s
    > EOF
    GNU Parted 3.2
    Using /dev/mmcblk0
    Welcome to GNU Parted! Type 'help' to view a list of commands.
    (parted) resizepart 4                                                     
    Warning: Partition /dev/mmcblk0p4 is being used. Are you sure you want to continue?
    Yes/No? yes                                                               
    End?  [38.3GB]? 74792959s                                                 
    (parted)                                                                  
    Information: You may need to update /etc/fstab.
    
    root@pine64hot:~# parted /dev/mmcblk0    
    GNU Parted 3.2
    Using /dev/mmcblk0
    Welcome to GNU Parted! Type 'help' to view a list of commands.
    (parted) unit s                                                           
    (parted) p                                                                
    Model: SD 00000 (sd/mmc)
    Disk /dev/mmcblk0: 125042688s
    Sector size (logical/physical): 512B/512B
    Partition Table: msdos
    Disk Flags: 
    
    Number  Start      End        Size       Type     File system  Flags
     1      40960s     143359s    102400s    primary  fat16        lba
     2      143360s    7577599s   7434240s   primary  ext4
     3      7602176s   41156607s  33554432s  primary  ext2
     4      41189376s  74792959s  33603584s  primary  btrfs
    
    (parted) q

So it worked but given that this was parted 3.2 on Xenial and Wheezy ships with 2.3 this might get a support nightmare when package version change without notice. Maybe it's better to document this to be part of image customization and provide a few examples how to do this.

zador-blood-stained, 2016-03-26:
> 1) Is there a specific reason we still create a MBR partition table and not GPT?

Looks like [support for GPT partition tables in mainline u-boot](http://git.denx.de/?p=u-boot.git;a=blob;f=doc/README.gpt;hb=HEAD) is disabled by default for most devices (including sunxi-based). If all other u-boot versions (for other boards) support GPT, then you can switch to GPT at least for SD cards (NAND bootloader still will need MBR with first FAT partition I believe).

> 2) Same applies to the use of fdisk for the partition resizing in firstrun (most probably parted being outdated as hell in Wheezy ;) )?

What isn't outdated as hell in Wheezy?

> Playing a bit around with it looks already pretty discouraging since it depends on the parted version used. 

I already tested it a while ago, newer parted versions have `resizepart` command and they accept partition size in percents, older have `resize` command (If I remember correctly).

ThomasKaiser, 2016-03-26:
> newer parted versions have resizepart command and they accept partition size in percents, older have resize command

The main problem is getting parted resize a partition that is already mounted. This normally requires interactive mode but some parted versions accept the 'yes' in between as CLI argument, some don't and the strings you have to provide in interactive mode differ also between versions (according to tutorials I found on the net it worked to specify `resizepart 4 100%` followed by `yes` while Xenial's parted only accepts `resizepart 4`, then `yes` and then `100%`.

So unless we maintain an own parted package (which IMO isn't worth the efforts) I would consider this too risky. Too bad since this 'use only `95%` when SD card size is 4GB or below' would've been nice (tested with 3 older cards and it's close to unbelievable how slow they get when you fill them entirely).

For the other use case we still have the ability to specify a fixed size and then do further partitioning later manually or by an included script. I'll leave this issue open for a few days, maybe we can collect some ideas around this stuff. And thx for the information regarding u-boot prefering MBR (in 2016 -- WTF?)

zador-blood-stained, 2016-03-26:
I just checked again: GPT partition table support is enabled by default.

zador-blood-stained, 2016-03-26:
> Too bad since this 'use only 95% when SD card size is 4GB or below' would've been nice (tested with 3 older cards and it's close to unbelievable how slow they get when you fill them entirely).

It still can be done by calculating partition size manually to leave ~5% (or fixed, i.e. 200MB) free space after main partition.

ThomasKaiser, 2016-03-26:
> GPT partition table support is enabled by default (at least in current u-boot version).

Ah, ok. Misunderstanding since you wrote _disabled_ before (or maybe I didn't get it). I tried the 200MB example:

	device="/dev/"$(lsblk -idn -o NAME | grep mmcblk0)
	QuotedDevice=$(echo "${device}" | sed 's:/:\\\/:g')
	Capacity=$(parted $device unit s print -sm | awk -F":" "/${QuotedDevice}/ {print \$2*\$4/1024}")
	if [ ${Capacity} -lt 4000000 ]; then
		LastSector=$(parted $device unit s print -sm | awk -F":" "/${QuotedDevice}/ {print \$2 - 409600}")
	fi
	PARTITIONS=$(($(fdisk -l $device | grep $device | wc -l)-1))
	PARTSTART=$(parted $device unit s print -sm | tail -1 | cut -d: -f2 | sed 's/s//') # start of first partition
	PARTEND=$(parted $device unit s print -sm | head -3 | tail -1 | cut -d: -f3 | sed 's/s//') # end of first partition
	STARTFROM=$(($PARTEND+1))
	[[ $PARTITIONS == 1 ]] && STARTFROM=$PARTSTART
	((echo d; echo $PARTITIONS; echo n; echo p; echo ; echo $STARTFROM; echo ${LastSector} ; echo w;) | fdisk $device) >/dev/null || true

And while this will work we have exactly the same problem with `parted` as before. If the version of parted we install in the image changes for whatever reason the output format then things will go terribly wrong. :)

ThomasKaiser, 2016-03-26:
It should better read

    parted $device unit s print -sm | awk -F":" "/${QuotedDevice}/ {print \$2 - (200 * 1024 * ( 1024 / \$4 ))}"

instead (in case SD cards appear that do not use 512 byte sectors for whatever reasons)

zador-blood-stained, 2016-03-26:
> Ah, ok. Misunderstanding since you wrote disabled before (or maybe I didn't get it).

Sorry, I just missed this initially:
`include\config_distro_defaults.h(64): #define CONFIG_EFI_PARTITION`

zador-blood-stained, 2016-03-26:
> And while this will work we have exactly the same problem with parted as before. If the version of parted we install in the image changes for whatever reason the output format then things will go terribly wrong. 

I believe defining different commands based on `lsb_release -sc` output should be enough for now, assuming new parted versions won't be backported to wheezy and trusty.

ThomasKaiser, 2016-03-26:
> I believe defining different commands based on `lsb_release -sc` output should be enough for now

Good idea, the `lsb_release` output is already stored in `$DISTRIBUTION` and currently the resize stuff is known to work from wheezy up to xenial. I will also add some sanity check to the `do_expand_rootfs` and did already some cosmetic changes to variable names.

Will then implement `/root/.rootfs_resize` behaviour with percentage and sectors (leaving MiB, KiB and so one as an exercise for someone else ;) )

ThomasKaiser, 2016-03-26:
BTW: Any objections against replacing this whole block

	root_device=$(mountpoint -d /)
	for file in /dev/* ; do
		CURRENT_DEVICE=$(printf "%d:%d" $(stat --printf="0x%t 0x%T" $file))
		if [ $CURRENT_DEVICE = $root_device ]; then
			root_partition=$file
			break;
		fi
	done
	rootfstype=$(blkid -s TYPE -o value $root_partition)

with

    rootfstype=$(findmnt / | awk -F" " '/\/dev\// {print $3}')

?

zador-blood-stained, 2016-03-26:
It can be even simpler
```
findmnt / -o FSTYPE -n
```
but Igor [replaced](https://github.com/igorpecovnik/lib/commit/d4f8441b9a3a629aae1cc864bd3a8cf9e93025d9) this with current code because it didn't work in some cases

ThomasKaiser, 2016-03-26:
Hmm...

	tk@opennms:~$ lsb_release -a
	No LSB modules are available.
	Distributor ID:	Ubuntu
	Description:	Ubuntu 14.04.4 LTS
	Release:	14.04
	Codename:	trusty

	tk@opennms:~$ findmnt --target / -o SOURCE -n
	/dev/mapper/opennms--vg-root

	tk@opennms:~$ findmnt / | awk -F" " '/\/dev\// {print $3}'
	ext4

	tk@opennms:~$ findmnt / -o FSTYPE -n
	ext4

zador-blood-stained, 2016-03-26:
Seems like in Trusty this
```
findmnt --target / -o SOURCE -n
```
doesn't point to `/dev/mmcblk0p1`, which breaks FS resizing later, but
```
findmnt / -o FSTYPE -n
```
should get FS type in any case.

ThomasKaiser, 2016-03-26:
Just checked a 12.04 installation. Same behaviour, you get `/dev/mapper/VG--SSD-Root` with the inappropriate call and the right one with the latter:

    tk@tux:~$ findmnt / -o FSTYPE -n
    ext4

Will change it to your variant above :)

ThomasKaiser, 2016-03-26:
> I mean that root_partition variable is used later

Hmm... I double checked but both `root_partition` and `root_device` seem to be only used to determine `rootfstype`.

BTW: `firstrun` still starts an `apt-get update >/dev/null` but IIRC we wanted to disable this. Should I?

zador-blood-stained, 2016-03-26:
> Hmm... I double checked but both root_partition and root_device seem to be only used to determine rootfstype.

It is used [here](https://github.com/igorpecovnik/lib/blob/master/scripts/firstrun#L184) and [here](https://github.com/igorpecovnik/lib/blob/master/scripts/firstrun#L186)

> BTW: firstrun still starts an apt-get update >/dev/null but IIRC we wanted to disable this. Should I?

Why not while you are reworking firstrun script. I'll check `unattended-upgrades` configuration and add base packages upgrade procedure to old debootstrap process later.

ThomasKaiser, 2016-03-26:
Great! And also great that I seem to be blind :\ Will do it now this way:

	ROOTFS=$(findmnt / | awk -F" " '/\/dev\// {print $2"\t"$3}')
	set ${ROOTFS}
	root_partition=$1
	rootfstype=$2

BTW: Is there any reason to keep the old debootstrap process since the new one provides so much more opportunities and faster builds? What about a huge 'deprecated' header notice to remove it with Armbian 6.x?

igorpecovnik, 2016-03-26:
`findmnt / | awk -F" " '/\/dev\// {print $2"\t"$3}'`
does not work in wheezy. 

You are free to remove apt-get update from first run.

zador-blood-stained, 2016-03-26:
> BTW: Is there any reason to keep the old debootstrap process since the new one provides so much more opportunities and faster builds? What about a huge 'deprecated' header notice to remove it with Armbian 6.x?

New process may break some things like USB/SATA/NAND installation (especially with non-ext4 rootfs), so it needs more testing and feedback to mark it stable and use as default process.

ThomasKaiser, 2016-03-27:
> `findmnt / | awk -F" " '/\/dev\// {print $2"\t"$3}'` does not work in wheezy.

So I added a switch for wheezy vs. the others and sent PR #245

closed 2016-03-27 by ThomasKaiser