armbian/build #244: Bug BuildAll

kouaw, 2016-03-26:
Hello,

it's error to build-all.sh on line 76. 

awk: line 1: regular expression compile failed (missing '(')
)#enabled

Thank for the bugfix ;)

igorpecovnik, 2016-03-27:
Hmm, it's working fine on my build server. What is your host target?

kouaw, 2016-03-27:
my build server is Ubuntu 14.04 LTS ( i'm using link in the doc ) on the virtualbox and vmware,

but my server can not be correct because I can not get either to compile u-boot properly

`[ o.k. ] Target directory [ /home/orion/output/debs/ ]
[ o.k. ] Building deb [ linux-u-boot-odroidc2_5.06_arm64.deb ]
[ error ] ERROR in function compile_uboot [ common.sh:161 ]
[ error ] Building u-boot failed, check configuration
[ o.k. ] Process terminated
`

Sorry for my english, i'm french user.

closed 2016-03-27 by igorpecovnik