armbian/build #250: Dialog + Putty | Resize window on SSH kills compile.sh

Fourdee, 2016-03-29:
Probably not much you can do about this, however, if you resize the Putty window whilst running compile.sh during dialog screen, script stops at the current line.
[image]

Offtopic:
Can I just say, I love the building scripts you've created and what ARMbian stands for. Saved me no end of time compiling the OPi PC kernel, really very impressive, great work :+1:

ThomasKaiser, 2016-03-29:
BTW: Since it seems you're dealing with 3.4.x and H3. Regarding Opi One with its more primitive voltage regulator you have to take care that the minimum cpufreq you allow is not below the lower value defined in the fex file. We chose 648 MHz as minimum (since it doesn't matter consumption wise when the value would be lower) but also ensured that we then set minimum cpufreq to 648 MHz (instead of 480 Mhz like on the other OPi variantes where this makes a little difference since we adjust VDD_CPUX here more fine grained). 

When I did some tests a few weeks ago with cpufreq lower than dvfs operating point defined then strange things happened: seems there exists a driver bug, then the voltage regulator switches to 1.3V instead of remaining at 1.1V and then operating at 480MHz the board gets hotter and consumes more power than at 648MHz)

And in case you're not doing this already a quick look in `/etc/init.d/armhwinfo` should be worth a look (we should call it `optimizearmhw` in the meantime)

zador-blood-stained, 2016-03-29:
> Probably not much you can do about this, however, if you resize the Putty window whilst running compile.sh during dialog screen, script stops at the current line.

You can add `PROGRESS_DISPLAY=plain` to avoid using `dialog` at all.

Fourdee, 2016-03-29:
@zador-blood-stained 

> You can add PROGRESS_DISPLAY=plain to avoid using dialog at all.

Legend :+1: 

@ThomasKaiser 
Thanks for the info. 
DietPi is currently using using Loboris kernel. I'am trying to build the ARMbian kernel to replace the Loboris one. I can't get the OPi-PC to boot after installing the 4 compiled packages, so I'am assuming I'am missing some steps for the OPi-PC?
[image]

I also noticed ```script.bin_OPI-PC_res``` and ```*.fex``` is not compiled in the ARMbian .deb packages?

Loboris /boot:
```
root@DietPi:/boot# ls -lha
total 35M
drwxr-xr-x  3 root root 6.0K Jan  1  1970 .
drwxr-xr-x 21 root root 4.0K Feb 11 20:42 ..
-rwxr-xr-x  1 root root  32K Dec  3 14:29 boot0_OPI.fex
-rwxr-xr-x  1 root root 2.0K Mar 29 18:10 config.txt
drwxr-xr-x  5 root root 4.0K Mar 29 18:10 dietpi
-rwxr-xr-x  1 root root 9.9K Mar 29 18:10 dietpi.txt
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_1080p50_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_1080p50_hdmi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_1080p60_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_1080p60_hdmi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_480p_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_480p_hdmi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_720p50_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_720p50_hdmi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_720p60_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-2_720p60_hdmi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PC_1080p50_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-PC_1080p50_hdmi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PC_1080p60_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-PC_1080p60_hdmi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PC_480p_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-PC_480p_hdmi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PC_720p50_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-PC_720p50_hdmi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PC_720p60_dvi
-rwxr-xr-x  1 root root  35K Dec  3 14:29 script.bin.OPI-PC_720p60_hdmi
-rwxr-xr-x  1 root root  37K Dec  3 14:29 script.bin.OPI-PLUS_1080p50_dvi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PLUS_1080p50_hdmi
-rwxr-xr-x  1 root root  37K Dec  3 14:29 script.bin.OPI-PLUS_1080p60_dvi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PLUS_1080p60_hdmi
-rwxr-xr-x  1 root root  37K Dec  3 14:29 script.bin.OPI-PLUS_480p_dvi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PLUS_480p_hdmi
-rwxr-xr-x  1 root root  37K Dec  3 14:29 script.bin.OPI-PLUS_720p50_dvi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PLUS_720p50_hdmi
-rwxr-xr-x  1 root root  37K Dec  3 14:29 script.bin.OPI-PLUS_720p60_dvi
-rwxr-xr-x  1 root root  36K Dec  3 14:29 script.bin.OPI-PLUS_720p60_hdmi
-rwxr-xr-x  1 root root 880K Dec  3 14:29 u-boot_OPI-emmc.fex
-rwxr-xr-x  1 root root 880K Dec  3 14:29 u-boot_OPI.fex
-rwxr-xr-x  1 root root  11M Dec  3 14:29 uImage
-rwxr-xr-x  1 root root  11M Dec  3 14:29 uImage_OPI-2
-rwxr-xr-x  1 root root  11M Dec  3 14:29 uImage_OPI-PLUS
```

igorpecovnik, 2016-03-29:
There is one package more, "root filesystem changes" and there are script.bin files and some other tweaks, check one of those files - I think it's not building within kernel only.

Example:
linux-jessie-root-bananapipro_5.07_armhf.deb

ThomasKaiser, 2016-03-29:
Well, loboris uses Allwinner's BSP u-boot (2011.09) and a different partition scheme. We use mainline u-boot 2016.03 and for most use cases a single partition scheme.

I would suspect the latter is the problem since when Armbian's u-boot package is installed then both boot.scr and kernel should be located on the 1st partition. I just realize that I only once dealt with Armbian and a 2 partition scheme (experimentally using F2FS for the rootfs). Ah, Igor answered already :)

zador-blood-stained, 2016-03-29:
If I remember correctly, fex/bin files are packed into board support deb package (linux-root-...). To build it with `KERNEL_ONLY=yes`, you need to set `RELEASE` to one of supported OS release codenames. 

In any case installing armbian kernel on non-armbian image may not be enough because of firstboot script that manages .bin files for different boards.

ThomasKaiser, 2016-03-29:
@Fourdee is your image using FAT for the 1st partition?

Fourdee, 2016-03-29:
@ThomasKaiser 
Yep, FAT, not sure if its 16/32 mind: ```/dev/mmcblk0p1: LABEL="BOOT" UUID="2352-F773" TYPE="vfat" PARTUUID="5a390d11-01"```.

I'am compiling again with ```KERNEL_ONLY=no```, will see if the extra package gets made.

ThomasKaiser, 2016-03-29:
IIRC then the package installation might fail since we're trying to use a symlink from `/boot/bin/${board}.bin` to `/boot/script.bin`.

Anyway if you want to maintain a set of .bins for different boards and resolution settings you could create them with a script and then simply store them all within /boot/ (but you should use our fex files for all the boards and there's one issue with Ethernet regarding Plus vs. the others that might prevent rebooting if the wrong script.bin is used at boot time -- that's the reason we still provide 2 different images for H3 boards)

Fourdee, 2016-03-29:
Ok, full image creation worked for the missing packages :+1: :
- linux-jessie-root-orangepih3_5.06_armhf.deb
- sunxi-tools_1.3-1_armhf.deb

Installed all .debs, and ```cp /boot/bin/orangepipc.bin /boot/script.bin``` won't boot. 

I think i'll write your ARMbian image tomorrow and see how the image is running with the single partition, uboot, .fex and script.bin. 
With DietPi, I try to use an existing image for the device where possible, so all of this complete image creation, uboot "stuff" is still a new area for me. Regardless I'am loving the ARMbian building tools, exactly what these SBC's and devs needed.

Fourdee, 2016-03-29:
> Anyway if you want to maintain a set of .bins for different boards and resolution settings you could create them with a script and then simply store them all within /boot/ (but you should use our [...]

```dietpi-config``` uses the a similar existing system to essentially copy the required .bin file, when user changes resolution: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-config#L606-L643

ThomasKaiser, 2016-03-29:
You find the fex files in `lib/config/` too (not only Oranges but Banana Pi M2+, Nano Pi M1+, Beelink X2 -- all H3 based). And you could also have a look into the `h3disp` utility that is a bit more flexible and provides more than just 3 different resolutions.

BTW: Using 'existing' images always scares me a bit since who knows what has been done before? By creating the whole thing automatically from scratch without any human intervention you know what you get. :)

Fourdee, 2016-03-30:
@ThomasKaiser 
Thanks :+1: 

I found them in a different location:
root@orangepipc:~# ls -lha /lib/config
ls: cannot access /lib/config: No such file or directory

```
root@orangepipc:~# find / -type f -name *.fex
/usr/lib/nand-sata-install/h3/boot0_OPI.fex
/usr/lib/nand-sata-install/h3/u-boot_OPI-emmc.fex
```

> BTW: Using 'existing' images always scares me a bit since who knows what has been done before? By creating the whole thing automatically from scratch without any human intervention you know what you get. :)

Yep, good advice, here's our current list:
- RPi image is based on Jessie Lite
- Odroid images are based on Meveric's Debian Jessie images.
- OPi PC based on Loboris image.

All images have a lot of work done on them before its DietPi ready (eg: stripping additional packages, setting up dietpi-services, ramlog etc etc).

Depending on how this turns out with the OPi PC, i will look into using ARMbian for all excluding RPi (which you guys dont support at the moment?).

ThomasKaiser, 2016-03-30:
> I found them in a different location

Sorry, misunderstanding since I was referring to `[lib/config/](https://github.com/igorpecovnik/lib/tree/master/config)` in our repo. And no, currently no support for Raspbian planned due to user base and technical/documentation challenges ('Why do you support RPi 2/3 but not A/B?')

Fourdee, 2016-03-30:
@ThomasKaiser 

> (https://github.com/igorpecovnik/lib/tree/master/config) in our repo

Aha, excellent :)

> 'Why do you support RPi 2/3 but not A/B?'

We support all RPi devices on DietPi (a/b/zero etc), excluding the compute module.

ThomasKaiser, 2016-03-30:
> We support all RPi devices on DietPi (a/b/zero etc)

I know. But supporting ARMv6 and proprietary boot procedures is nothing we want to spend time on :)

Fourdee, 2016-03-30:
>  But supporting ARMv6 and proprietary boot procedures is nothing we want to spend time on :)

Ok, I think I can sum this up with two questions, to try and get some closure on this:

#### The main question is:
Would you guys consider adding an option, that allows the user to compile the ARMbian image, with a optional FAT partition at the start with ```/boot``` on the partition?

This is critical for DietPi as we store a settings file in the FAT partition, which allows our users (mainly Windows) to setup WiFi creds, automation options prior to boot: https://github.com/Fourdee/DietPi/blob/master/dietpi.txt

```/boot``` on the FAT partition also seems a common practice with all SBC's core devs, with the benefit of allowing the user to modify ```boot.ini``` (eg: Odroids) prior to booting on a MS Windows system.

As you already mentioned, you have symlinks active in /boot, so this may be a bigger modification than expected.

#### The other question is:
If the above is not possible, or rejected, how would I (or anyone else in my situation) go about converting a single ext4 partition ARMbian image, into a FAT/ext4 partitioned image with ```/boot``` on the FAT partition?

Fourdee, 2016-03-30:
@ThomasKaiser 
Intresting, you created the following script: http://kaiser-edv.de/tmp/H9rWPf/fix-thermal-problems.sh :+1:. Quoted here: http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=785&fromuid=29411

For now, your script will be run during our v115 update, to all OPi-PC users.

zador-blood-stained, 2016-03-30:
> Would you guys consider adding an option, that allows the user to compile the ARMbian image, with a optional FAT partition at the start with /boot on the partition?

Already supported internally for several devices with old u-boot version, but there is no user defined option to use this with any device.

zador-blood-stained, 2016-03-30:
Added (untested): https://github.com/zador-blood-stained/lib/commit/f135eca2b22e51c3fed1c5f4e87a5ada8918a26d
Once merged to main repo, you should be able to use `BOOTSIZE=x` parameter to create FAT boot partition of `x` megabytes.

ThomasKaiser, 2016-03-30:
@Fourdee you should be aware that zador's fix might not be enough. I do not even know how the combination of loboris' settings/kernel behaved when ran with the 'wrong' networking settings (Plus and Plus 2 use an external GbE capable PHY and all the other Oranges the internal Fast Ethernet PHY) but with our setup currently we ship one single kernel image (unlike loboris who used two different ones) and rely on fex settings.

And when using the wrong fex settings the kernel might crash at shutdown. This is something we didn't spent that much time on but it might become a showstopper. Since 'wrong' script.bin leads to an Orange Pi PC freezing when used with fex for the Plus and vice versa.

On top: Many more H3 devices will appear soon (all the SBCs being a rip-off of Xunlong's work, compare with FriendlyARM's NanoPi M1 or SinoVoip's M2+) so you will soon face the situation to support even more 'devices'.

igorpecovnik, 2016-06-23:
Topic related issues is solved, closing.

closed 2016-06-23 by igorpecovnik