armbian/build #253: Compile Odroid C2

kouaw, 2016-03-31:
Hello, 

Sorry the bug of compiling odroid C2 is not fix :

look the message : 
`
../fip/fip_create: 1: ../fip/fip_create: Syntax error: "(" unexpected
make[1]: *** [fip.bin] Error 2
make: *** [sub-make] Error 2
[ o.k. ] Target directory [ /home/orion/output/debs/ ]
[ o.k. ] Building deb [ linux-u-boot-odroidc2_5.07_arm64.deb ]
[ error ] ERROR in function compile_uboot [ common.sh:163 ]
[ error ] Building u-boot failed, check configuration
[ o.k. ] Process terminated
`

igorpecovnik, 2016-04-01:
Compilation works - I just tested from clean sources. Perhaps you are using too recent compiler. Are you using recommended host setup?

kouaw, 2016-04-01:
I use a laptop with ubuntu 14.04 SSD / quad core.

I already clear the source folder but this changes nothing.

I am willing to compile tested for another image but I feared that the problems is in the program fip_create

kouaw, 2016-04-01:
Compile Odroid C1 is ok for created .deb but no image created ^^

igorpecovnik, 2016-04-02:
Hmm, hard to say what is wrong, what about compiler? If you upgrade it by any chance, this is the reason. Can you try manual compile as instructed by Hardkernel? My script is not doing anything different, except we use system compiler, which is proper one, if you use Ubuntu Trusty.

Odroid C1 is not done yet, but you should be able to make an image. The boot process is broken.

kouaw, 2016-04-02:
hello, i try manual compile via instructed by Hardkernel but is not good compiling :+1: 
`  OBJCOPY u-boot.hex
../fip/fip_create: 1: ../fip/fip_create: Syntax error: "(" unexpected
make[1]: *** [fip.bin] Erreur 2
make: *** [sub-make] Erreur 2
`

kouaw, 2016-04-03:
Hello, I clean and reload Ubuntu.
It's Okay now. Sorry

closed 2016-04-03 by kouaw