armbian/build #258: Add support for A20-Olinuxino-Lime2-eMMC

chradev, 2016-04-06:
There is a new option to A20-Olinuxino-Lime2 HW rev. E with eMMC (4 GB) instead of NAND. eMMC chip is connected to A20 via MMC2 and can be used to boot from.

It could be great if one is able to:
- boot from eMMC with rootfs (read only preferable) on it;
- alternatively boot from any other bootable device on SATA (HDD/SSD), USB (HDD/Flash) or SD Card (on MMC1) if present (in that order preferable);
- in case of other bootable device presence and user button pressed (at reset or power on) to boot from eMMC forcibly.

I have A20-Olinuxino-Lime2-eMMC boards and time to take part in the process.

igorpecovnik, 2016-04-09:
I saw you solved this? Do you have an stable / working patch now?

chradev, 2016-04-09:
I have currently solved the problem with eMMC only for the kernel 3.4 and 4.4 but doing it without some of the patches (for dts files) needed. There is also patches needed for U-Boot and boot scripts.

On the other hand the board support should be split for Lime2-NAND and Lime2-eMMC boards. Lime2 without any option could be supported in both cases.

There is also one issue not solved for now but not only for Lime2 boards - slow GBit Ethernet which Olimex claim to be solved in their new images for Lime2 (-4GB and -eMMC) boards. In my opinion it is good to be solved as well.

Unfortunately, I am not sure of my ability to create all needed patches because of luck of experience.

chradev, 2016-04-10:
Hi Igor,

Tonight is happy night for me. I succeed to patch U-Boot (v2015.10) and kernel (4.4.6) to work with eMMC on A20-Olinuxino-Lime2-eMMC board. Slow GBit Ethernet has also a fix in U-Boot.

U-Boot staff (should work for default case as well - not tested):
userpatches/u-boot/u-boot-next/a20-lime2-emmc-uboot.patch
userpatches/u-boot/u-boot-next/a20-gbit-fix-uboot.patch

Mainline kernel staff:
userpatches/kernel/sunxi-next/a20-lime2-emmc-kernel-next.patch

There you are above files archived:
(Unfortunately I cannot attach the zip file to the issue)

Find more information including final patches [here](http://forum.armbian.com/index.php/topic/853-armbian-customization/page-2#entry7491) and [here](http://forum.armbian.com/index.php/topic/853-armbian-customization/#entry7494).

More testing is advisable before adding to repository.
It is good idea to add following new boards lime2-nand and lime2-emmc especially for Armbian default (legacy kernel 3.4) because NAND and eMMC Lime2 options are exclusive.

Next step is to prepare u-boot script to boot from available devices.
Script (like nand-sata-install) to copy system on eMMC is also needed.

Best regards
Chris

chradev, 2016-04-17:
Hi Igor,

I have tested patches for vanilla kernel (4.5.1) and mainline U-Boot (v2015-10):
 * patch/kernel/sunxi-next/add-emmc-lime2.patch
 * patch/u-boot/u-boot-next/add-emmc-lime2.patch
and they work fine booting from both SD and eMMC.
I will try to test other patches but I have a question about  config/lime2-emmc.fex.
Have I to run compile.sh with `BOARD=lime2-emmc` and will other staff comply with it?

One more question: 
I do not see slow Ethernet fix as proposed by Olimex and tested to work by me.
I see other disabled patch sunxi-gmac-fix.patch for U-Boot instead.

Is Hans de Goede's solution working / tested and will be official one?

Best regards
Chris

igorpecovnik, 2016-04-18:
We are (temporally) using U-boot 2016.01 for A10 / A20 boards where those problems should not exists. I only tested on Bananas / Cubietruck where same problems emerge. My Lime2 is in use and I hard to do experiments.

Tnx for test.

chradev, 2016-04-18:
Thanks Igor,
I will try to migrate to U-Boot 2016.01.
Unfortunately, I have encountered a problem with latest changes (or not) "bricking"my Lime2 board.
The build was with your patches but reverting to my did not solve the problem (but who knows where the rabbit is hidden). See my post [here](http://forum.armbian.com/index.php/topic/853-armbian-customization/#entry7895)

chradev, 2016-04-20:
Hi Igor,

The problem I reach is "[two SD card boot problem](http://forum.armbian.com/index.php/topic/853-armbian-customization/page-3#entry7895)" and it is not related to patches.

chradev, 2016-04-22:
I am facing at a problem to write to eMMC from U-Boot (2016.05-rc1 and 2015.10).
My try to update and save U-Boot environment finishes with fail to write to eMMC

=> setenv boot_targets scsi0 usb0 fel mmc0 mmc2 pxe dhcp
=> saveenv
Saving Environment to MMC...
Writing to MMC(0)... mmc write failed
failed

Any thoughts how to fix it?

igorpecovnik, 2016-04-22:
No, this sounds like a bug to me but am also not much competent around u-boot. Try denx mailing list.

zador-blood-stained, 2016-04-22:
Looks more like u-boot configuration issue. There are different ways to store environment, including writing to file on ext4 FS and writing to "raw' sector on MMC. In your case u-boot tries to use the latter, but I believe it needs more build time configuration (or your environment is too big).
```
	printf("Writing to %sMMC(%d)... ", copy ? "redundant " : "", dev);
	if (write_env(mmc, CONFIG_ENV_SIZE, offset, (u_char *)env_new)) {
		puts("failed\n");
		ret = 1;
		goto fini;
	}
```

chradev, 2016-04-22:
The size of environment much less than the space on the device
`Environment size: 3482/131068 bytes`
On the other hand the same action works fine on SD/MMC

chradev, 2016-04-23:
> 
I am facing at a problem to write to eMMC from U-Boot (2016.05-rc1 and 2015.10).
My try to update and save U-Boot environment finishes with fail to write to eMMC
=> setenv boot_targets scsi0 usb0 fel mmc0 mmc2 pxe dhcp
=> saveenv
Saving Environment to MMC...
Writing to MMC(0)... mmc write failed
failed

I have tested the failing case on the second a20-Olinuxino-Lime2-eMMC board I have.

And it works fine!
Definitely the problem is in the HW.

igorpecovnik, 2016-04-23:
Good to hear this! Let's close this issue.

closed 2016-04-23 by igorpecovnik

lauhub, 2016-06-16:
Following Chradev instructions to build an image, when I try to boot kernel, it always reboot and restart the system, in an endless loop (see [endless-reboot-loop.log.txt](https://github.com/igorpecovnik/lib/files/318668/endless-reboot-loop.log.txt)).

The loop does not occur when booting from SD card.

I also have the same problem to write to eMMC from u-boot with two boards (Lime2 Rev. E): 
```
=> saveenv
Saving Environment to MMC...
Writing to MMC(0)... mmc write failed
failed
```
I tried to remove the cheap-mmc patch (u-boot-02-support-cheap-mmc.patch) without success.

Chradev, did you solve this for your first board ?

Would you have any leads to troubleshoot or solve this issue ?

(PS: I re-open this issue, so do not hesitate to tell me if I was wrong to do this and if I should have open another discussion)

chradev, 2016-06-16:
Hi lauhub,

As I can see in endless-reboot-loop.log.txt you try to load system from eMMC which is device 0 for uboot. Your boot script send to kernel "... root=/dev/mmcblk0p1 ..." but the kernel assigns eMMC as mmcblk1 which is causing the system to reboot.

You can use my solution to send root with PARTUUID as described in: 
[http://forum.armbian.com/index.php/topic/853-armbian-customization/page-4#entry8032](url)

About the problem to write to eMMC from u-boot I try to comment the issue with Olimex but without success and have not a solution. Currently I overcome this issue by modification of default environment parameters in uboot as described in:
[http://forum.armbian.com/index.php/topic/853-armbian-customization/page-4#entry8063](url)

Best regards
Chris

lauhub, 2016-06-17:
Hi Chris,
Thanks a lot, I could finally make it work. All your instructions helped a lot, that is a great work !
I'll also try to compile u-boot with debug mode, but at first try I had compilation errors.
Best regards