armbian/build #263: Build automation & testing via qemu or other emulation

faddat, 2016-04-07:
So I'm really into build automation.  Is there currently an automated (per-commit) system that tests lib?  

Has anyone thought  about he best way to cobble one together?  

I'm gathering thoughts before I try to do it and push it, so if you've got ideas on this, I'd love to hear them.

zador-blood-stained, 2016-04-07:
So what do you want to test?

bash code syntax? Easy to implement like this
```
for f in *.sh; do
  bash -n $f || exit -1
done
```
and connect to something like Travis CI.

Kernel and u-boot patches? Things start to look more complicated because now we depend on external repositories and there are at least 10 different kernel/u-boot source variations and it's not that easy to detect which sources family needs testing after "lib" repository update.

Image creation? Things start to look even more complicated because currently superuser rights are required for loop device access (can be reworked to use libguestfs and fuse, but it will require significant changes). In addition debootstrap and package installation depends on network speed (unless apt-cacher is used) and again there are different release and device targets.

In the end even if image was built successfully it still can fail to boot properly for a variety of reasons.

faddat, 2016-04-07:
I hear you.  

Actually this is part of what I asked: to determine what can/cannot be te4sted.  Generally the board I am focued on right now is opipc, but you can extrapolate these ideas to any board....

My thought expeiment was thus:  If I can virtualize the thing, I can get rapid dev cycles.  And if I can get rapid dev cycles, well you get the idea, more, cooler things, done faster.  In a few days of using lib consistently I've noticed that sometimes a change to one thing will take down another.  My ideal automation would build all oses & kernels for all boards each time there was a commit to master.  

I can configure the systems, but will need pointers on the current known limits of testability.  Can you tell me any more about why I can't test an image?  Simply because I cannot replicate its actual state, or...?

Thanks!

-Jake

igorpecovnik, 2016-04-07:
Image is hard to test. How you would know that u-boot will properly init hardware?

zador-blood-stained, 2016-04-07:
> Can you tell me any more about why I can't test an image? Simply because I cannot replicate its actual state, or...?

Simply because you can't emulate SoC specific hardware/IP blocks. Well, some devices can be tested via FEL mode in more or less automated way, but it's more hardware engineering than software build automation.

faddat, 2016-04-07:
Thanks for the information guys.  I think that in the end I am indeed looknig to thest the whole image.  I'd like to set up a build machine for this project that spits out all of the images when changes are made, that way, well, feedback loops stay tight, and awesomeness occurs.  How does U-boot work?  I am thinking maybe I can make a test of some sort, but maybe not go all the way in.  If I undersatnd ight, I could check u-boot in software (its characteristics, as opposed to if it boots) and then I could chroot into /.  Is that right?

igorpecovnik, 2016-04-08:
A progress will be made if we extract valuable information from auto building and if we do the kernel / uboot and board support packages in auto manner. This level is good enough.

I already done something since some time. It's possible to build all images but am doing no checkups. Run with:

BUILD_ALL="yes" # do all enabled images
BUILD_ALL="demo" # only shows what will be built

faddat, 2016-04-08:
K.  I'm giong to build all, and then try to figure out what can or cannot be automated in terms of checking image quality.  The manual testing sounds like a LOT of work-- Thanks for that :).

faddat, 2016-04-09:
I have recently found kernelci.org.   I have asked Steven to send them a board, which partially solves this.... But only partially.  I am going to close this issue and create a new one describing the feature/flow I hope to eventually achieve, for the sake of clarity.

closed 2016-04-09 by faddat