armbian/build #266: orangepiPC

faddat, 2016-04-07:
okay, here's my status-- my Opi boots but only on legacy.  using dev or next does just about nothing-- no lights (other than the ethernet lights) and no bootage.  

If you need/want more info, just ask :)

igorpecovnik, 2016-04-08:
only legacy and possibly DEV. I forgot if lights are coming up here too.

next don't work.

ThomasKaiser, 2016-04-08:
Orange Pi PC booted through FEL based on latest code/patches: http://pastebin.com/0wxkhn7P

Leds and network and everything else works:

    ./compile.sh KERNEL_ONLY=no BOARD=orangepih3 PROGRESS_DISPLAY=plain RELEASE=jessie BUILD_DESKTOP=no EXTENDED_DEBOOTSTRAP=yes ROOTFS_TYPE=fel PROGRESS_LOG_TO_FILE=yes BRANCH=dev FEL_AUTO=yes

faddat, 2016-04-08:
For FEL, I use a USB drive and hold the button, correct?

zador-blood-stained, 2016-04-08:
No, you connect board OTG port to your build host, hold FEL button while resetting or powering on only and check [this](https://github.com/igorpecovnik/lib/blob/master/documentation/fel-boot.md) and [this](https://linux-sunxi.org/FEL/USBBoot) for more info.

FYI last time I checked trusty target didn't work with root with FEL/NFS.

faddat, 2016-04-08:
Could FEL be used todo automated testing, for example:  I've got 20 boards hooked up to 20 usb ports.  I can reboot them remotely, and since I can time the boots I can ensure the right image is in place at boot time.... or not so much?

ThomasKaiser, 2016-04-08:
Yes and no, see #198

faddat, 2016-04-12:
Anyone know how to make a quick UART reader for the serial out?  Here's the istuation:  Armbian 5.07 doesn't work on opi-pc no matter how roms are made.  I get a solid network light, but DHCP never happes according to my router.  Then I also get a green light, which coaasionally turns off and on again.  When I use the 3.4 image at Armbian, that works.

ThomasKaiser, 2016-04-12:
I still don't get what you're talking about? What is a rom? And are you asking for a serial console adapter like [this](http://www.ebay.com/itm/400565980256)?

"Armbian 5.07" doesn't work but 'the 3.4 image at Armbian" does? Sorry?

If you want to try out vanilla kernel, then [this image](http://kaiser-edv.de/tmp/4U4tkD/Armbian_5.07_Orangepih3_Debian_jessie_4.6.0-rc1.7z) is confirmed to work with onboard Ethernet and 4.6.0-rc1.

faddat, 2016-04-12:
Please ignore a reasonable amount of that message, in fact.... It is now dleted.  Thoday was a shit day, what can I say?  

:)  Tomorrow will be better...

closed 2016-04-12 by faddat