armbian/build #269: Project Intent

faddat, 2016-04-08:
I have always been a very big plan of the rockchip platform 3188.  Less so its more expensive counterparts, 3288, and 3368, which don't seem to offer very much price/performance bang.  Anyway, my question is this:

Would the "intent" of lib include 3188 boards?  

E:g:  If I am making build scripts for 3188 devices, do you think I should work them into lib at the same time? I haven't yet done a deep study of lib's who's and how's but I'll be doing that soon.  Is it a situation of well, it's armbian, so anything using ARM is fair game or.... ?

Thanks

igorpecovnik, 2016-04-09:
There is no particular reason why we don't have Rockchip boards. For adding the board you must be confident with manual building, than we can move it to the script. I can assist.

faddat, 2016-04-09:
Okay, I will first work on getting confident.  Right now I am iffy.  With the build process, I've done it but not always with the best results.  Also because of some things going on with Klouds I think that I will be unable to put in the time for about another month or so.

Thanks!

closed 2016-04-09 by faddat