armbian/build #271: Building without Docker or Vagrant on Ubuntu Xenial

faddat, 2016-04-11:
I'm getting 2 errors on Ubuntu Xenial:

[image]

The script continues, so I am not sure if these are showstoppers or not.

igorpecovnik, 2016-04-11:
USB redirecor is known to be broken, so it's safe to ignore and also this Invalid argument is not critical. Try booting the board ;)

BTW: try also to use alternative building method:

EXTENDED_DEBOOTSTRAP="yes"
FORCE_USE_RAMDISK="yes

It's faster.

faddat, 2016-04-11:
Wonderful!  Thank you!

Jacob Gadikian
E-mail: faddat@gmail.com
SKYPE: faddat
Phone/SMS: +84 167 789 6421

closed 2016-04-14 by faddat

igorpecovnik, 2016-04-14:
BTW, usb redirecor is fixed, one error less :)
https://github.com/igorpecovnik/lib/commit/eda047936ba53788128a48cf49bd4acd9ba20b29