armbian/build #272: Why does a search for DMA get rejected?

madGambol, 2016-04-12:
I'm looking for updates or discussions about SPI DMA on the Cubieboard. When I enter SPI or DMA I get a notification that those search terms aren't allowed! I don't understand. Those are "forbidden" topics at the Arabian forums? ????? How is that possible?

I'm trying to find the fix for the 63 byte limit on the SPI device. I need more than that under DMA support.

Thanks,

Chris Arena
Portsmouth, Rhode Island

ThomasKaiser, 2016-04-12:
Are you kidding?! Instead of reading the reason that has been displayed to you ("One or all of your search keywords were below 4 characters...") you really open an issue here?

Please think about closing this issue and doing a Google search for eg. 'SPI DMA site:armbian.com' if you think kernel stuff for A10/A20 **would** Armbian/Cubieboard related

madGambol, 2016-04-13:
Close the issue.

I was simplifying. I had done spi dma and other variations that either
ended with that message or 404. It seemed like a malfunction.

I've never, ever seen such an arbitrary restriction. You have to go off
site to search in the site? What a unexpected surprise!

Since I can't seem to enter a topic to ask a question I looked for any way
to make contact. Thanks.

Chris

... there is nothing either good or bad, but thinking makes it so.
-- Shakespeare

igorpecovnik, 2016-04-13:
This is a limitation of forum software. Unfortunately changing this to "3 characters search" is far from simple even it seems logical to only alter one parameter. But it's not. Since it require extra work, this task become just one of not urgent / not important ones which sits and waits that I'll find that extra time. Once this will be fixed ... since it annoys me too.

closed 2016-04-13 by igorpecovnik