armbian/build #273: Preselected values, like IP address

faddat, 2016-04-14:
So, one of my myriad uses for lib is distributing images that are immediatley ready to perform a distributed network function when they reach the user.  I know that there are many ways, but I wanted to check in and see if there was a recommendation regarding the best way for:

* Entering the IP address.  I'm looking at setting up systemd-networkd and enabling DHCP on the network interface, then tunneling to join the rest of our network, regardless of the address given by DHCP (except of course for addresses that can't reach the rest of our network)

    Current plan:  Modify script to enable systemd-networkd, and put the .network and .netdev files in the right places

I'm going to end up adding to this thread as I think of things, but that's about all I've got at the moment.  Later today I will give build automation on a cluster a shot, and I'll let you know how it's going.  I'm going to put up a lib repository mirror at git.klouds.org so I can get the webhooks set up.  Not sure why I mention it, but it seems worthy of mention.  I will submit pull requests and the like here regardless of having a mirror.  

(Slightly Off-Topic)
I want to make something like lib for Android, specifically CyanogenMod.  I've been plugging away at that for a while now, and what's been keeping me back is the lack of availablity of some of the binaries that are usually pulled from the device at build time.  

After CiyanogenMod, Ubuntu Touch.

zador-blood-stained, 2016-04-14:
> Entering the IP address. I'm looking at setting up systemd-networkd and enabling DHCP on the network interface, then tunneling to join the rest of our network, regardless of the address given by [...]

Sounds like a job for `customize-image.sh` script. Something like this:
```
# kill existing network configuration
echo > /etc/network/interfaces
# enable necessary services
systemctl --no-reload enable systemd-networkd systemd-networkd-wait-online
# create or copy configuration
cp /tmp/overlay/eth0.network /etc/systemd/network/eth0.network
# etc
```

faddat, 2016-04-14:
@zador-blood-stained Thank you!  Cool, that's what I was thinking as well.  

Has anyone here built android much?  Some devices have pretty good support for newer kernels, I know.  I'm going to go and check to see if the port of ubuntu touch is complete for my unwise-purchase-of-a-cell-phone, my Nexus 6.

closed 2016-04-26 by igorpecovnik