armbian/build #274: Legacy build stops when choosing Xenial target

ThomasKaiser, 2016-04-14:
I tried to build a Xenial image for Banana Pi M2+ (stops at u-boot stage since the u-boot-dev patch needs to be copied to default) and then orangepih3. Stops reproducible with both debootstrap and debootstrap-ng (different messages). NG below:

    dpkg-deb: building package 'linux-firmware-image-sun8i' in '../linux-firmware-image-sun8i_5.07_armhf.deb'.
    dpkg-deb: building package 'linux-headers-sun8i' in '../linux-headers-sun8i_5.07_armhf.deb'.
    dpkg-deb: building package 'linux-image-sun8i' in '../linux-image-sun8i_5.07_armhf.deb'.
    [ o.k. ] Creating board support package [ orangepih3 ]
    [ o.k. ] Fingerprinting [ Armbian 5.07 Orangepih3 Ubuntu xenial default Linux ]
    [ o.k. ] Building package [ linux-xenial-root-orangepih3 ]
    [ o.k. ] Starting build process for [ orangepih3 xenial ]
    [ o.k. ] Creating new rootfs for [ xenial ]
    [ o.k. ] Installing base system [ Stage 1/2 ]
    I: Retrieving InRelease 
    I: Checking Release signature
    I: Valid Release signature (key id 790BC7277767219C42C86F933B4FE6ACC0B21F32)
    I: Retrieving Packages 
    I: Validating Packages 
    I: Found packages in base already in required: locales 
    I: Resolving dependencies of required packages...
    I: Resolving dependencies of base packages...
    I: Checking component main on http://localhost:3142/ports.ubuntu.com...
    E: Couldn't find these debs: debconf-utils
    [ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:173 ]
    [ error ] Debootstrap base system first stage failed 
    [ o.k. ] Process terminated 
    [ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:559 ]
    [ error ] debootstrap-ng was interrupted 
    [ o.k. ] Process terminated 

Any idea what's wrong? Building jessie works just fine.

zador-blood-stained, 2016-04-14:
This is caused by `debconf-utils` moving (?) to "universe" repository component [[1]](http://packages.ubuntu.com/xenial/debconf-utils), so `debconf-utils` should be removed from [here](https://github.com/igorpecovnik/lib/blob/master/debootstrap-ng.sh#L168) and [here](https://github.com/igorpecovnik/lib/blob/master/debootstrap.sh#L102)

ThomasKaiser, 2016-04-14:
Seems to work, thx! I would suspect you fix this with your next commit?

zador-blood-stained, 2016-04-14:
Based on [this](https://wiki.ubuntu.com/XenialXerus/ReleaseSchedule) I would suspect that this change in Xenial is final, so [here we go](https://github.com/zador-blood-stained/lib/commit/b69a1400782ea7fa1280568069c1a7bf085b9278).
Hopefully it doesn't break older releases.

closed 2016-04-14 by ThomasKaiser