armbian/build #276: Can't find default configuration "arch/../configs/FriendlyARM_NanoPi_M1_defconfig"!

atlury, 2016-04-17:
***
*** Can't find default configuration "arch/../configs/FriendlyARM_NanoPi_M1_defconfig"!
***
make[1]: *** [FriendlyARM_NanoPi_M1_defconfig] Error 1
make: *** [FriendlyARM_NanoPi_M1_defconfig] Error 2

atlury, 2016-04-17:
Its happening while trying to build the latest dev kernel...just a while back...I selected NanoPi m1 wifi in the board selection menu

zador-blood-stained, 2016-04-17:
NanoPi M1 was added only yesterday and only for default branch. You can try copying `lib/patch/u-boot/u-boot-default/u-boot-99-opi-change-build-settings.patch` to `userpatches/u-boot/u-boot-dev/` and repeat compilation, but I believe this configuration is not tested yet.

atlury, 2016-04-17:
Thanks, its compiling now. Shall test the configuration and report.

closed 2016-04-21 by igorpecovnik