armbian/build #279: 64bit feature of ext4 breaks u-boot

oleid, 2016-04-21:
As described in https://github.com/igorpecovnik/lib/pull/277,  images created on debian testing won't boot, since u-boot only sees an empty boot folder.

Thus, something like  `-O ^64bit` needs to be passed to mkfs.ext4 when creating the filesystem for the boot partition.

I _think_ the ext2fstools of ubuntu 16.04 might be recent enough to trigger the bug, too.

zador-blood-stained, 2016-04-21:
> I think the ext2fstools of ubuntu 16.04 might be recent enough to trigger the bug, too.

http://packages.ubuntu.com/ru/xenial/e2fsprogs

Just unpacked the package and checked `/etc/mke2fs.conf`
```
[fs_types]
	ext3 = {
		features = has_journal
	}
	ext4 = {
		features = has_journal,extent,huge_file,flex_bg,uninit_bg,dir_nlink,extra_isize
		auto_64-bit_support = 1
		inode_size = 256
	}
	ext4dev = {
		features = has_journal,extent,huge_file,flex_bg,uninit_bg,dir_nlink,extra_isize
		inode_size = 256
		options = test_fs=1
	}
```

zador-blood-stained, 2016-04-21:
```
       auto_64-bit_support
              This relation is a boolean  which  specifies  whether  mke2fs(8)
              should  automatically  add  the  64bit  feature if the number of
              blocks for the file system requires this feature to be  enabled.
              The resize_inode feature is also automatically disabled since it
              doesn't support 64-bit block numbers.
```
So if I understand this correctly, Xenial 16.04 won't be affected, and as for Stretch, it's not on the list of recommended build hosts and it's a *testing* branch, so this issue should be addressed once it is officially released.

However there is another possible issue - using `nand-sata-install` script on Jessie image upgraded to testing, and this should be addressed sooner.

zador-blood-stained, 2016-06-04:
Fixed by 7506d9d3c903b76054a83fe57d4181e46759b1ad, 28ba22b7aba917cbd012e32386556d5c0fd87f73 and cb3c5445f88391625a7ef0f31e6c86743dd69877

closed 2016-06-04 by zador-blood-stained