armbian/build #282: Allwinner legacy backdoor?

ThomasKaiser, 2016-04-30:
Please have a look at http://irclog.whitequark.org/linux-sunxi/2016-04-29#16314390;

The file is still present in the newer sun8i BSP variant and I would assume sun7i and others are also affected?

Don't know how to deal with it.

ThomasKaiser, 2016-04-30:
Hmm... sun7i doesn't have it but sun8i is affected:

    tk@bananapim3:~$ id
    uid=1000(tk) gid=1000(tk) groups=1000(tk),20(dialout),27(sudo),29(audio),44(video),46(plugdev),108(netdev)
    tk@bananapim3:~$ echo "rootmydevice" > /proc/sunxi_debug/sunxi_debug 
    tk@bananapim3:~$ id
    uid=0(root) gid=0(root) groups=0(root),20(dialout),27(sudo),29(audio),44(video),46(plugdev),108(netdev),1000(tk)

zador-blood-stained, 2016-04-30:
I saw this on IRC yesterday. Even though this is more Local Privileges Escalation than a backdoor, IMO it should be disabled.

ThomasKaiser, 2016-04-30:
I agree that it should be disabled. Based on my understanding this privilege escalation combined with any other small bug might lead to an network enabled exploit (I would assume a php, nginx or apache process redirected to `/proc/sunxi_debug/sunxi_debug` could also become root this way?)

zador-blood-stained, 2016-04-30:
> I would assume a php, nginx or apache process redirected to /proc/sunxi_debug/sunxi_debug could also become root this way?

Yes, most likely.

Removing sunxi_debug.o from [this](https://github.com/igorpecovnik/linux/blob/sun8i/arch/arm/mach-sunxi/Makefile#L52) line (and deleting file `sunxi_debug.c`) may be the simplest solution if it doesn't break compilation.

ThomasKaiser, 2016-04-30:
Can you try it out? At least whether the build fails or not? My main build host is still busy doing other stuff and the 2nd host is also down (for yet unknown reasons -- too far away to look after it)

zador-blood-stained, 2016-04-30:
Build succeeds, so [this](https://github.com/zador-blood-stained/lib/blob/7a5aeda32d20a74b0ae5214828fe75e676cdf82b/patch/kernel/sun8i-default/remove-sunxi-debug.patch) should work

ThomasKaiser, 2016-04-30:
Could you also provide debs to be able to test? :-)

zador-blood-stained, 2016-04-30:
https://www.dropbox.com/sh/y8gudnbnpwm193w/AAB2BeRMosLIekOC1CPQU-a3a?dl=0

ThomasKaiser, 2016-04-30:
Fix confirmed to work by 'original submitter' KotCzarny himself :) http://irclog.whitequark.org/linux-sunxi/2016-04-30#16321288

So please push the fix. And I hope Igor checks download log and in case no one downloaded BPi M2+ 5.10 images we could simply re-release the M2+ image with fix included (otherwise we would've to increase version number already to 5.11 if I understand correctly?)

zador-blood-stained, 2016-04-30:
It's in my repository already, so it's a matter of merging my branch into this.

kotc, 2016-04-30:
@ThomasKaiser : happy to help, thanks for quick response!

ThomasKaiser, 2016-04-30:
@kotc: Thx for bringing this to our attention. Since we're rolling out a new major release this weekend this was almost perfectly timed. I also thought about fixing the issue for all of the many loboris images out there (already cloned his kernel repo since he didn't maintain it for maybe half a year) but thought again about. Users better switch to Armbian instead :)

igorpecovnik, 2016-04-30:
Bugs and problems usually arise when I do travel :) No downloads, so we can stay on 5.10 but I guess we will have an update to 5.11 in any case very soon since it's almost impossible to bring such update without any more or less serious problem. 

Merged, so closing.

closed 2016-04-30 by igorpecovnik