armbian/build #283: s500-bootloader.bin not initialized

vonfritz, 2016-05-01:
I know actions semi devices are WIP and the roseapple is disabled. Maybe you should disable the lemaker guitar to, i don't think that the build boots with this bootloader.

Greetings

igorpecovnik, 2016-05-01:
I was able to boot with it but I guess they have more versions of Guitar out there ...

vonfritz, 2016-05-01:
Well it should be indifferent which version, the bootloader  initializes the SoM and not the Baseboard. The SoM is always the same.
The S500-bootloader.bin in your repo has the same checksum as the  uninitialized bootloader.bin from 7 months ago (initial commit)
Can you please retry to boot with this bootloader when you have time?
it would be very strange if the board boots with this uninitialized bootloader.
just be curios.

Thanks
it is, however, the motherboard and always the same, and the bootloader initializes the one and not the basic card

Well the S500-bootloader.bin has the same checksum as the  uninitialized bootloader.bin from 7 months ago (initial commit)
Can you please retry to boot with this bootloader when you have time?
it would be very strange if the board boots with this uninitialized bootloader.
just be curios.

Thanks

vonfritz, 2016-05-01:
forgot, roseapple needs or has a different .ini as the guitar.
How do you think to manage this?
Two different bootloader or one unitialized and then via script chose the right .ini and then build the right bootloader?

igorpecovnik, 2016-05-01:
You are right. It does not boot, but I don't understand how it was possible to boot during my testing? I was playing with Guitar months ago and last week, Roseapple I made 1-2 tries and left since I have a lot of other things right now. I am interested that both boards are fixed and operational at basic level.

We have separate boot loader for virtually each board so this is not an issue. Where do I get a proper bootloader.bin for each boards?

vonfritz, 2016-05-01:
Well you can build one or try this one:
https://github.com/vonfritz/lib/commit/b3999aacb883349ae60f459f0dc0bad93241c3e9
The bootloader should work with all 5 base boards.
I don't have a guitar (board), test if it works. Maybe consider that you should rename it as it 's not usable for the roseapple.

igorpecovnik, 2016-05-02:
Thanks, will try to build later. Renaming and some script rework follows.

vonfritz, 2016-05-02:
Ok I think we can close it for now.
Will try to build a image for the roseapple and look if it works.

closed 2016-05-02 by vonfritz

igorpecovnik, 2016-05-02:
Also doing this right now and I am getting kernel panic at boot on Roseapple and I need to use their bootloader.

vonfritz, 2016-05-03:
Can you confirm that it boots properly on the guitar and that it has no issues?

igorpecovnik, 2016-05-03:
Yes, confirmed. My baseboard is revision B.