armbian/build #284: Armbian 5.10 higher temperature

bremme, 2016-05-02:
I just updated one of my Orange Pi One's (H3) to Armbian 5.10 this afternoon.  I only use this pi for a couple lightweight tasks, running an MQTT broker (mosquitto) and a NodeJS web app. I noticed that the idle temperature went from 39 C on 5.05 to 48 C on 5.10. I didn't check with the 5.05 version, but using 5.10, the cpu frequency is 648 Mhz when idle and the governor is set to interactive.

Its not a real problem, but just wondering what could cause this new behavior. Temperature measurement, cpu voltage, governor, frequency...?

If I can help by supplying additional information, let me know.

ThomasKaiser, 2016-05-02:
Can you please adjust `/etc/default/cpufrequtils` so that minimum cpufreq is 480000, reboot and report back?

bremme, 2016-05-02:
I adjusted the frequency and did a reboot. The temperature is still 48 C. I have a little heat sink on the H3. I cooled down the cpu by blowing some air over the heat sink till 38 C. But the temperature increased again to 48 C. The cpu use is 1% using armbianmonitor -m. It's just strange that the temperature is about 10 degree C higher, while I'm not running more stuff on the pi.

```shell
System load:   0.00            	Up time:       19 min		
Memory usage:  9 % of 494Mb  	IP:
CPU temp:      48°C           	
Usage of /:    10% of 15G 
```

```shell
20:13:22:  480MHz  0.00   1%   0%   0%   0%   0%   0%   48°C
20:13:27:  480MHz  0.00   1%   0%   0%   0%   0%   0%   47°C
20:13:32:  480MHz  0.00   1%   0%   0%   0%   0%   0%   47°C
20:13:37:  480MHz  0.00   1%   0%   0%   0%   0%   0%   47°C
20:13:42:  480MHz  0.00   1%   0%   0%   0%   0%   0%   47°C
20:13:47:  480MHz  0.00   1%   0%   0%   0%   0%   0%   47°C
```

```shell
$ cat /etc/default/cpufrequtils 
ENABLE=true
MIN_SPEED=480000
MAX_SPEED=1200000
GOVERNOR=interactive
```

ThomasKaiser, 2016-05-02:
Thx for the update. The difference in shown temperature might be u-boot/clock related. When we started supporting H3 back in March I did notice that the reported internal SoC temperatur was significantly lower compared to the same board running a loboris OS image with 100% identical dvfs settings.

So since u-boot is now also updated maybe that makes the difference. Would be worth the efforts to check just this (using 5.05 with older u-boot and then only exchanging u-boot) but my Orange Pi One is now still on a journey far far away ;)

ThomasKaiser, 2016-05-03:
Problem solved. I made a mistake when taking over FriendlyARM's dvfs settings. Currently Orange Pi One is running at 1.3GHz all the time, ARISC messages should be spit out when you do a `dmesg` and you'll need to adjust script.bin to get back old behaviour. Fix/explanation by @zador-blood-stained : http://forum.armbian.com/index.php/topic/724-quick-review-of-orange-pi-one/?view=getlastpost

bremme, 2016-05-03:
Thanks a lot! I applied the suggested fix and the temperature dropped back to 39 C at idle. If other people experience the same behavior on Armbian 5.10 using an Orange Pi One, here is the exact fix that I applied:

```shell
# backup script.bin just in case
$ sudo cp /boot/bin/orangepione.bin /boot/bin/orangepione.bin.backup
# convert bin to fex
$ sudo bin2fex /boot/bin/orangepione.bin /boot/bin/orangepione.fex
# edit fex file
$ sudo nano /boot/bin/orangepione.fex
# look for the dvfs_table section and change to:
[dvfs_table]
pmuic_type = 1
pmu_gpio0 = port:PL06<1><1><2><1>
pmu_level0 = 11300
pmu_level1 = 1100
max_freq = 1200000000
min_freq = 480000000
LV_count = 5
LV1_freq = 1200000000
LV1_volt = 1300
LV2_freq = 1008000000
LV2_volt = 1300
LV3_freq = 816000000
LV3_volt = 1100
LV4_freq = 648000000
LV4_volt = 1100
LV5_freq = 480000000
LV5_volt = 1100
# save and exit
# convert fex back to bin
$ sudo fex2bin /boot/bin/orangepione.fex /boot/bin/orangepione.bin
# if you haven't already done so, edit /etc/default/cpufrequtils
# make it look like this:
ENABLE=true
MIN_SPEED=480000
MAX_SPEED=1200000
GOVERNOR=interactive
# save and exit
# reboot pi
$ sudo reboot
```

closed 2016-05-03 by bremme

fandres, 2018-07-22:
Hello  @bremme, do you have any idea how to do this in Armbian 5.38? my Orange pi zero has a temprature of 70 °, these files do not exist.
or could you help me
Thank you

bremme, 2018-08-10:
@fandres I haven't played around with Armbian in a while. So I don't know the exact procedure. But I guess the .fex files should still be somewhere. So a similar procedure should apply to your problem as well. But I would be a little surprised if this bug would still be present in Armbian 5.38?