armbian/build #285: Armbian 5.10 keeps showing updates to install on login

bremme, 2016-05-02:
Just a minor issue, with perhaps an easy fix. When I login through ssh on my Orange Pi One it shows:

```shell
[ 3 updates to install: apt-get upgrade ]
```

It keeps displaying this after several `$ sudo apt-get update && sudo apt-get upgrade` 's and reboots. How can I fix this?

igorpecovnik, 2016-05-03:
This info is cached / refreshed on two days. Not the best implementation but your login is fast.
https://github.com/igorpecovnik/lib/blob/master/scripts/update-motd.d/40-updates#L131-L149

zador-blood-stained, 2016-06-30:
Update logic was [changed](https://github.com/igorpecovnik/lib/commit/3c147016ee6f5a67bb3ab0f2857470e1bb6e6b88), now it should work more reliably.

closed 2016-06-30 by zador-blood-stained