armbian/build #286: USB OTG does not work with mainline kernel on Lime2

chradev, 2016-05-02:
I am trying to use USB OTG on A20-Olinuxino-Lime2-eMMC with own build of Armbian up to 5.11 and mainline kernel up to 4.5.2 without success.

I have tested 
- both 'host' and 'otg' types for dr_mode in device tree;
- connect USB OTG to other board's USB Host;
- attach USB mass storage and other devices;
- attach other self powered devices;
- try to load modules like extcon_gpio, extcon_usb_gpio;
- rebuild kernel with CONFIG_USB_GPIO_VBUS=y
and many others without any success.

I have checked the HW is working fine as 'host' and 'device' with latest Olimex image with Debian Jessie and  kermel 3.4.103. A month or more ago I have also tested it to work with Armbian 5.07 and kernel 3.4.111 (modifying fex file). I have also verify that device tree seams to be o.k.

In the boot messages USB OTG is recognized by sunxi musb driver but usb0-vbus is finally disabled. There are few confusing messages as well.

`[    3.254996] reg-fixed-voltage usb0-vbus: could not find pctldev for node /soc@01c00000/pinctrl@01c20800/usb0_vbus_pin@0, deferring probe
[    3.255033] reg-fixed-voltage usb1-vbus: could not find pctldev for node /soc@01c00000/pinctrl@01c20800/usb1_vbus_pin@0, deferring probe
[    3.255067] reg-fixed-voltage usb2-vbus: could not find pctldev for node /soc@01c00000/pinctrl@01c20800/usb2_vbus_pin@0, deferring probe
[    3.257948] usbcore: registered new interface driver usbfs
[    3.258023] usbcore: registered new interface driver hub
[    3.258127] usbcore: registered new device driver usb
[    3.369469] sun4i-usb-phy 1c13400.phy: could not find pctldev for node /soc@01c00000/pinctrl@01c20800/usb0_id_detect_pin@0, deferring probe

...

[    3.846532] usb_phy_generic.0.auto supply vcc not found, using dummy regulator
[    3.847087] musb-hdrc musb-hdrc.1.auto: MUSB HDRC host driver
[    3.847109] musb-hdrc musb-hdrc.1.auto: new USB bus registered, assigned bus number 5
[    3.847554] usb usb5: New USB device found, idVendor=1d6b, idProduct=0002
[    3.847569] usb usb5: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    3.847579] usb usb5: Product: MUSB HDRC host driver
[    3.847588] usb usb5: Manufacturer: Linux 4.5.2-sunxi musb-hcd
[    3.847598] usb usb5: SerialNumber: musb-hdrc.1.auto
[    3.848439] hub 5-0:1.0: USB hub found
[    3.848513] hub 5-0:1.0: 1 port detected
[    3.877070] usb0-vbus: disabling`

igorpecovnik, 2016-05-03:
I thought this was solved. It would be best to address the one who is working on this implementation - I am out if ideas. @Hans de Goede https://groups.google.com/forum/#!forum/linux-sunxi

chradev, 2016-05-16:
Hi Igor,

Unfortunately, I did not find the solution of the problem.

Thanks to a help from sunxi form  I find a solution to make 
[USB OTG to work in Host only mode](https://groups.google.com/d/msg/linux-sunxi/uulK5lf7XrU/QUMtNbVpAQAJ).

Fortunately, it is enough for me as a beginning but any way it is semi-solution.
So any ideas are welcome.

Best regards
Chris

zador-blood-stained, 2016-12-28:
Unfortunately don't think anything can be done here (by the Armbian team) without actual hardware, so closing this.

closed 2016-12-28 by zador-blood-stained