armbian/build #288: Occasional Freezes observed for OrangePI Plus 2 Running 5.10 on EMMC

avinashga23, 2016-05-04:
Hi,

From past 2 days i have observed occasional freezes on my orange pi plus 2 running 5.10 on built in EMMC. When this happens the system is unreachable to ping, if already in SSH session the terminal stops responding to the key strokes, sometimes this comes out of the freezes. First i thought it can be the network issue, later i have observed this even after keeping the board within 2 ft of my wifi router.

Regards,
--
Avinash

ThomasKaiser, 2016-05-05:
Until now I do not see any evidence of freezes since you seem to report WiFi connection losses for whatever reasons.

Please think about adding `while true ; do date >/var/log/heartbeat.log ; sync ; sleep 60 ; done` prior to `exit 0` in `/etc/rc.local` to further nail the problem down.

avinashga23, 2016-05-06:
I guess its the network issue. I have not observed this from past 24 hrs.

closed 2016-05-06 by avinashga23