armbian/build #290: H3 | Hard power cycle = wont boot

Fourdee, 2016-05-04:
Hi Guys,

I'am attempting to create a new DietPi image built with ARMbian build tools. This is in relation to the recent debug exploit code you found in the 3.4 kernel:  https://github.com/Fourdee/DietPi/issues/312

#### The issue
After the initial ARMbian setup, hard power cycle = no boot

#### After writing the ARMbian image I compiled, all I do is:
- Copy dietpi scripts to ```/boot/``` (```dietpi.txt``` and folder ```dietpi``` from this page https://github.com/Fourdee/DietPi)
- Poweron

After going through the initial ARMbian setup and creating an account (dietpi):
```
# create user dietpi
passwd #set to dietpi as root@
userdel -f dietpi
cat << _EOF_ > /etc/apt/sources.list
deb http://ftp.debian.org/debian jessie main contrib non-free
deb http://ftp.debian.org/debian jessie-updates main contrib non-free
deb http://security.debian.org jessie/updates main contrib non-free
deb http://ftp.debian.org/debian jessie-backports main contrib non-free
_EOF_
apt-get clean
apt-get update
 ```
Soft reboots are fine. Hard power cycle results in no boot (green led on device). I've also tested with writing the image using ```dd``` and win32diskimage, same issue.

#### Build notes
Server used to build:
- VM x64 14.04.4 LTS
```
root@vm-crosscomp:~# uname -a
Linux vm-crosscomp 3.13.0-85-generic #129-Ubuntu SMP Thu Mar 17 20:50:15 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

 The ARMbian build tools completed successfully with the following settings:
- Full image build
- Distro = Jessie
- Server only (no desktop)

Image generated = Armbian_5.11_Orangepih3_Debian_jessie_3.4.112.raw: http://dietpi.com/downloads/testing/Armbian_5.11_Orangepih3_Debian_jessie_3.4.112.zip

compile.sh: http://dietpi.com/downloads/testing/compile.sh

lib/configuration.sh: http://dietpi.com/downloads/testing/configuration.sh

Any ideas what might be causing the issue? Maybe my OPi PC is at fault?

igorpecovnik, 2016-05-05:
You mean this:?
http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=1339&highlight=power

Fourdee, 2016-05-05:
@igorpecovnik 
I don't believe so. This issue in no relation to the power button (i never use it).

In short:
- Power on device, do stuff, reboot multiple times = working fine
- Shutdown system, unplug power, plug power back in = won't boot (solid green light)

When shutting down the system, the flashing leds (Red Green) change to solid Green. The Green never turns off.

igorpecovnik, 2016-05-05:
It must boot when you put power plug back. Connect serial console and see what's happening ... if you have some strange device on USB, u-boot might break.

Fourdee, 2016-05-06:
@igorpecovnik 

Found the cause:
Failed ```firstrun``` service with function ```do_expand_rootfs``` corrupts file system on a 2 partition ARMbian build tool generated image.

Error codes and tests:
https://github.com/Fourdee/DietPi/issues/312#issuecomment-217420990

igorpecovnik, 2016-05-06:
Than we have a problem, thanks. Will be fixed asap.

zador-blood-stained, 2016-06-05:
I think this is fixed now. Feel free to reopen if there are issues with current build script version.

closed 2016-06-05 by zador-blood-stained