armbian/build #291: Orangepih3 (Orange pi pc)

faddat, 2016-05-06:
Seems a regression from before the refactor-- I have lost the ethernet port when building "dev" + Debian 7/8 & Ubuntu 14.04/16.04.

Anyone else?

ThomasKaiser, 2016-05-06:
Me2. And at that moment I gave up a few weeks ago. The whole stuff needs a rework and we should then also try to include '@megous' THS patches](https://github.com/megous/linux/tree/orange-pi) to get thermal readouts and throttling.

faddat, 2016-05-06:
Alright.  I'm going to try getting my hands dirty inside of lib here.  I'll let you know how things progress.

faddat, 2016-05-06:
And if we are starting from the top for the build sequence for orangepi, should I maybe do something like:

- [ ] Determine Build Steps
 - [ ] Build u-boot 
 - [ ] Test u-boot
 - [ ] Build kernel
 - [ ] Get Rootfs
 - [ ] Make an image file
 - [ ] Test image file on board
- [ ] Integrate into lib
 - [ ] Orient myself in the code
 - [ ] Implement the sequence discovered above

ThomasKaiser, 2016-05-06:
I would believe u-boot patches are still fine. I would disable [0000-sun8i-rgmii-ethernet.patch](https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sun8i-dev/0000-sun8i-rgmii-ethernet.patch) and use megous' repo as base so that sun8i.conf reads

	LINUXKERNEL='https://github.com/megous/linux'
	KERNELBRANCH='orange-pi'
	LINUXSOURCE='linux-sun8i-mainline'

Maybe that works already. When I tested back then my OPi PC didn't boot any more. And maybe that was related to different voltage definitions in u-boot. Please compare [his settings](https://github.com/megous/u-boot/commit/e43b43009667d48becd97d1bd43dfa23c7646ed4) with [mine](https://github.com/igorpecovnik/lib/blob/master/patch/u-boot/u-boot-dev/u-boot-99-opi-change-build-settings.patch#L11-L12). But maybe that wasn't the reason. I simply gave up a few weeks ago since it got too frustrating.

faddat, 2016-05-06:
Kernel configuration to the nth degree, coupled with all sorts of crazy
patching?  Frustrating?  I have no idea what you're talking about.....
;).

Will report back.....

Jacob Gadikian
E-mail: faddat@gmail.com
SKYPE: faddat
Phone/SMS: +84 167 789 6421

faddat, 2016-05-06:
Okay, so no progress with the settings you suggested, currently trying sunxi-wip and sunxi-next.  

We'll see how it goes :).

faddat, 2016-05-09:
Got no love yesterday, kept pushing through and got boardlove today:

**u-boot** u-boot-sunxi
**Linux:**  Linux-mainline + linux-next
**Ethernet patches** at http://sunxi.montjoie.ovh/ethernet/
**rootfs** Ubuntu 16.10 

..... and we have a winnner!  

BUT

otoh, I have NO idea how to get this into armbian.  Also I guess it's going to take me at least a few more hours to have this in a place where I'd feel comfortable with anyone else using it..... if you follow these directions, it may cause your boards, house, wife, husband, and/or pants to spontaneously combust.  

You've been warned.

ThomasKaiser, 2016-05-13:
Just a note. Wens updated the EPHY/EMAC stuff in his branch an hour ago: https://github.com/wens/linux/commit/299a7d8b7b45a3d265655fb3ea0dd827f1295811

Unfortunately neither time to try it out nor a build host available (still busy doing other stuff)

faddat, 2016-05-13:
No worries, I'll try testing it out, and I'll update lib if it works out.  Thanks for keeping an eye out!

kometchtech, 2016-05-13:
I compile a dev in Ubuntu 16.04 Dcoker on error has occurred.

```
  x   LD      drivers/media/built-in.o                                                                x  
  x Makefile:962: recipe for target 'drivers' failed                                                  x  
  x make: *** [drivers] Error 2                                                                       x  
  x   LD      net/built-in.o                                                                          x  
  
[ error ] ERROR in function compile_kernel [ common.sh:183 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated 
```

faddat, 2016-05-13:
I got the same error.  Currently trying orangepih3 instead, that is building alright.

zador-blood-stained, 2016-05-13:
> rootfs Ubuntu 16.10

Well, I can imagine that in October of 2016 H3 Ethernet would work :smile: 

> I got the same error. Currently trying orangepih3 instead, that is building alright.

I had to disable both patches for `sun8i-dev` to compile wens `h3-emac` branch.

Results on Orange Pi One:
Working Ethernet @ first boot, no Ethernet @ second boot. Switching from default orange pi pc dtb to orange pi one doesn't work - opi one DTB appears to be broken.

faddat, 2016-05-13:
That's really strange with the switch from working to not on 1st/2nd boot.  Any idea what might have caused that?  I can't really see how the disk expansion would do it.... ? *puzzling*.

zador-blood-stained, 2016-05-13:
> I can't really see how the disk expansion would do it.... ? puzzling.

I feel like it's more about cold/warm boot and improper emac reset or something like this...

Another thing - changing `allow-hotplug eth0` (that should be the default IMO) to `auto eth0`, and after resetting power I'm stuck at "Raising network interfaces" :disappointed:  

Right now I want to try [this](https://github.com/montjoie/linux/tree/sun8i-emac-wip) branch, [referenced](https://linux-sunxi.org/Sun8i_emac) on linux-sunxi wiki, in case it works better.

faddat, 2016-05-13:
Just wrote the image file produced from lib:master for orangepih3 to an sdcard, 

Regarding Montjoie-linux, I think it depends what version of the kernel you run it against, and AFAIK, montjoie-linux is linux + these patches.  

So basically, I did this already, and wish to warn you that I think you're most likely to end up with something that has a blinky ethernet light, but no ethernet :/.

faddat, 2016-05-13:
[image]

it built alright, but it died at that screen.....

zador-blood-stained, 2016-05-13:
> So basically, I did this already, and wish to warn you that I think you're most likely to end up with something that has a blinky ethernet light, but no ethernet :/.

Well, `dmesg` output with different DTBs still may provide some useful details for me.

> it built alright, but it died at that screen.....

but this is u-boot (still), it wasn't built with *libtelepathy* support, it can't die because it's *about to load* non-working kernel.

faddat, 2016-05-13:
The kernel worked.  u-boot choked on the USB keyboard I had plugged in.

faddat, 2016-05-13:
I know that they are Device Tree Blobs (or binaries) and that they do driver stuff.  I know that they can be baked into the kernel zImage-- and from there it's fuzzy.

zador-blood-stained, 2016-05-13:
`.dts`/`.dtb` files have essentially the same purpose as `.fex`/`.bin` for legacy kernel - describing hardware configuration for the kernel, only instead on plain INI-style syntax you have tree-like structure with more nesting levels, ability to include files and extend and override properties.

Appending DTB to kernel image is legacy for dumb u-boot versions and is not used in Armbian.
```
config ARM_APPENDED_DTB
	bool "Use appended device tree blob to zImage (EXPERIMENTAL)"
	depends on OF
	help
	  With this option, the boot code will look for a device tree binary
	  (DTB) appended to zImage
	  (e.g. cat zImage <filename>.dtb > zImage_w_dtb).

	  This is meant as a backward compatibility convenience for those
	  systems with a bootloader that can't be upgraded to accommodate
	  the documented boot protocol using a device tree.

	  Beware that there is very little in terms of protection against
	  this option being confused by leftover garbage in memory that might
	  look like a DTB header after a reboot if no actual DTB is appended
	  to zImage.  Do not leave this option active in a production kernel
	  if you don't intend to always append a DTB.  Proper passing of the
	  location into r2 of a bootloader provided DTB is always preferable
	  to this option.
```

faddat, 2016-05-13:
Can you tell me how you got the kernel to cough up all that context?

Thanks!

Jacob Gadikian
E-mail: faddat@gmail.com
SKYPE: faddat
Phone/SMS: +84 167 789 6421

zador-blood-stained, 2016-05-13:
You mean description of kernel options? Easy: https://github.com/torvalds/linux/blob/master/arch/arm/Kconfig#L1840-L1858

In other words - just grep through Kconfig files in kernel source tree when you know exact option name (without `CONFIG_` prefix) or some words in its description.

faddat, 2016-05-13:
Thanks!

-jake

Jacob Gadikian
E-mail: faddat@gmail.com
SKYPE: faddat
Phone/SMS: +84 167 789 6421

zador-blood-stained, 2016-05-13:
I remembered something I read ~1 month ago and decided to try. So I connected my Orange Pi to 100MBit router instead of 1GBit switch. Result - working Ethernet :smile:

faddat, 2016-05-13:
??!!¿?

Using dev/h3 or using dev/orangepi-pc?

Awesome though!

zador-blood-stained, 2016-05-13:
I was using orangepione target and wens' branch, but it doesn't matter that much since in any case orange pi pc DTB is used.

Small update: It looks like emac fails to reset at boot if plugged into 1GBit switch, after it got properly initialized, it will work fine if you connect it to 1G switch.

faddat, 2016-05-13:
Let me see if I've got it straight:

Dtb - determines the drivers that the kernel uses by describing the
hardware.

zador-blood-stained, 2016-05-14:
DT determines what drivers should kernel use for each device and it gives drivers device-specific data (and DT also provides some other configuration details).

faddat, 2016-05-14:
@igorpecovnik,  

       mv *.deb $DEST/debs/ || exit_with_error "Failed moving kernel DEBs"

```
       eval ${KERNEL_TOOLCHAIN:+env PATH=$KERNEL_TOOLCHAIN:$PATH} 'make -j1 $KERNEL_PACKING KDEB_PKGVERSION=$REVISION LOCALVERSION="-"$LINUXFAMILY \
                KBUILD_DEBARCH=$ARCH ARCH=$ARCHITECTURE DEBFULLNAME="$MAINTAINER" DEBEMAIL="$MAINTAINERMAIL" CROSS_COMPILE="$CCACHE $KERNEL_COMPILER" 2>&1 ' \
                ${PROGRESS_LOG_TO_FILE:+' | tee -a $DEST/debug/compilation.log'} \
                ${OUTPUT_DIALOG:+' | dialog --backtitle "$backtitle" --progressbox "Creating kernel packages..." $TTY_Y $TTY_X'} \
                ${OUTPUT_VERYSILENT:+' >/dev/null 2>/dev/null'}

        cd ..
        mv *.deb $DEST/debs/ || exit_with_error "Failed moving kernel DEBs"
}

```

Does || mean bash for "[if this fails?](http://www.tldp.org/LDP/abs/html/testconstructs.html#TESTCONSTRUCTS1)"  That is where my build (orangepi_pc / dev kernel / xenial) dies on me.  

Has anyone at all tried lib with distcc?  That could be the issue here, I put some distcc into my bashrc.  I noticed that lib is using linaro 5.3 now, so my next step will be to try:

1) no distcc
...and if that works
2) w/ linaro on the other 4 boxes.  

Thanks armbianers!

zador-blood-stained, 2016-05-14:
> Does || mean bash for "if this fails?" 

Yes.

> Has anyone at all tried lib with distcc?

No, but you should probably read ccache manual - it requires specific configuration to work. Also "some distcc" may not affect cross-compilation at all.

faddat, 2016-05-18:
Closing because I think it's kinda-resolved.

closed 2016-05-18 by faddat