armbian/build #295: Lamobo R1 fails to boot with premade jessie/vanilla

faddat, 2016-05-10:
I recently got a serial debug usb cable, and this is the reason for most of my recent winnage with opi.  Here's the debug output:

```
U-Boot SPL 2016.01-armbian-s500 (Apr 29 2016 - 05:01:49)
DRAM: 1024 MiB
CPU: 912000000Hz, AXI/AHB/APB: 3/2/2
Trying to boot from MMC

U-Boot 2016.01-armbian-s500 (Apr 29 2016 - 05:01:49 +0200) Allwinner Technology

CPU:   Allwinner A20 (SUN7I)
I2C:   ready
DRAM:  1 GiB
MMC:   SUNXI SD/MMC: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
SCSI:  SUNXI SCSI INIT
SATA link 0 timeout.
AHCI 0001.0100 32 slots 1 ports 3 Gbps 0x1 impl SATA mode
flags: ncq stag pm led clo only pmp pio slum part ccc apst 
Net:   eth0: ethernet@01c50000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
USB2:   USB EHCI 1.00
USB3:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
scanning bus 2 for devices... 2 USB Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
2426 bytes read in 212 ms (10.7 KiB/s)
## Executing script at 43100000
0 bytes read in 151 ms (0 Bytes/s)
0 bytes read in 161 ms (0 Bytes/s)
29089 bytes read in 801 ms (35.2 KiB/s)
4017323 bytes read in 413 ms (9.3 MiB/s)
6047000 bytes read in 598 ms (9.6 MiB/s)
Kernel image @ 0x46000000 [ 0x000000 - 0x5c4518 ]
## Loading init Ramdisk from Legacy Image at 42000000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    4017259 Bytes = 3.8 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 49000000
   Booting using the fdt blob at 0x49000000
   Loading Ramdisk to 49c2b000, end 49fffc6b ... OK
   Loading Device Tree to 49c20000, end 49c2a1a0 ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.

U-Boot SPL 2016.01-armbian-s500 (Apr 29 2016 - 05:01:49)
DRAM: 1024 MiB
CPU: 912000000Hz, AXI/AHB/APB: 3/2/2
Trying to boot from MMC

U-Boot 2016.01-armbian-s500 (Apr 29 2016 - 05:01:49 +0200) Allwinner Technology

CPU:   Allwinner A20 (SUN7I)
I2C:   ready
DRAM:  1 GiB
MMC:   SUNXI SD/MMC: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
SCSI:  SUNXI SCSI INIT
SATA link 0 timeout.
AHCI 0001.0100 32 slots 1 ports 3 Gbps 0x1 impl SATA mode
flags: ncq stag pm led clo only pmp pio slum part ccc apst 
Net:   eth0: ethernet@01c50000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
USB2:   USB EHCI 1.00
USB3:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
scanning bus 2 for devices... 2 USB Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
2426 bytes read in 211 ms (10.7 KiB/s)
## Executing script at 43100000
183 bytes read in 156 ms (1000 Bytes/s)
0 bytes read in 161 ms (0 Bytes/s)
29089 bytes read in 801 ms (35.2 KiB/s)
4017323 bytes read in 412 ms (9.3 MiB/s)
6047000 bytes read in 597 ms (9.7 MiB/s)
Kernel image @ 0x46000000 [ 0x000000 - 0x5c4518 ]
## Loading init Ramdisk from Legacy Image at 42000000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    4017259 Bytes = 3.8 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 49000000
   Booting using the fdt blob at 0x49000000
   Loading Ramdisk to 49c2b000, end 49fffc6b ... OK
   Loading Device Tree to 49c20000, end 49c2a1a0 ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.

```

It just hangs there.  I'm going to try it out with the latest lib and I'll check back in.

faddat, 2016-05-10:
2016.01-armbian-s500  

Does that s500 refer to the s500 chipset?  I was having a peruse of the log just now and noticed that....

zador-blood-stained, 2016-05-10:
Did you change boot.cmd/boot.scr parameter "console=tty0" to "console=ttyS0,115200"

> Does that s500 refer to the s500 chipset? I was having a peruse of the log just now and noticed that....

There were similar issues with u-boot version naming, though actual u-boot configuration should be right.

faddat, 2016-05-10:
@zador-blood-stained I didn't make any changes to boot.scr-- should I or..... ?

closed 2016-05-10 by faddat

zador-blood-stained, 2016-05-10:
Yes, you should change "console" parameter - otherwise you won't get any kernel messages on serial console. Don't forget to recompile `boot.cmd` to `boot.scr`.