armbian/build #296: Odroid C2 compile error

kometchtech, 2016-05-13:
When using the Armbian building script and trying to create the image of the Odroid C2 errors.

```
[ error ] ERROR in function compile_uboot [ common.sh:89 ]
[ error ] U-boot file not found [ u-boot.bin ]
```

igorpecovnik, 2016-05-13:
https://github.com/igorpecovnik/lib/commit/5bb94da344393a0319338b02ff5ec0bde20cdad5

works now

closed 2016-05-13 by igorpecovnik

kometchtech, 2016-05-13:
Tried again, error is.

```
../include/linux/compiler-gcc.h:114:30: fatal error: linux/compiler-gcc5.h: No such file or directory
make[3]: *** [include/autoconf.mk] Error 1
../scripts/Makefile.autoconf:64: recipe for target 'include/autoconf.m
make[2]: *** [silentoldconfig] Error 1
../Makefile:481: recipe for target 'silentoldconfig' failed
make[1]: *** No rule to make target 'include/config/auto.conf', needed by 'include/config/uboot.release'.  Stop.
make: *** [sub-make] Error 2
Makefile:147: recipe for target 'sub-make' failed
```

igorpecovnik, 2016-05-13:
Remove sources and try again. I just did and working fine ... 

What is your host system?

kometchtech, 2016-05-13:
It is ubuntu16.04 on VirtualBox.

igorpecovnik, 2016-05-13:
Ubuntu 16.04 is not yet 100% reliable and well tested - I am still mostly working with 14.04.

kometchtech, 2016-05-13:
Try and try again with OK Ubuntu 14.04.

kometchtech, 2016-05-13:
tried again in Ubuntu 14.04, but become a similar error.
```
[ error ] ERROR in function compile_uboot [ common.sh:89 ]
[ error ] U-boot file not found [ u-boot.bin ]
[ o.k. ] Process terminated
```

igorpecovnik, 2016-05-13:
I guess, now it should work on Ubuntu 16.04 too
https://github.com/igorpecovnik/lib/commit/feaf8901f757bcb8769d1a2ea8a132e17fff92fe

zador-blood-stained, 2016-05-13:
> I guess, now it should work on Ubuntu 16.04 too

Tested and works, if you have 4.9 linaro aarch64 toolchain (i.e. `gcc-linaro-4.9-2016.02-x86_64_aarch64-linux-gnu`) in `$SRC/toolchains`

kometchtech, 2016-05-13:
I tried testing with Ubuntu 16.04.
Now follows the intended results?
```
[ o.k. ] Cleaning [ u-boot-odroidc2/odroidc2-v2015.01 ]
[ o.k. ] Cleaning [ linux-odroidc2/odroidc2-3.14.y ]
[ error ] ERROR in function source [ main.sh:211 ]
[ error ] Could not find required u-boot toolchain [ < 5.0 ]
[ o.k. ] Process terminated 
```

zador-blood-stained, 2016-05-13:
> Now follows the intended results?

Yes, intended. U-boot won't compile with newer toolchains (5.x), so now you get informative message instead of compilation error. You can download [this](https://releases.linaro.org/components/toolchain/binaries/4.9-2016.02/aarch64-linux-gnu/gcc-linaro-4.9-2016.02-x86_64_aarch64-linux-gnu.tar.xz) toolchain and unpack it to `toolchains` directory near `lib`, `output` and `sources`, it should allow you to compile C2 u-boot in Ubuntu 16.04.

kometchtech, 2016-05-13:
Of course.