armbian/build #299: extras/mt7601.sh fails: repository no longer exists

bit, 2016-05-15:
extras/mt7601.sh tries to clone https://github.com/porjo/mt7601/ which fails asking for a password.

closed 2016-05-15 by igorpecovnik

flexiti, 2016-05-19:
WiFi USB Mediatek mt7601 will not work now ? ( I just try connect it do OrangePi PC)

tusharvb19, 2017-08-18:
Need help!
got this while following
https://forum.armbian.com/index.php?/topic/1819-solved-orange-pi-pc-and-2-ralink-mt7601u-dongle-usb-id-148f7601/

root@ILA:~/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913# make
make -C tools
make[1]: Entering directory '/root/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/tools'
gcc -g bin2h.c -o bin2h
make[1]: Leaving directory '/root/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/tools'
/root/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/tools/bin2h
cp -f os/linux/Makefile.6 /root/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/os/linux/                                                                                         Makefile
make -C /lib/modules/3.4.113-sun8i/build SUBDIRS=/root/DPO_MT7601U_LinuxSTA_3.0                                                                                         .0.4_20130913/os/linux modules
make[1]: *** /lib/modules/3.4.113-sun8i/build: No such file or directory.  Stop                                                                                         .
Makefile:394: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2
root@ILAN:~/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913# make
make -C tools
make[1]: Entering directory '/root/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/tools'
gcc -g bin2h.c -o bin2h
make[1]: Leaving directory '/root/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/tools'
/root/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/tools/bin2h
cp -f os/linux/Makefile.6 /root/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913/os/linux/                                                                                         Makefile
make -C /lib/modules/3.4.113-sun8i/build SUBDIRS=/root/DPO_MT7601U_LinuxSTA_3.0                                                                                         .0.4_20130913/os/linux modules
make[1]: *** /lib/modules/3.4.113-sun8i/build: No such file or directory.  Stop                                                                                         .
Makefile:394: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2
root@ILA:~/DPO_MT7601U_LinuxSTA_3.0.0.4_20130913#

GusAntoniassi, 2018-04-14:
Got this problem too when trying to compile ADS7846 driver for Orange Pi PC.
What solved for me was:
```bash
$~ sudo apt-get update
$~ sudo apt-get -y upgrade
$~ sudo apt-get install linux-headers-$(uname -r)
```
In my case this was a fresh install of Armbian, so the packages were probably outdated.