armbian/build #306: OrangePi+ u-boot compile "Duplicate node name"

kotrs, 2016-05-17:
Hello, I am not able to compile u-boot for OrangePi+ ... The compilation finishes with these errrors

(duplicate_node_names): Duplicate node name /clocks/apb0_clk───────────────────────────────────┘  
      UPD     include/generated/timestamp_autogenerated.h   ERROR (duplicate_node_names): Duplicate node name /clocks/clk@01f01428
                              ERROR (duplicate_node_names): Duplicate node name /soc/reset@01f014b0
                                                                                                   ERROR (duplicate_node_names): Duplicate node name /soc/pinctrl@01f02c00
                                                                      ERROR: Input tree has errors, aborting (use -f to force output)
                                 make[2]: *** [arch/arm/dts/sun8i-h3-orangepi-2.dtb] Error 2
                                                                                            make[1]: *** [arch-dtbs] Error 2
                        make: *** [dts/dt.dtb] Error 2
                                                      make: *** Waiting for unfinished jobs....
[ error ] ERROR in function compile_uboot [ common.sh:89 ]
[ error ] U-boot file not found [ u-boot-sunxi-with-spl.bin ]
[ o.k. ] Process terminated

igorpecovnik, 2016-05-17:
Yes, I see. We have new u-boot and patches must be adjusted.

igorpecovnik, 2016-05-17:
It's already fixed.

https://github.com/zador-blood-stained/lib/commit/5b05e265720918c8af1ea8ee1cba0660ec8cbdd3

closed 2016-05-17 by igorpecovnik