armbian/build #307: Different motd displayed on orange Pi PC

kometchtech, 2016-05-18:
To build in Orange Pi PC dev and motd Pi M2 + banana and they listed.
```
 ____                                  ____  _   __  __ ____       
| __ )  __ _ _ __   __ _ _ __   __ _  |  _ \(_) |  \/  |___ \  _   
|  _ \ / _` | '_ \ / _` | '_ \ / _` | | |_) | | | |\/| | __) || |_ 
| |_) | (_| | | | | (_| | | | | (_| | |  __/| | | |  | |/ __/_   _|
|____/ \__,_|_| |_|\__,_|_| |_|\__,_| |_|   |_| |_|  |_|_____||_|  
                                                                   

Welcome to ARMBIAN Debian GNU/Linux 8 (jessie) 4.6.0-sun8i 

System load:   0.05             Up time:       32 sec    
Memory usage:  3 % of 1002Mb    IP:            192.168.122.178
Usage of /:    8% of 15G     
```

igorpecovnik, 2016-05-18:
This auto recognition is somehow fragile and we are changing it.

igorpecovnik, 2016-06-02:
Auto recognition is disabled on new images.

closed 2016-06-02 by igorpecovnik