armbian/build #309: Orangepipc default desktop build fails

engshien, 2016-05-18:
Build of Orangepipc default jessie desktop fails (console build succeeds):

[Build host running Ubuntu 16.04 Xenial]

```
...
[ o.k. ] Creating board support package [ orangepipc ]
[ o.k. ] Fingerprinting [ Armbian 5.11 Orangepipc Debian jessie default Linux ]
[ o.k. ] Building package [ linux-jessie-root-orangepipc ]
[ o.k. ] Starting build process for [ orangepipc jessie ]
[ o.k. ] Extracting jessie-ng-armhf.e50...317.tgz [ 0 days old ]
jessie-ng-armhf.e50...317.tgz: 45.6MiB [ 237MiB/s] [======================================================================================================================================>] 100%
mount: mount point /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/proc does not exist
mount: mount point /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/sys does not exist
mount: mount point /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/dev does not exist
mount: mount point /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/dev does not exist
mount: mount point /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/dev/pts does not exist
[ o.k. ] Applying distribution specific tweaks for [ jessie ]
/home/ewu/dev/armbian/igorpecovnik/lib/distributions.sh: line 23: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/apt/apt.conf.d/71-no-recommends: No such file or directory
cp: cannot create regular file '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/apt/apt.conf.d/50unattended-upgrades': No such file or directory
cp: cannot create regular file '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/apt/apt.conf.d/02periodic': No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/ssh/sshd_config: No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/apt/apt.conf.d/50unattended-upgrades: No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/apt/apt.conf.d/50unattended-upgrades: No such file or directory
cp: cannot create regular file '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/init/ttyS0.conf': No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/init/ttyS0.conf: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/default/console-setup: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
touch: cannot touch '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/root/.not_logged_in_yet': No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
cp: cannot create regular file '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/tmp/': Not a directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ o.k. ] Installing packages [ linux-image-sun8i ]
mount: mount point /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/tmp does not exist
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ o.k. ] Installing u-boot [ linux-u-boot-orangepipc ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ o.k. ] Installing headers [ linux-headers-sun8i ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
cp: cannot create regular file '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/boot/boot.bmp': No such file or directory
checkdir:  cannot create extraction directory: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/lib/firmware
           No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
cp: cannot create regular file '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/lirc/lircd.conf': No such file or directory
[ o.k. ] Installing board support package [ orangepipc ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ o.k. ] Creating boot scripts [ orangepipc ]
cp: cannot create regular file '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/boot/boot.cmd': No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/boot/boot.cmd: No such file or directory
/home/ewu/dev/armbian/igorpecovnik/lib/boards.sh: line 78: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/default/cpufrequtils: No such file or directory
/home/ewu/dev/armbian/igorpecovnik/lib/boards.sh: line 79: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/default/cpufrequtils: No such file or directory
/home/ewu/dev/armbian/igorpecovnik/lib/boards.sh: line 80: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/default/cpufrequtils: No such file or directory
/home/ewu/dev/armbian/igorpecovnik/lib/boards.sh: line 81: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/default/cpufrequtils: No such file or directory
rm: cannot remove '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/resolv.conf': No such file or directory
cp: target '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/network/' is not a directory
ln: failed to create symbolic link '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/network/interfaces': No such file or directory
umount: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/tmp: mountpoint not found
[ o.k. ] Installing desktop [ XFCE ]
mount: mount point /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/tmp does not exist
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
/home/ewu/dev/armbian/igorpecovnik/lib/desktop.sh: line 71: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/wicd/manager-settings.conf: No such file or directory
/home/ewu/dev/armbian/igorpecovnik/lib/desktop.sh: line 72: /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/wicd/manager-settings.conf: No such file or directory
sed: can't read /home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/default/nodm: No such file or directory
cp: cannot create regular file '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/etc/X11/xorg.conf': No such file or directory
cp: cannot create directory '/home/ewu/dev/armbian/igorpecovnik/output/cache/sdcard/tmp/debs': No such file or directory
[ o.k. ] Installing desktop-extras for sunxi [ sunxi ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ error ] Installation failed [ desktop-extras for sunxi ]
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:529 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

zador-blood-stained, 2016-05-18:
> jessie-ng-armhf.e50...317.tgz: 45.6MiB

Cache file size is too small - probably there were errors at previous image build attempt. Please delete it manually from `output/cache/rootfs` (or adding "cache" to `CLEAN_LEVEL` in compile.sh) and try again.

engshien, 2016-05-19:
Thank you. Good to know if I see this again.

closed 2016-05-19 by engshien