armbian/build #311: Mediatek Mt7601 usb wifi

flexiti, 2016-05-19:
Help please, 
How install this device driver now ? ( previous source not existing )

flexiti, 2016-05-19:
Orange PI PC, Jessie desktop
I downloaded the previous, ready to use image from [http://www.armbian.com/orange-pi-pc/] 
and I try  as described here :   [https://github.com/igorpecovnik/lib/blob/master/documentation/user-faq.md#how-to-build-a-wireless-driver-]
but I have no success  (errors below) . 
When I try to build from current sources, there is no sources of the MT7601 driver.
 
---------------------------------------------

> root@orangepipc:/usr/src/linux-headers-3.4.112-sun8i# make scripts
> scripts/kconfig/conf --silentoldconfig Kconfig
> fs/Kconfig:287: can't open file "fs/fs_compat.kconfig"
> /usr/src/linux-headers-3.4.112-sun8i/scripts/kconfig/Makefile:33: recipe for target 'silentoldconfig' failed
> make[2]: *** [silentoldconfig] Error 1
> /usr/src/linux-headers-3.4.112-sun8i/Makefile:493: recipe for target 'silentoldconfig' failed
> make[1]: *** [silentoldconfig] Error 2
> make: *** No rule to make target 'include/config/auto.conf', needed by 'scripts'.  Stop.
> root@orangepipc:/usr/src/linux-headers-3.4.112-sun8i#
> root@orangepipc:/usr/src# cd mt7601-3.0.0.4/
> root@orangepipc:/usr/src/mt7601-3.0.0.4# make ARCH=arm
> make -C tools
> make[1]: Entering directory '/usr/src/mt7601-3.0.0.4/tools'
> gcc -g bin2h.c -o bin2h
> /usr/bin/ld: cannot open output file bin2h: Read-only file system
> collect2: error: ld returned 1 exit status
> Makefile:2: recipe for target 'all' failed
> make[1]: *** [all] Error 1
> make[1]: Leaving directory '/usr/src/mt7601-3.0.0.4/tools'
> Makefile:330: recipe for target 'build_tools' failed
> make: *** [build_tools] Error 2
> root@orangepipc:/usr/src/mt7601-3.0.0.4#
>

ThomasKaiser, 2016-05-20:
> Read-only file system

Seems you have a local problem too. And yes, it's already known that the sources aren't available any more at the old location but it needs some work to [fix this](https://github.com/igorpecovnik/lib/commit/4bcb5e51f73ad91f6a8d03bb2dbf8c27160f6456)

flexiti, 2016-05-20:
> Read-only file system

I suppose, however, that this may not be a local problem, simply copied the image from ambian.com and immediately ran this script. What kind of problem would that be? any suggestions ?

igorpecovnik, 2016-05-20:
We will fix this driver eventually but for now it's disabled.

Read only file system indicates some troubles with media, file system corruption. Check SD card quality, use other one.

flexiti, 2016-05-20:
All over again, another card (16G Kingston Class 10 UH1)
image from [http://www.armbian.com/orange-pi-pc/] 
cd /usr/src/linux-headers-$(uname -r)
make scripts

> mariuszb@orangepipc:~$ cd /usr/src/linux-headers-$(uname -r)
> mariuszb@orangepipc:/usr/src/linux-headers-3.4.112-sun8i$ make scripts
> scripts/kconfig/conf --silentoldconfig Kconfig
> fs/Kconfig:287: can't open file "fs/fs_compat.kconfig"
> /usr/src/linux-headers-3.4.112-sun8i/scripts/kconfig/Makefile:33: recipe for target 'silentoldconfig' failed
> make[2]: *** [silentoldconfig] Error 1
> /usr/src/linux-headers-3.4.112-sun8i/Makefile:493: recipe for target 'silentoldconfig' failed
> make[1]: *** [silentoldconfig] Error 2
> make: *** No rule to make target 'include/config/auto.conf', needed by 'scripts'.  Stop.
> mariuszb@orangepipc:/usr/src/linux-headers-3.4.112-sun8i$
> 

and  second part:
cd /usr/src/mt7601-3.0.0.4
make ARCH=arm

> root@orangepipc:/usr/src/mt7601-3.0.0.4# make ARCH=arm
> make -C tools
> make[1]: Entering directory '/usr/src/mt7601-3.0.0.4/tools'
> gcc -g bin2h.c -o bin2h
> make[1]: Leaving directory '/usr/src/mt7601-3.0.0.4/tools'
> /usr/src/mt7601-3.0.0.4/tools/bin2h
> cp -f os/linux/Makefile.6 /usr/src/mt7601-3.0.0.4/os/linux/Makefile
> make -C /lib/modules/3.4.112-sun8i/build SUBDIRS=/usr/src/mt7601-3.0.0.4/os/linux modules
> make[1]: Entering directory '/usr/src/linux-headers-3.4.112-sun8i'
>   CC [M]  /usr/src/mt7601-3.0.0.4/os/linux/../../os/linux/rt_profile.o
> /bin/sh: 1: /usr/src/linux-headers-3.4.112-sun8i/scripts/recordmcount: not found
> scripts/Makefile.build:307: recipe for target '/usr/src/mt7601-3.0.0.4/os/linux/../../os/linux/rt_profile.o' failed
> make[2]: *** [/usr/src/mt7601-3.0.0.4/os/linux/../../os/linux/rt_profile.o] Error 127
> Makefile:1368: recipe for target '_module_/usr/src/mt7601-3.0.0.4/os/linux' failed
> make[1]: *** [_module_/usr/src/mt7601-3.0.0.4/os/linux] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-3.4.112-sun8i'
> Makefile:394: recipe for target 'LINUX' failed
> make: *** [LINUX] Error 2
> root@orangepipc:/usr/src/mt7601-3.0.0.4#

zador-blood-stained, 2016-05-20:
Kernel headers scripts compilation is broken in this version. This will be fixed in new kernel release or you can install prebuilt kernel from [here](http://forum.armbian.com/index.php/topic/1231-testers-wanted-improving-ths-settings/#entry9269).

flexiti, 2016-05-20:
I do not know if I understood you well (I'm not good in Linux), but
- I did something like that: install prebuilt kernel You pointed me and:

> root@orangepipc:/home/mariuszb/Downloads# sudo dpkg --install linux-image-sun8i_5.11_armhf.deb
> (Reading database ... 152047 files and directories currently installed.)
> Preparing to unpack linux-image-sun8i_5.11_armhf.deb ...
> Unpacking linux-image-sun8i (5.11) over (5.10) ...
> Setting up linux-image-sun8i (5.11) ...
> update-initramfs: Generating /boot/initrd.img-3.4.112-sun8i
> root@orangepipc:/home/mariuszb/Downloads

-Then reboot
and again make scripts, the effect is the same

> mariuszb@orangepipc:/usr/src/linux-headers-3.4.112-sun8i$ make scripts
> scripts/kconfig/conf --silentoldconfig Kconfig
> fs/Kconfig:287: can't open file "fs/fs_compat.kconfig"

zador-blood-stained, 2016-05-20:
You need to install linux-headers package too.

flexiti, 2016-05-20:
Thank you. @zador-blood-stained 
I am one step further :)

> cd /usr/src/linux-headers-$(uname -r)
> make scripts

is ok now

Now, I did this:

> root@orangepipc:/usr/src/mt7601-3.0.0.4# make ARCH=arm

there were several warnings
but finally:

>  CC [M]  /usr/src/mt7601-3.0.0.4/os/linux/../../os/linux/usb_main_dev.o
>   CC [M]  /usr/src/mt7601-3.0.0.4/os/linux/../../common/rtusb_dev_id.o
>   CC [M]  /usr/src/mt7601-3.0.0.4/os/linux/../../common/frq_cal.o
>   LD [M]  /usr/src/mt7601-3.0.0.4/os/linux/mt7601Usta.o
>   Building modules, stage 2.
>   MODPOST 1 modules
>   CC      /usr/src/mt7601-3.0.0.4/os/linux/mt7601Usta.mod.o
>   LD [M]  /usr/src/mt7601-3.0.0.4/os/linux/mt7601Usta.ko
> make[1]: Leaving directory '/usr/src/linux-headers-3.4.112-sun8i'
> cp -f /usr/src/mt7601-3.0.0.4/os/linux/mt7601Usta.ko /tftpboot 2>/dev/null || :
> root@orangepipc:/usr/src/mt7601-3.0.0.4#

.... And I do not know what to do next :)

zador-blood-stained, 2016-05-20:
I don't have any MT7601 devices, so no idea.
Please check output of `modinfo mt7601Usta`, `lsmod` and `lsusb` to confirm that this module is properly installed and can handle your device.

flexiti, 2016-05-20:
root@orangepipc:/usr/src/mt7601-3.0.0.4/os/linux# **lsmod**
> Module                  Size  Used by
> mt7601Usta            564290  0      
> mali_drm                2732  1
> drm                   178255  2 mali_drm
> mali                  123208  0
> ump                    29379  3 mali
> 8189es                994547  0

root@orangepipc:/usr/src/mt7601-3.0.0.4/os/linux# **modinfo** mt7601Usta
> modinfo: ERROR: **Module mt7601Usta not found**.   

root@orangepipc:/usr/src/mt7601-3.0.0.4/os/linux# **lsusb**
> Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 007 Device 002: ID 1997:2433
> Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
> Bus 004 Device 002: ID 148f:7601 Ralink Technology, Corp.     **;; its ok**
> 

root@orangepipc:/usr/src/mt7601-3.0.0.4/os/linux# iwconfig wlan0

> wlan0     No such device

igorpecovnik, 2016-05-20:
Locate this file and try:

insmod mt7601Usta.ko

Check if driver was loaded

dmesg

flexiti, 2016-05-20:
@igorpecovnik  see above

flexiti, 2016-05-20:
@igorpecovnik @ThomasKaiser 
what types of wifi dongle work with Armbian? It can simply buy a instead of wasting time, although this type was described as acting - unfortunately does not work in the previous version and is no longer running in new version

igorpecovnik, 2016-05-20:
No idea, we don't maintain list of all supported devices by Linux kernel.
http://www.linux-drivers.org

Ask on forum what do they use.

flexiti, 2016-05-20:
Ufff, its working from current sources now :)  I have compiled latest Armbian source and then added wifi driver.
I have to organize everything what I've done but not today.
I'll let you know what I did if it needed
Most vital observation: This driver  is  shown as a **ra0**, not as **wlan0**
BTW. this is ubuntu image, I will check also on Debian
[image]

igorpecovnik, 2016-05-21:
Yes, not all of drivers show up as wlan0 ...area of (wireless) networking is big (mess). We only try to bring up chips which are soldered to the boards. Even this can be sometimes very challenging.

flexiti, 2016-05-21:
So.. (I puted copy of Mediatek driver on my git)

sudo apt-get install linux-headers-generic build-essential git
sudo apt-get install git
git clone https://github.com/flexiti/mt7601.git
cd mt7601
make clean
make
make install
mkdir -p /etc/Wireless/RT2870STA/
cp RT2870STA.dat /etc/Wireless/RT2870STA/
modprobe mt7601Usta

in addition in desktop mode in the network options I have changed network Wifi prefix from wlan0 to  ra0, and mark my Wifi network to be automatically connected

Tested on latest Armbian 5.11 Debian Jessy and  Armbian 5.11 Ubuntu Trusty) 
Network config example:
    

> auto ra0
>     iface ra0 inet static     
>     address 192.168.0.56
>     netmask 255.255.255.0
>     gateway 192.168.0.1 
>     wpa-ssid YOURSID
>     wpa-psk YOURPASS
>     dns-nameservers 8.8.8.8 8.8.4.4 

The original vendor driver can be downloaded from MediaTek's website (http://www.mediatek.com/en/downloads1/downloads/mt7601u-usb/).

@igorpecovnik  
Can You mark it as solved ?
BTW. I have connected only Wifi (192.168.0.56), but on terminal welcome message I see wired eth1 address (192.168.0.55)

igorpecovnik, 2016-05-21:
welcome screen is not working properly if connected via wlan ... need to fix once.

closed 2016-05-21 by igorpecovnik

Valerkav, 2016-05-25:
what is it, by step "make"?:
-------------------------------------------
**pi@raspberrypi:~/mt7601 $** make
make -C tools
make[1]: Entering directory '/home/pi/mt7601/tools'
gcc -g bin2h.c -o bin2h
make[1]: Leaving directory '/home/pi/mt7601/tools'
/home/pi/mt7601/tools/bin2h
cp -f os/linux/Makefile.6 /home/pi/mt7601/os/linux/Makefile
make -C /lib/modules/4.4.9+/build SUBDIRS=/home/pi/mt7601/os/linux modules
make[1]: *** /lib/modules/4.4.9+/build: No such file or directory.  Stop.
Makefile:394: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2
-----------------------------------------------------------

flexiti, 2016-05-25:
try .. sudo apt-get install linux-generic

saif619, 2017-05-14:
[image]
i am not able to install the driver , I have attached the screenshot of the error 
Please help

btmerz, 2017-07-02:
Hello,
I have the same problem with my OrangePi PC with Armbian and a MT7601 wifi dongle not working.
So I followd the instructions from flexiti:
sudo apt-get install linux-headers-generic build-essential git
sudo apt-get install git
git clone https://github.com/flexiti/mt7601.git
cd mt7601
make clean
make
make install
mkdir -p /etc/Wireless/RT2870STA/
cp RT2870STA.dat /etc/Wireless/RT2870STA/
modprobe mt7601Usta

The commands ran all without any error.
But after a reboot and checking with ifconfig there is still no new device visible. Any help yould be great.
Bernd

igorpecovnik, 2017-07-03:
https://forum.armbian.com/index.php?/search/&q=MT7601

EliasKotlyar, 2017-09-03:
Hello everyone,

My personal experience on compiling on a Olinuxino A20:
1. sudo apt-get install  build-essential git
(linux-headers-generic - is not avaible)
4. cd /usr/src/linux-headers-4.11.6-sunxi/scripts/
5. make recordmcount (needed cause its failing if not compiled)
2. cd /tmp
3. git clone https://github.com/EliasKotlyar/mt7601
cd mt7601
make clean
make
make install
mkdir -p /etc/Wireless/RT2870STA/
cp RT2870STA.dat /etc/Wireless/RT2870STA/
modprobe mt7601Usta

martinayotte, 2017-09-03:
From what I know, MT7601 is already part of the Mainline kernel ...

EliasKotlyar, 2017-09-03:
Hello martinayotte,

Yes, you are right. Its avaible on the mainline kernel from 4.2 upwards(https://github.com/art567/mt7601usta). Unfortunately, it was not working as expected - it gave me a lot of kernel messages when trying to connect. The connection was not working at all. After searching around, i found this issue and tried to compile this solution: https://github.com/flexiti/mt7601. It was not succesfull, because of some changes in the new kernels. After patching, and compiling, it started working...so i shared my solution for this problem. 

Greetings
Elias

pansila, 2018-03-20:
It works on kernel 4.9 (armbian 5.25) but crashs on kernel 4.11 (armbian 5.31) when associating to an AP.

igorpecovnik, 2018-03-20:
Modern kernels have this driver build in. I am not sure from which version but they are present in latest 4.14.y

pansila, 2018-03-20:
@igorpecovnik New mac80211-based mt7601 driver had been merged to upstream since 4.2.
https://github.com/torvalds/linux/commit/c869f77d6abb5d5f9f2f1a661d5c53862a9cad34

pansila, 2018-03-21:
Afte upgraded to 4.14 (Armbian 5.31), it works again.