armbian/build #312: check disque space before build

damien7851, 2016-05-20:
I have run this tools on tiny virtual machine.
It's work. but at the end fail because no left space on disc on raw create
I have freed space on disc but it's no possible to generate raw with data already generated.

it will be an good improvement for the future .

ThomasKaiser, 2016-05-20:
According to our documentation [you need 20GB free space](https://github.com/igorpecovnik/lib/blob/master/documentation/geek-faq.md#how-to-build-armbian-image-or-custom-kernel)

So you're reporting that you have +20GB free space but building fails due to a disk full situation?

damien7851, 2016-05-20:
I have not read correctly the documentation.
I intended just to add a check of free space.
compilation is long and it is better to avoid this kind of stupid errors.

zador-blood-stained, 2016-05-20:
It's impossible to know in advance how much space will be needed for compilation/image building attempt, and inserting free space checks before each step is counterproductive IMO.

damien7851, 2016-05-20:
Okay, but it's possible to add if 20gb are free at beginning only.

by cons should verify the proper end of dd otherwise it must wait until the copy fails rootfs.

it's an tiny improvement!

damien7851, 2016-05-24:
This issue should be marked improvement.

closed 2016-05-24 by damien7851