armbian/build #313: jessie 4.5.2 boot issue on pcduino3

williamcbc, 2016-05-20:
Hi, 

I've using Armbian_5.10_Pcduino3_Debian_jessie_4.5.2 on my pcDuino3, but the uboot always freeze  at "Uncompressing Linux... done, booting the kernel."
I used Rufus with dd mode to burn the raw file to the sdcard.
I've tried different sd-card but without luck, the uboot is still not moving forward.
I've also tried to fix the RTC issue, I added a fixtime.sh to the /bin and make a new boot.scr, still not working on this case. Is there any suggestion? Thanks!!

P.S. Attached is the boot log.

[bootlog.txt](https://github.com/igorpecovnik/lib/files/274998/bootlog.txt)

igorpecovnik, 2016-05-20:
First problem is that none of us have any PCduino board - support was added on theoretical base. It should work but ... 

I would need more than this - are you getting anything on HDMI screen? Can you enable console - ttyS0,115200 and verbosity, parameter loglevel set to 9 and recompile boot.cmd to boot.scr on some linux machine and try again.

williamcbc, 2016-05-21:
Hi Igor,

After set console to ttyS0,115200 in the boot.cmd, recompiled and after several time hardware reset, the system are booting normally! It seemed like in pcDuino3 boot.cmd, the console port is set to tty1 in default instead of ttyS0. And thanks for your help and your great work! Thanks!

igorpecovnik, 2016-05-21:
Nice to hear this.

closed 2016-05-21 by igorpecovnik